raspberrypi/firmware #1782: hdmi_enable_4kp60 breaks when an overclock is applied to the HEVC freq.

ryanteck, 2023-01-29:
**Is this the right place for my bug report?**
Relates to HEVC Frequency adjustments so I would say that this is the GPU firmware.

**Describe the bug**
The `hdmi_enable_4kp60` setting when enabled in the configuration file, seems to break / conflict manually specifying a frequency for the HEVC block (using `hevc_freq`).

As per https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/config_txt/overclocking.adoc#specific-to-raspberry-pi-4-raspberry-pi-400-and-cm4 when the `hdmi_enable_4kp60` function is enabled the core clock is automatically overclocked to 550Mhz for this to work.

When I then adjust the `hevc_freq` frequency to anything other than 550 it seems to then cause an issue where it will not be able to initialise the connection to the TV properly.

If I set the `hevc_freq` manually to 550 then it works perfectly fine, with `force_turbo` enabled we can see that the frequencies of the core, v3d and havoc are all at 550. This is the same as if there’s no frequency manually specified for this.
```
frequency(48)=1500345728
frequency(1)=549993152
frequency(46)=550006336
frequency(76)=549993152
```

But then even the smallest of overclocks to the hevc core (555Mhz) seems to break it.
```
frequency(48)=1500345728
frequency(1)=277501472
frequency(46)=277501472
frequency(76)=555002944
```

The interesting thing is that the v3d and core frequency then seem to halved. But even setting these manually to 550 they still seem to run at the reduced frequency.

**To reproduce**
1) Flash Raspberry Pi OS (I used with Desktop, Non Full, 32 Bit) to an SD card.
2) Add / uncomment and adjust the following lines in the main section of /boot/config.txt:
```
# Enable DRM VC4 V3D driver
dtoverlay=vc4-fkms-v3d,cma-512
max_framebuffers=1
dtoverlay=rpivid-v4l2
```
3) Add the following to the end of the /boot/config.txt file under the [all] section.
```
[all]
gpu_mem=128
enable_uart=1
hdmi_enable_4kp60=1
#hevc_freq=555
force_turbo=1
```
4) Connect it to a 4K Display (I’ve only got 1 4K TV to test with so have been using this) and power the Pi up
5) Confirm that display works as intended, then edit the /boot/config.txt and uncomment the hevc_freq=555 line.
6) Reboot the Pi, confirm if the display has initialised as intended.

**Expected behaviour**
I would expect the other cores to stay at the same frequency, and just the HEVC core to increase (which happens with `hdmi_enable_4kp60` disabled).

```
frequency(48)=1500345728
frequency(1)=277501472
frequency(46)=277501472
frequency(76)=555002944
```

**Actual behaviour**
The other cores seem to fail to keep their frequency, and it fails to initialise a 4kp60 display.
```
frequency(48)=1500345728
frequency(1)=277501472
frequency(46)=277501472
frequency(76)=555002944
```

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? Raspberry Pi 4, 2GB, Rev 1.1 & 1.2 (tested on 2 Pis)
* Which OS and version (`cat /etc/rpi-issue`) Raspberry Pi reference 2022-09-22 Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 8a42abcd1dbd8c9c1fdfca4e0c3778255b2f9cc4, stage4
* Which firmware version (`vcgencmd version`) Jan  5 2023 10:46:54 Copyright (c) 2012 Broadcom version 8ba17717fbcedd4c3b6d4bce7e50c7af4155cba9 (clean) (release) (start)
* Which kernel version (`uname -a`)? Linux moonpi1 5.15.84-v7l+ #1613 SMP Thu Jan 5 12:01:26 GMT 2023 armv7l GNU/Linux

I’ve attached the raspinfo below, along with this I have tried 2 different Pi 4s with different HDMI cables. The only thing I’m unable to rule out is my TV as I’ve only got the 1 4K :)

**Logs**
If applicable, add the relevant output from `dmesg` or similar.
In the logfile attached.
[raspiinfo.txt](https://github.com/raspberrypi/firmware/files/10529528/raspiinfo.txt)

**Additional context**
I believe that's all the information, the reason why I'm trying to do this is that the 4K streaming I'm trying to do is a bit stuttery and I've found that overclocking the HEVC block whilst connected to a 1080p TV resolves the stuttering. However it's then a shame that I can't seem to do it when actually on my 4K TV.

popcornmix, 2023-01-29:
I'll look into the specifics during the week, but the real issue is the five gpu clocks (core, h264, hevc, isp, v3d) are not independent. They share a PLL, which means only one gets the exact frequency, the other's just get the closest they can from integer dividers of this PLL.

The fix is to only use gpu_freq=600 (or whatever frequency you want) which will boost them all, and so won't suffer from limitations of a shared PLL.

ryanteck, 2023-01-29:
Thanks for getting back so quick exp over the weekend @popcornmix ,

It seems that `gpu_freq` also doesn't work alongside the `hdmi_enable_4kp60` command either. I've set it to 900Mhz (to see if that works) whilst commenting out the other lines so my config now looks like this:

```
[all]
gpu_mem=128
enable_uart=1
hdmi_enable_4kp60=1
#core_freq=550
#v3d_freq=550
#hevc_freq=555
gpu_freq=900
over_voltage=12
force_turbo=1
```

But the freqs all seem to have stayed at 550Mhz still:
```
frequency(48)=1500345728
frequency(1)=549993152
frequency(46)=550006336
frequency(76)=549993152
```

The TV does connect successfully but no overclock 

Interestingly when I run raspinfo you can see that it thinks / decodes that the cores should be all set at 550. 

```
config.txt
----------

arm_boost=1
arm_freq=1500
audio_pwm_mode=514
camera_auto_detect=1
config_hdmi_boost=5
core_freq=550
core_freq_min=220
disable_commandline_tags=2
disable_l2cache=1
disable_overscan=1
display_auto_detect=1
display_hdmi_rotate=-1
display_lcd_rotate=-1
dvfs=3
enable_gic=1
enable_uart=1
force_eeprom_read=1
force_pwm_open=1
force_turbo=1
framebuffer_ignore_alpha=1
framebuffer_swap=1
gpu_freq=900
gpu_freq_min=250
h264_freq=550
hdmi_enable_4kp60=1
hdmi_samples_limit=0x1f4000
hevc_freq=550
init_uart_clock=0x2dc6c00
isp_freq=550
lcd_framerate=60
mask_gpu_interrupt0=1024
mask_gpu_interrupt1=0x10000
max_framebuffer_height=2160
max_framebuffer_width=3840
max_framebuffers=2
over_voltage=12
over_voltage_avs=-36250
pause_burst_frames=1
program_serial_random=1
total_mem=2048
v3d_freq=550
hdmi_force_cec_address:0=65535
hdmi_force_cec_address:1=65535
hdmi_pixel_freq_limit:0=0x23c34600
hdmi_pixel_freq_limit:1=0x9a7ec80
device_tree=-
overlay_prefix=overlays/
hdmi_cvt:0=
hdmi_cvt:1=
hdmi_edid_filename:0=
hdmi_edid_filename:1=
hdmi_timings:0=
hdmi_timings:1=
```

ryanteck, 2023-01-29:
Well I've done some more experimenting throughout the day.

The gpu_freq command seems to get ignored with the `hdmi_enabled_4kp60` option. However from the info on the above raspinfo commands I've realised which options that it's then setting.

It does look correct that the pll is to blame, primarily directly between that and the clock frequency seems to be exactly half if I don't set it to be the same clock frequencies.

This seems to then get close enough, I'll leave this issue open but it's up to you if you want to close it. Maybe the better way to re-word it is that `hdmi_enable_4kp60` ignores `gpu_freq`. 

```
[all]
gpu_mem=128
enable_uart=1
#hdmi_enable_4kp60=0
over_voltage=10
force_turbo=1
# Try to replicate hdmi_enable_4kp60
hdmi_pixel_freq_limit:0=0x23c34600
hdmi_pixel_freq_limit:1=0x9a7ec80
max_framebuffer_height=2160
max_framebuffer_width=3840
gpu_freq=850
core_freq=850
core_freq_min=220
```