raspberrypi/firmware #1784: Unexpected video= parameter pops on the kernel command line

substring, 2023-02-09:
**Describe the bug**
I'm trying to output to a CRT through HDMI since it's the only way to have KMS + DRI, using the VGA666 doesn't produce /dev/dri.

Setting a video mode on Pi4 + KMS driver is complicated, the bootloader (or I don't know what) adds unwanted kernel parameters like `video=HDMI-A-1:640x480M@60` either based on the hdmi_timing, either from nowhere.

When forcing an EDID in config.txt and kernel command line, there is still a mode set by video= and is the one set for the framebuffer, and is wrong for CRTs, resulting in no display, or an ever scrolling image.

CVT timings are mostly not good for CRTs.

The only case I got a stable picture on a CRT TV is using a mode which would overlap the one generated by the command line, discovered that with pure luck + fkms

**To reproduce**
Simply set a hdmi_timing, hdmi_cvt or an EDID, then `cat /proc/cmdline` to see that unexected video= parameter

**Expected behaviour**
this video= shouldn't be added when an edid is forced (from cmdline.txt or config.txt). I understand it may be necessary with hdmi_timings since you need a way to pass down the resolution. A parameter to disable adding that video= would be warmly welcome

**Actual behaviour**
Setting a resolution by any mean in config.txt is translated to a `video=` parameter, without any control over it

**System**
[raspinfo](https://gist.github.com/substring/1fbf80509fed4965bb231f8270ef5304)

**Logs**

**Additional context**

6by9, 2023-02-10:
https://www.raspberrypi.com/documentation/computers/config_txt.html#disable_fw_kms_setup
```disable_fw_kms_setup=1```

substring, 2023-02-10:
AWESOME THANK YOU !

closed 2023-02-10 by substring