raspberrypi/firmware #1785: EEPROM doesn't provide any information whether it was booted from ramdisk

amaksoft, 2023-02-13:
If i set up `boot_ramdisk=1` in my `config.txt` and EEPROM loads my next stage bootloader from `boot.img` file it's not reflected in the device tree `/chosen/bootloader`.
Since the ramdisk may contain some extra metadata files for my next stage bootloader it would really help to know that i need to read that metadata from there. Since main use case is secure boot, it makes sense to have a single signed image to hold all files. Does it make sense to you?

If you agree, i suggest to add some `uint32_t` flags to` /chosen/bootloader` that will reflect that it was booted from ramdisk and also whether secure boot was enabled. Also it would really help to have memory address where ramdisk was loaded and it's size (if you keep it in memory of course. Otherwise at least give enough information for the next boot stage to access it)
e.g. 
```dts
chosen {
        ....
        bootloader {
                boot-mode = <0x00000001>;
                boot-flags = <0x00000003>; # <- bit0 = 1 => ramdisk mode, bit1 = 1 => secure boot enabled
                ramdisk-addr = <0x0000c0de 0x00f12345>; # <- ramdisk was loaded into 0x0000c0de it's length is 0x00f12345
                signature-addr = <0x0000c0de 0x00f12345>; # <- ramdisk signature was loaded into 0x0000c0de it's length is 0x00f12345
                ....
        };
};
```

amaksoft, 2023-02-14:
That's what i do right now. But config rather represents _requested state_, not _actual state_.
So it's still a bit of a guess from the next stage perspective.
Device tree's `/chosen` section is the second stage bootloader _actual state_ reporting API.

timg236, 2023-02-14:
I've tagged this as an enhancement in-case someone wants to implement the 'ramdisk mode' indicator via DT. 

Passing data to the OS should really go via initramfs so I think that aspect would need much more thought