raspberrypi/firmware #1788: Unable to auto detect 2560x1440 monitors

IanSB, 2023-02-21:
I'm working on this bare metal vintage computer scaler project which runs on the Pi zero and zero 2W:
https://github.com/IanSB/RGBtoHDMI/wiki
and I want to add auto detection support for 2560x1440 and 4K monitors.

I've added the following to the config.txt file:
hdmi_pixel_freq_limit=340000000
which is well above the 241.1 Mhz frequency required by 2560x1440 but the Pi will not auto detect and display that resolution.

If I connect the Pi to a 4K monitor it will select and display 3840x2160@30Hz as expected (that has an even higher clock frequency).

When connected to my BenQ PD3200Q 2560x1440 monitor:
If hdmi_group is set to 0 or 1 (auto detect or CEA) then the Pi will output 1280x720@60Hz
If hdmi_group is set to 2 (DMT) then the Pi will output 1920x1080@60Hz

I also have an IIyama 2560x1440 monitor and that displays the following:
If hdmi_group is set to 0 or 1 (auto detect or CEA) then the Pi will output 1920x1080@60Hz
If hdmi_group is set to 2 (DMT) then the Pi will output 2048x1152@60Hz which is a really obscure resolution although it is defined as the last entry in Standard Display Modes.

I've dumped the EDID of both monitors and attached them both as binary and text files in hex. If you copy the contents of a text file and paste it here you can view the EDID info:
http://www.edidreader.com/

This shows that the full timing info for 2560x1440 is included in Block 0 Standard Timing Information Descriptor for both monitors.

I also note that 1280x720 is the first entry in the Block 1 descriptors on the BenQ and it seems to be using that in CEA mode.
On the IIyama the first entry in the Block 1 descriptors is 1920x1080 which probably explains the above difference

[EDID-2560x1440.zip](https://github.com/raspberrypi/firmware/files/10797844/EDID-2560x1440.zip)

I can get the correct display by manually setting the resolution in config.txt but I would like to use auto detect as that works well with all other monitor resolutions and makes the converter plug and play.
I've tried the latest master and stable builds.

6by9, 2023-02-22:
The firmware has only ever automatically adopted standard CEA or DMT modes. 2560x1440 is not a standard mode, therefore it will never be selected.

Your best bet would be to use an EDID= filter in config.txt to detect your particular monitor and apply the relevant hdmi_mode, hdmi_group, and hdmi_timings settings.

IanSB, 2023-02-22:
>Your best bet would be to use an EDID= filter in config.txt to detect your particular monitor and apply the relevant hdmi_mode, hdmi_group, and hdmi_timings settings.

That wouldn't help as I'm trying to support any monitor, not just a specific one.

One other option would be to force higher resolution monitors to fall back to 1080p when auto resolution is selected by limiting the bandwidth to 165Mhz but even that doesn't work as there is inconsistent behaviour in CEA mode where it's picking 1280x720 instead of 1920x1080 with the BenQ monitor.
All standard 1920x1080 modes are listed in the CEA video data block for the BenQ and one is flagged as the "Native" resolution (for the purposes of CEA mode) and that is the one I would expect to be chosen instead of 1280x720:

Native: TRUE
VIC: 16
Format: 1920x1080p
Field Rate: 59.94Hz/60Hz
Picture AR: 16:9
Pixel AR: 1:1

Why is the CEA native resolution being ignored?

popcornmix, 2023-02-22:
```
pi@pi4:~ $ /opt/vc/bin/edidparser BenQ-EDID.bin 
Enabling fuzzy format match...
Parsing BenQ-EDID.bin...
HDMI:EDID version 1.3, 1 extensions, screen size 71x40 cm
HDMI:EDID features - videodef 0x80 !standby !suspend active off; colour encoding:RGB444|YCbCr444|YCbCr422; sRGB is default colourspace; preferred format is native; does not support GTF
HDMI:EDID found monitor S/N descriptor tag 0xff
HDMI:EDID found monitor range descriptor tag 0xfd
HDMI:EDID monitor range offsets: V min=0, V max=0, H min=0, H max=0
HDMI:EDID monitor range: vertical is 50-76 Hz, horizontal is 30-90 kHz, max pixel clock is 270 MHz
HDMI:EDID monitor range does not support GTF
HDMI:EDID found monitor name descriptor tag 0xfc
HDMI:EDID monitor name is BenQ_PD3200Q
HDMI:EDID failed to find a matching detail format for 2560x1440p hfp:47 hs:32 hbp:81 vfp:3 vs:5 vbp:33 pixel clock:241 MHz
HDMI:EDID calculated refresh rate is 60 Hz
HDMI:EDID guessing the format to be 2560x1440p @60 Hz
HDMI:EDID found unknown detail timing format: 2560x1440p hfp:47 hs:32 hbp:81 vfp:3 vs:5 vbp:33 pixel clock:241 MHz
HDMI:EDID established timing I/II bytes are 25 6B 80
HDMI:EDID found DMT format: code 4, 640x480p @ 60 Hz in established timing I/II
HDMI:EDID found DMT format: code 6, 640x480p @ 75 Hz in established timing I/II
HDMI:EDID found DMT format: code 9, 800x600p @ 60 Hz in established timing I/II
HDMI:EDID found DMT format: code 11, 800x600p @ 75 Hz in established timing I/II
HDMI:EDID found DMT format: code 16, 1024x768p @ 60 Hz in established timing I/II
HDMI:EDID found DMT format: code 18, 1024x768p @ 75 Hz in established timing I/II
HDMI:EDID found DMT format: code 36, 1280x1024p @ 75 Hz in established timing I/II
HDMI:EDID standard timings block x 8: 0x8180 81C0 8100 A9C0 B300 D1C0 0101 0101 
HDMI:EDID found DMT format: code 35, 1280x1024p @ 60 Hz (5:4) in standard timing 0
HDMI:EDID found DMT format: code 85, 1280x720p @ 60 Hz (16:9) in standard timing 1
HDMI:EDID found DMT format: code 28, 1280x800p @ 60 Hz (16:10) in standard timing 2
HDMI:EDID found DMT format: code 83, 1600x900p @ 60 Hz (16:9) in standard timing 3
HDMI:EDID found DMT format: code 58, 1680x1050p @ 60 Hz (16:10) in standard timing 4
HDMI:EDID found DMT format: code 82, 1920x1080p @ 60 Hz (16:9) in standard timing 5
HDMI:EDID parsing v3 CEA extension 0
HDMI:EDID monitor support - underscan IT formats:yes, basic audio:yes, yuv444:yes, yuv422:yes, #native DTD:1
HDMI:EDID found preferred CEA detail timing format: 1280x720p @ 60 Hz (4)
HDMI:EDID found CEA detail timing format: 720x480p @ 60 Hz (2)
HDMI:EDID found CEA detail timing format: 720x576p @ 50 Hz (17)
HDMI:EDID found CEA detail timing format: 720x576p @ 50 Hz (17)
HDMI:EDID found CEA format: code 16, 1920x1080p @ 60Hz (native)
HDMI:EDID found CEA format: code 31, 1920x1080p @ 50Hz 
HDMI:EDID found CEA format: code 34, 1920x1080p @ 30Hz 
HDMI:EDID found CEA format: code 33, 1920x1080p @ 25Hz 
HDMI:EDID found CEA format: code 32, 1920x1080p @ 24Hz 
HDMI:EDID found CEA format: code 5, 1920x1080i @ 60Hz 
HDMI:EDID found CEA format: code 20, 1920x1080i @ 50Hz 
HDMI:EDID found CEA format: code 4, 1280x720p @ 60Hz 
HDMI:EDID found CEA format: code 19, 1280x720p @ 50Hz 
HDMI:EDID found CEA format: code 18, 720x576p @ 50Hz 
HDMI:EDID found CEA format: code 17, 720x576p @ 50Hz 
HDMI:EDID found CEA format: code 3, 720x480p @ 60Hz 
HDMI:EDID found CEA format: code 2, 720x480p @ 60Hz 
HDMI:EDID found CEA format: code 1, 640x480p @ 60Hz 
HDMI:EDID found CEA format: code 7, 1440x480i @ 60Hz 
HDMI:EDID found CEA format: code 6, 1440x480i @ 60Hz 
HDMI:EDID found audio format 2 channels PCM, sample rate: 32|44|48 kHz, sample size: 16|20|24 bits
HDMI:EDID found HDMI VSDB length 7
HDMI:EDID HDMI VSDB has physical address 1.0.0.0
HDMI:EDID HDMI VSDB supports AI:no, dual link DVI:no
HDMI:EDID HDMI VSDB deep colour support - 48-bit:no 36-bit:yes 30-bit:yes DC_yuv444:yes
HDMI:EDID HDMI VSDB max TMDS clock 250 MHz
HDMI:EDID HDMI VSDB has no latency information
HDMI:EDID filtering formats with pixel clock unlimited MHz or h. blanking unlimited
HDMI:EDID best score mode initialised to CEA (1) 640x480p @ 60 Hz with pixel clock -140642532 MHz (score 25)
HDMI:EDID best score mode is now CEA (1) 640x480p @ 60 Hz with pixel clock 25 MHz (score 61864)
HDMI:EDID best score mode is now CEA (2) 720x480p @ 60 Hz with pixel clock 27 MHz (score 66472)
HDMI:EDID CEA mode (3) 720x480p @ 60 Hz with pixel clock 27 MHz has a score of 66472
HDMI:EDID best score mode is now CEA (4) 1280x720p @ 60 Hz with pixel clock 74 MHz (score 5190888)
HDMI:EDID DMT mode (4) 640x480p @ 60 Hz with pixel clock 25 MHz has a score of 18432
HDMI:EDID CEA mode (5) 1920x1080i @ 60 Hz with pixel clock 74 MHz has a score of 149416
HDMI:EDID CEA mode (6) 1440x480i @ 60 Hz with pixel clock 27 MHz has a score of 45736
[... 5 lines cut ...]
HDMI:EDID DMT mode (16) 1024x768p @ 60 Hz with pixel clock 65 MHz has a score of 47185
HDMI:EDID CEA mode (17) 720x576p @ 50 Hz with pixel clock 27 MHz has a score of 66472
HDMI:EDID CEA mode (18) 720x576p @ 50 Hz with pixel clock 27 MHz has a score of 66472
HDMI:EDID DMT mode (18) 1024x768p @ 75 Hz with pixel clock 78 MHz has a score of 14745
HDMI:EDID CEA mode (19) 1280x720p @ 50 Hz with pixel clock 74 MHz has a score of 117160
HDMI:EDID CEA mode (20) 1920x1080i @ 50 Hz with pixel clock 74 MHz has a score of 128680
HDMI:EDID DMT mode (28) 1280x800p @ 60 Hz with pixel clock 83 MHz has a score of 86440
[... cut ...]
HDMI:EDID DMT mode (83) 1600x900p @ 60 Hz with pixel clock 108 MHz has a score of 111400
HDMI:EDID DMT mode (85) 1280x720p @ 60 Hz with pixel clock 74 MHz has a score of 80296
HDMI0:EDID preferred mode remained as CEA (4) 1280x720p @ 60 Hz with pixel clock 74 MHz
HDMI:EDID has HDMI support and audio support
/opt/vc/bin/edidparser exited with code 0
```
The firmware treats the first CEA detailed timing listed in edid as preferred:
```
HDMI:EDID found preferred CEA detail timing format: 1280x720p @ 60 Hz (4)
```

pelwell, 2023-02-22:
What does `preferred format is native` in that log indicate?

pelwell, 2023-02-22:
In trying to answer my own question I found this: https://en.wikipedia.org/wiki/Extended_Display_Identification_Data#Structure,_version_1.4
The description for bit 1 of byte 24 says:
> Preferred timing mode specified in descriptor block 1. For EDID 1.3+ the preferred timing mode is always in the first Detailed Timing Descriptor. In that case, this bit specifies whether the preferred timing mode includes native pixel format and refresh rate.

Which to the uninitiated like me says that the firmware is behaving correctly.

IanSB, 2023-02-23:
>The description for bit 1 of byte 24 says:
Preferred timing mode specified in descriptor block 1. For EDID 1.3+ the preferred timing mode is always in the first Detailed Timing Descriptor. In that case, this bit specifies whether the preferred timing mode includes native pixel format and refresh rate.

That bit is in the standard EDID and refers to the first Detailed Timing Descriptor in that part which is 2560x1440 as shown in the above EDID parser output with the intention that those detailed timings are to be used to generate that resolution when in DMT mode which is how non-standard resolutions are supported but that functionality isn't implemented as indicated above.

The CEA description in wikipedia is a little ambiguous as it indicates that the first detailed timing descriptor in the CEA extension should be used in version 1 but it is not totally clear if that applies to later ones.

Anyway, I have managed to work around the problem by reading the EDID in my code and dynamically rewriting config.txt with an appropriate mode definition if a 2560x1440 monitor is detected. This does require an additional automatic reboot when moving to and from such a monitor which is a little messy but as this is a bare metal project the boot time is only a few seconds.

closed 2023-02-24 by IanSB