raspberrypi/firmware #1791: RPI 3B fails with 7 blinks on green while booting

pelerner, 2023-03-12:
RPI 3B fails after the recent update ("Raspberry Pi 3 Model B Rev 1.2 with 1GB")
Systems doesn't come online, just gives seven blinks on green.

The following Arch Linux Arm packages caused the problem ...

- raspberrypi-bootloader-20230310-1
- linux-rpi-6.1.16-3-aarch64

Same SD-Card runs fine on Raspberry Pi 4 Model B Rev 1.5 with 8GB.
For reference see https://archlinuxarm.org/forum/viewtopic.php?f=67&t=16418

Last-known-good commit is:
https://github.com/raspberrypi/firmware/commit/889323796bb8feb99c7a96ba5e6445895cab47ce

Breaking commit is:
https://github.com/raspberrypi/firmware/commit/379d5bfa60bf3b0afff413a18344aaaf3bde0083

Peter

pelwell, 2023-03-12:
Which kernel files are present on the card? And what are the config.txt settings?

pelerner, 2023-03-12:
Files

```
d-----        09.06.2021     14:41                overlays                                                             
-a----        08.06.2021     15:17          25870 bcm2708-rpi-b-plus.dtb                                               
-a----        08.06.2021     15:17          25218 bcm2708-rpi-b-rev1.dtb                                               
-a----        08.06.2021     15:17          25607 bcm2708-rpi-b.dtb                                                    
-a----        08.06.2021     15:17          25529 bcm2708-rpi-cm.dtb                                                   
-a----        08.06.2021     15:17          26545 bcm2708-rpi-zero-w.dtb                                               
-a----        08.06.2021     15:17          25352 bcm2708-rpi-zero.dtb                                                 
-a----        08.06.2021     15:17          26745 bcm2709-rpi-2-b.dtb                                                  
-a----        08.06.2021     15:17          26894 bcm2710-rpi-2-b.dtb                                                  
-a----        08.06.2021     15:17          29011 bcm2710-rpi-3-b-plus.dtb                                             
-a----        08.06.2021     15:17          28392 bcm2710-rpi-3-b.dtb                                                  
-a----        08.06.2021     15:17          26890 bcm2710-rpi-cm3.dtb                                                  
-a----        08.06.2021     15:17          49218 bcm2711-rpi-4-b.dtb                                                  
-a----        08.06.2021     15:17          49214 bcm2711-rpi-400.dtb                                                  
-a----        08.06.2021     15:17          49892 bcm2711-rpi-cm4.dtb                                                  
-a----        08.06.2021     15:17          18554 bcm2835-rpi-a-plus.dtb                                               
-a----        08.06.2021     15:17          18434 bcm2835-rpi-a.dtb                                                    
-a----        08.06.2021     15:17          18841 bcm2835-rpi-b-plus.dtb                                               
-a----        08.06.2021     15:17          18713 bcm2835-rpi-b-rev2.dtb                                               
-a----        08.06.2021     15:17          18580 bcm2835-rpi-b.dtb                                                    
-a----        08.06.2021     15:17          18695 bcm2835-rpi-cm1-io1.dtb                                              
-a----        08.06.2021     15:17          18949 bcm2835-rpi-zero-w.dtb                                               
-a----        08.06.2021     15:17          18506 bcm2835-rpi-zero.dtb                                                 
-a----        08.06.2021     15:17          19833 bcm2836-rpi-2-b.dtb                                                  
-a----        08.06.2021     15:17          20120 bcm2837-rpi-3-a-plus.dtb                                             
-a----        08.06.2021     15:17          20989 bcm2837-rpi-3-b-plus.dtb                                             
-a----        08.06.2021     15:17          20525 bcm2837-rpi-3-b.dtb                                                  
-a----        08.06.2021     15:29          52456 bootcode.bin                                                         
-a----        08.06.2021     15:17            161 cmdline.txt                                                          
-a----        08.06.2021     15:17            120 config.txt                                                           
-a----        08.06.2021     15:29           7310 fixup.dat                                                            
-a----        08.06.2021     15:29           5443 fixup4.dat                                                           
-a----        08.06.2021     15:29           3188 fixup4cd.dat                                                         
-a----        08.06.2021     15:29           8450 fixup4db.dat                                                         
-a----        08.06.2021     15:29           8454 fixup4x.dat                                                          
-a----        08.06.2021     15:29           3188 fixup_cd.dat                                                         
-a----        08.06.2021     15:29          10300 fixup_db.dat                                                         
-a----        08.06.2021     15:29          10298 fixup_x.dat                                                          
-a----        09.06.2021     14:42        6803552 initramfs-linux.img                                                  
-a----        08.06.2021     15:17        8584808 kernel7.img                                                          
-a----        08.06.2021     15:29        2956576 start.elf                                                            
-a----        08.06.2021     15:29        2232384 start4.elf                                                           
-a----        08.06.2021     15:29         796284 start4cd.elf                                                         
-a----        08.06.2021     15:29        3728168 start4db.elf                                                         
-a----        08.06.2021     15:29        2984424 start4x.elf                                                          
-a----        08.06.2021     15:29         796284 start_cd.elf                                                         
-a----        08.06.2021     15:29        4800168 start_db.elf                                                         
-a----        08.06.2021     15:29        3708072 start_x.elf                                                          
-a----        08.06.2021     15:17          19852 bcm2837-rpi-cm3-io3.dtb
```

Config.txt:
```
# See /boot/overlays/README for all available options

dtoverlay=vc4-kms-v3d
initramfs initramfs-linux.img followkernel
```

hilderingt, 2023-03-13:
Same problem here:

Raspberry Pi 3 B+
Kernel: 6.1.6 aarch64

Files:
`/boot/bcm2710-rpi-2-b.dtb
/boot/bcm2710-rpi-3-b.dtb
/boot/bcm2710-rpi-3-b-plus.dtb
/boot/bcm2710-rpi-cm3.dtb
/boot/bcm2710-rpi-zero-2.dtb
/boot/bcm2710-rpi-zero-2-w.dtb
/boot/bcm2711-rpi-400.dtb
/boot/bcm2711-rpi-4-b.dtb
/boot/bcm2711-rpi-cm4.dtb
/boot/bcm2711-rpi-cm4-io.dtb
/boot/bcm2711-rpi-cm4s.dtb
/boot/bcm2837-rpi-3-a-plus.dtb
/boot/bcm2837-rpi-3-b.dtb
/boot/bcm2837-rpi-3-b-plus.dtb
/boot/bcm2837-rpi-cm3-io3.dtb
/boot/bcm2837-rpi-zero-2-w.dtb
/boot/bootcode.bin
/boot/cmdline.txt
/boot/config.txt
/boot/fixup4cd.dat
/boot/fixup4.dat
/boot/fixup4db.dat
/boot/fixup4x.dat
/boot/fixup_cd.dat
/boot/fixup.dat
/boot/fixup_db.dat
/boot/fixup_x.dat
/boot/initramfs-linux.img
/boot/kernel8.img
/boot/overlays
/boot/start4cd.elf
/boot/start4db.elf
/boot/start4.elf
/boot/start4x.elf
/boot/start_cd.elf
/boot/start_db.elf
/boot/start.elf
/boot/start_x.elf`

config.txt:
`# See /boot/overlays/README for all available options

dtoverlay=vc4-kms-v3d
initramfs initramfs-linux.img followkernel

display_auto_detect=1

# Uncomment to enable bluetooth
#dtparam=krnbt=on

[pi4]
# Run as fast as firmware / board allows
`

hilderingt, 2023-03-13:
Same here.

Raspberry Pi 3 B+
raspberrypi-bootloader-20230310-1
linux-rpi-6.1.16-3-aarch64

Files:
```
/boot/bcm2710-rpi-2-b.dtb
/boot/bcm2710-rpi-3-b.dtb
/boot/bcm2710-rpi-3-b-plus.dtb
/boot/bcm2710-rpi-cm3.dtb
/boot/bcm2710-rpi-zero-2.dtb
/boot/bcm2710-rpi-zero-2-w.dtb
/boot/bcm2711-rpi-400.dtb
/boot/bcm2711-rpi-4-b.dtb
/boot/bcm2711-rpi-cm4.dtb
/boot/bcm2711-rpi-cm4-io.dtb
/boot/bcm2711-rpi-cm4s.dtb
/boot/bcm2837-rpi-3-a-plus.dtb
/boot/bcm2837-rpi-3-b.dtb
/boot/bcm2837-rpi-3-b-plus.dtb
/boot/bcm2837-rpi-cm3-io3.dtb
/boot/bcm2837-rpi-zero-2-w.dtb
/boot/bootcode.bin
/boot/cmdline.txt
/boot/config.txt
/boot/fixup4cd.dat
/boot/fixup4.dat
/boot/fixup4db.dat
/boot/fixup4x.dat
/boot/fixup_cd.dat
/boot/fixup.dat
/boot/fixup_db.dat
/boot/fixup_x.dat
/boot/initramfs-linux.img
/boot/kernel8.img
/boot/overlays
/boot/start4cd.elf
/boot/start4db.elf
/boot/start4.elf
/boot/start4x.elf
/boot/start_cd.elf
/boot/start_db.elf
/boot/start.elf
/boot/start_x.elf
```

config.txt:
```
# See /boot/overlays/README for all available options

dtoverlay=vc4-kms-v3d
initramfs initramfs-linux.img followkernel

display_auto_detect=1

# Uncomment to enable bluetooth
#dtparam=krnbt=on

[pi4]
# Run as fast as firmware / board allows
arm_boost=1
```

df8oe, 2023-03-13:
Same here.

Raspberry Pi 3 B+
- raspberrypi-bootloader-20230310-1
- linux-rpi-6.2.3-2-aarch64

Files:
`   30436 11. Mär 12:45 bcm2710-rpi-2-b.dtb
    32188 11. Mär 12:45 bcm2710-rpi-3-b.dtb
    32799 11. Mär 12:45 bcm2710-rpi-3-b-plus.dtb
    30331 11. Mär 12:45 bcm2710-rpi-cm3.dtb
    31364 11. Mär 12:45 bcm2710-rpi-zero-2.dtb
    31364 11. Mär 12:45 bcm2710-rpi-zero-2-w.dtb
    52743 11. Mär 12:45 bcm2711-rpi-400.dtb
    52695 11. Mär 12:45 bcm2711-rpi-4-b.dtb
    53276 11. Mär 12:45 bcm2711-rpi-cm4.dtb
    38349 11. Mär 12:45 bcm2711-rpi-cm4-io.dtb
    50578 11. Mär 12:45 bcm2711-rpi-cm4s.dtb
    20713 11. Mär 12:45 bcm2837-rpi-3-a-plus.dtb
    20979 11. Mär 12:45 bcm2837-rpi-3-b.dtb
    21443 11. Mär 12:45 bcm2837-rpi-3-b-plus.dtb
    20098 11. Mär 12:45 bcm2837-rpi-cm3-io3.dtb
    20524 11. Mär 12:45 bcm2837-rpi-zero-2-w.dtb
    52460 28. Sep 04:11 bootcode.bin
      149 21. Feb 19:50 cmdline.txt
       84 11. Mär 12:45 cmdline.txt.pacnew
      603 19. Okt 08:06 config.txt
      254 11. Mär 12:45 config.txt.pacnew
     3170 28. Sep 04:11 fixup4cd.dat
     5399 28. Sep 04:11 fixup4.dat
     8379 28. Sep 04:11 fixup4db.dat
     8379 28. Sep 04:11 fixup4x.dat
     3170 28. Sep 04:11 fixup_cd.dat
     7262 28. Sep 04:11 fixup.dat
    10228 28. Sep 04:11 fixup_db.dat
    10226 28. Sep 04:11 fixup_x.dat
  7599973 13. Mär 08:34 initramfs-linux.img
 24973824 11. Mär 12:45 kernel8.img
    24576 13. Mär 09:14 overlays
   803964 28. Sep 04:11 start4cd.elf
  3744808 28. Sep 04:11 start4db.elf
  2249280 28. Sep 04:11 start4.elf
  2996680 28. Sep 04:11 start4x.elf
   803964 28. Sep 04:11 start_cd.elf
  4816712 28. Sep 04:11 start_db.elf
  2973536 28. Sep 04:11 start.elf
  3720360 28. Sep 04:11 start_x.elf`

config.txt:
`# See /boot/overlays/README for all available options

dtoverlay=dwc2

initramfs initramfs-linux.img followkernel

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

# disable WIFI and Bluetooth
dtoverlay=pi3-disable-wifi
dtoverlay=pi3-disable-bt
dtoverlay=disable-wifi
dtoverlay=disable-bt

hdmi_safe=1

dtoverlay=waveshare35a
`

pelwell, 2023-03-13:
The kernel selection logic did change with this release, and something clearly got broken. It shouldn't take long to find and fix.

@hilderingt @df8oe Try with an explicit `arm_64bit=1` as a workaround (it shouldn't be necessary because that's all you have on your card).

@pelerner Your configuration is confusing - your post says `linux-rpi-6.1.16-3-aarch64`, but the kernel on the card is `kernel7.img` - the name for a 32-bit kernel. You can try with an explicit `arm_64bit=0`, and if that doesn't work try `arm_64bit=1`.

damdo, 2023-03-13:
I'm seeing this too.
Both on `Raspberry Pi Model B+ Rev 1.2` and `Raspberry Pi Model B`.
Starting since https://github.com/raspberrypi/firmware/commit/379d5bfa60bf3b0afff413a18344aaaf3bde0083

damdo, 2023-03-13:
Maybe that's unrelated to the failures users are seeing on arm64.
Could this happen because I'm using a [5.15.y kernel version](https://github.com/gokrazy-community/kernel-rpi-os-32/commit/f375b56d2ff2c9720f1e20abd3cc7a59b87521bf) together with [this firmware version](https://github.com/raspberrypi/firmware/commit/379d5bfa60bf3b0afff413a18344aaaf3bde0083)?

pelwell, 2023-03-13:
I can't immediately think of why that could be a problem, but I'll try the 5.15.84 kernel  with the new firmware when I get a moment.

pelwell, 2023-03-13:
@damdo It boots for me with that combination on a Rev 1.2 B+:
```
pi@raspberrypi:~$ uname -a && vcgencmd version && tail -1 /proc/cpuinfo
Linux raspberrypi 5.15.84+ #1610 Mon Dec 19 18:44:39 GMT 2022 armv6l GNU/Linux
Mar 10 2023 18:47:56
Copyright (c) 2012 Broadcom
version a75e3569c8471617565045aeb735d07d85df29c3 (clean) (release) (start)
Model           : Raspberry Pi Model B Plus Rev 1.2
``

df8oe, 2023-03-13:
Where do you place the workaround? In config.txt? Or as kernel parameter in cmdline.txt?

pelwell, 2023-03-13:
Yes - in config.txt.

df8oe, 2023-03-13:
That has no effect here - does not boot.

pelwell, 2023-03-13:
Be careful - if you've appended it to the file you posted above then it will be within the `[pi4]` section. Try with this addition instead:
```
[all]
arm_64bit=1
```

df8oe, 2023-03-13:
You are right - it was the wrong section. And many other lines have been misplaced, too :) Thank you - workaround is successful!

pelwell, 2023-03-13:
Thanks for persevering. We'll probably push an update out tomorrow, at which point you should be able to remove the workaround.

hilderingt, 2023-03-13:
@pelwell Thanks for your help. Works well.

pelwell, 2023-03-14:
The latest rpi-update release should not require the arm_64bit=0 workaround.

closed 2023-03-22 by pelwell