raspberrypi/firmware #1792: RPI 3B+ doesn't boot with latest firmware

nicoske, 2023-03-16:
I have several RPI 3B+ for our custom project that doesn't booth with latest firmware.

Last working firmware is : https://github.com/raspberrypi/firmware/commit/306332746ef18cb62992296dad0486396cf07c27
Tested https://github.com/raspberrypi/firmware/commit/379d5bfa60bf3b0afff413a18344aaaf3bde0083 and https://github.com/raspberrypi/firmware/commit/2c9ca22c4aedd1a8a6bc14698be5861525f2bfdf

HDMI output is blank (active with signal but no output), no LEDs (only red), 

Files in vfat partition:

```
-rwxr-xr-x 1   32162 2023-03-15 22:23:18.000000000 +0100 bcm2710-rpi-3-b.dtb
-rwxr-xr-x 1   32773 2023-03-15 22:23:18.000000000 +0100 bcm2710-rpi-3-b-plus.dtb
-rwxr-xr-x 1   52476 2023-03-15 22:23:18.000000000 +0100 bootcode.bin
-rwxr-xr-x 1     876 2023-03-15 22:23:18.000000000 +0100 boot.scr
drwxr-xr-x 2    2048 2023-03-15 22:23:18.000000000 +0100 broadcom
-rwxr-xr-x 1     461 2023-03-15 22:23:18.000000000 +0100 config.txt
-rwxr-xr-x 1    7266 2023-03-15 22:23:18.000000000 +0100 fixup.dat
drwxr-xr-x 2   24576 2023-03-15 22:23:18.000000000 +0100 overlays
-rwxr-xr-x 1 2975552 2023-03-15 22:23:18.000000000 +0100 start.elf
-rwxr-xr-x 1  410504 2023-03-15 22:23:18.000000000 +0100 u-boot.bin
-rwxr-xr-x 1 8634320 2023-03-15 22:23:18.000000000 +0100 zImage

```

config.txt

```
[all]
disable_splash=1
kernel=u-boot.bin
disable_overscan=1
dtparam=audio=on
display_auto_detect=1

[pi3]
dtoverlay=vc4-kms-v3d,cma-512

[pi4]
dtoverlay=vc4-kms-v3d,cma-512
arm_64bit=1

[cm4]
dtoverlay=dwc2,dr_mode=host
dtoverlay=gt9xx,interrupt=7,reset=8
dtoverlay=hdmi-backlight-hwhack-gpio,gpio_pin=13,active_low=1
gpio=13=op,dh

```

Note: I did my test bypassing uboot, setting kernel to zImage.

pelwell, 2023-03-16:
Using your config.txt but with `kernel=u-boot.bin` commented out it bootsfor me on a 3B+.
Do you have a serial cable you could use to debug? Or a second Pi and 3 IDC patch leads?

pelwell, 2023-03-16:
And with the working firmware, what does `uname -a` report?

pelwell, 2023-03-16:
I'm convinced that explicitly setting `arm_64bit=0` will solve your problem. A recent change to the way the kernel selection works failed to take into account the case that config.txt includes a `kernel=` setting but no `arm_64bit=`, and the firmware ended up starting the ARMs in 64-bit mode. In future, the rule will be that unless the given name matches a known 64-bit kernel name, i.e. `kernel8.img`, then it will be assumed to be a 32-bit image, even on platforms where 64-bit kernels are preferred (as is now the case on Pi 4s).

nicoske, 2023-03-16:
I'll add `arm_64bit=0` to the `[rpi3]` section of the config.txt and get back to you.
I'm building the kernel myself (and use device tree and overlays from it) and use `zImage` as filename.
I always sync the kernel version and firmware.

Thanks

nicoske, 2023-03-17:
I confirm that using `arm_64bit=0` using https://github.com/raspberrypi/firmware/commit/2c9ca22c4aedd1a8a6bc14698be5861525f2bfdf works fine.
And I also confirm that removing `arm_64bit` config works fine with https://github.com/raspberrypi/firmware/commit/a585b376a2e7e657287543d196ae1f8881ede559

Thanks a lot!

popcornmix, 2023-03-17:
Thanks for confirming.

closed 2023-03-17 by popcornmix