raspberrypi/firmware #1794: The dtoverlay for BH1750 sensor missing inputs?

Jibun-no-Kage, 2023-03-16:
Per https://www.kernel.org/doc/Documentation/ABI/testing/sysfs-bus-iio the BH1750 sensor should have several inputs including:

What:		/sys/.../iio:deviceX/in_illuminance_input
What:		/sys/.../iio:deviceX/in_illuminance_raw
What:		/sys/.../iio:deviceX/in_illuminanceY_input
What:		/sys/.../iio:deviceX/in_illuminanceY_raw
What:		/sys/.../iio:deviceX/in_illuminanceY_mean_raw
What:		/sys/.../iio:deviceX/in_illuminance_ir_raw
What:		/sys/.../iio:deviceX/in_illuminance_clear_raw
KernelVersion:	3.4
Contact:	linux-iio@vger.kernel.org
Description:
		Illuminance measurement, units after application of scale
		and offset are lux.

But when I enable the BH1750 dtoverlay on Raspberry Pi I cannot see 'in_illuminance_input' file.  I can see the scale and raw inputs.

ls -l /sys/devices/platform/soc/20804000.i2c/i2c-1/1-0023/iio:device0/
total 0
-rw-r--r-- 1 root root 4096 Mar 16 15:02 in_illuminance_integration_time
-rw-r--r-- 1 root root 4096 Mar 16 14:55 in_illuminance_raw
-rw-r--r-- 1 root root 4096 Mar 16 14:56 in_illuminance_scale
-r--r--r-- 1 root root 4096 Mar 16 15:02 integration_time_available
-r--r--r-- 1 root root 4096 Mar 16 15:02 name
lrwxrwxrwx 1 root root    0 Mar 16 15:02 of_node -> ../../../../../../../firmware/devicetree/base/soc/i2c@7e804000/bh1750@23
drwxr-xr-x 2 root root    0 Mar 16 14:55 power
lrwxrwxrwx 1 root root    0 Mar 16 15:02 subsystem -> ../../../../../../../bus/iio
-rw-r--r-- 1 root root 4096 Mar 16 14:50 uevent

Is this a limit of the dtoverlay or the Linux kernel support?  For what I am doing i can use the raw value, but am hoping otherwise if possible.

pelwell, 2023-03-16:
> the BH1750 sensor should have several inputs including

I'm not sure where you get the "should" from. The description in the linked file says:

```
Description:
		If known for a device, scale to be applied to <type>Y[_name]_raw
```
Notice the "If known for a device". The BH1750 driver doesn't claim to have such data, so the corresponding properties do not appear.

Jibun-no-Kage, 2023-03-17:
@pelwell, should, yup, could rather.  Now, wonder what the TSL2561 can provide? :)