raspberrypi/firmware #1796: 1-Wire in Parasite Power configuration (1-Wire using 2 wires) does not work (again)

iz8mbw, 2023-03-21:
Hello.
It seems 1-Wire in Parasite Power doesn't work again, reports always 85 degrees.

Some info for you:
```
root@rpi3:~# uname -a
Linux rpi3 6.1.19-v8+ #1637 SMP PREEMPT Tue Mar 14 11:11:47 GMT 2023 aarch64 GNU/Linux
```
```
root@rpi3:~# cat /etc/debian_version
11.6
```
```
root@rpi3:~# dtoverlay -l
No overlays loaded
root@rpi3:~#
```

```
root@rpi3:~# dmesg | grep -i wire
[    9.279917] Driver for 1-wire Dallas network protocol.
[    9.287128] gpio-4 (onewire@4): enforced open drain please flag it properly in DT/ACPI DSDT/board file
[    9.317419] w1_master_driver w1_bus_master1: Attaching one wire slave 3b.00000014fe14 crc 17
```

root@rpi3:~# cat /boot/config.txt | grep w1-gpio
dtoverlay=w1-gpio,gpiopin=4,pullup=on

As you can see `w1` reports 85 degrees:
```
root@rpi3:/sys/devices/w1_bus_master1/3b-00000014fe14# cat w1_slave
50 05 ff ff f0 ff ff ff 5c : crc=5c YES
50 05 ff ff f0 ff ff ff 5c t=85000
```

iz8mbw, 2023-03-21:
@popcornmix  @pelwell

dp111, 2023-03-21:
> Just made a test:
> I changed from 3V3 to 5V the power supply to pin GPIO4 and now 1-Wire
> works (always using a 4.7khom resistor).
> Well, why now 3V3 is not more enough?

I suspect the issue is the host isn't driving the bus high to charge the
capacitor at the correct time. This may have worked in the past by luck or
the timing just being slightly different.

See
https://www.analog.com/media/en/technical-documentation/data-sheets/DS18S20-PAR.pdf
Page 3.

iz8mbw, 2023-03-22:
I don't know if this can the be the motivation.
With 3.3 V I'm unable to read sensor temperature but as I wrote the 1-Wire bus can see the chip, can see the serial number and can see the chip resolution but when I interrogate it for temperature it answer with 85000 (85 °C).
I'll try to change the resistor from 4.7 kohm to 3.9 kohm and switch back from 5 V to 3.3 V.

dp111, 2023-03-22:
Did you see the bit in the link which says the temperature conversion takes
extra current and so needs a stronger pull so they recommend that the host
drives the bus high to provide the power?

pelwell, 2023-03-22:
That's what the w1_gpio driver attempts to do, by forcing the pin to be an output driven high for a while.

iz8mbw, 2023-03-22:
If you need test on new kernel/firmware I can do it.