raspberrypi/firmware #1797: New Kernel let crashes the desktop gui

thxstr, 2023-03-22:
Before days the system was uptodate and everything works fine (in particular: Dietpi, Mate-Desktop on my Raspberrypi). Now I updated the Kernelversion (the last update that the system got) and the GUI is crashing from time to time randomly. So the Display is working for seconds/minutes and then the Desktop seems to crash (why ever). 
If I go to the oldstable kernel (with "sudo rpi-update oldstablesudo rpi-update oldstable") everything is fine again (like before days), but an rpi-update will bring the same effect again to the desktop. Could it be that there is a problem in the current kernel-version?

To be exact I did a fresh installation today again, with the same result. The .xsession-errors shows:
`Xsession: X session started for dietpi at Di 21. Mär 10:52:34 CET 2023
WARNING: tempfile is deprecated; consider using mktemp instead.
dbus-update-activation-environment: setting DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus
dbus-update-activation-environment: setting DISPLAY=:0
dbus-update-activation-environment: setting XAUTHORITY=/home/dietpi/.Xauthority
localuser:dietpi being added to access control list
dbus-update-activation-environment: setting QT_ACCESSIBILITY=1
dbus-update-activation-environment: setting USER=dietpi
dbus-update-activation-environment: setting XDG_SESSION_TYPE=x11
dbus-update-activation-environment: setting HOME=/home/dietpi
dbus-update-activation-environment: setting DESKTOP_SESSION=lightdm-xsession
dbus-update-activation-environment: setting XDG_SEAT_PATH=/org/freedesktop/DisplayManager/Seat0
dbus-update-activation-environment: setting DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus
dbus-update-activation-environment: setting LOGNAME=dietpi
dbus-update-activation-environment: setting XDG_SESSION_CLASS=user
dbus-update-activation-environment: setting PATH=/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games
dbus-update-activation-environment: setting XDG_SESSION_PATH=/org/freedesktop/DisplayManager/Session0
dbus-update-activation-environment: setting XDG_RUNTIME_DIR=/run/user/1000
dbus-update-activation-environment: setting DISPLAY=:0
dbus-update-activation-environment: setting LANG=de_DE.UTF-8
dbus-update-activation-environment: setting XDG_SESSION_DESKTOP=lightdm-xsession
dbus-update-activation-environment: setting XAUTHORITY=/home/dietpi/.Xauthority
dbus-update-activation-environment: setting XDG_GREETER_DATA_DIR=/var/lib/lightdm/data/dietpi
dbus-update-activation-environment: setting SHELL=/bin/bash
dbus-update-activation-environment: setting GDMSESSION=lightdm-xsession
dbus-update-activation-environment: setting QT_ACCESSIBILITY=1
dbus-update-activation-environment: setting GPG_AGENT_INFO=/run/user/1000/gnupg/S.gpg-agent:0:1
dbus-update-activation-environment: setting PWD=/home/dietpi
dbus-update-activation-environment: setting XDG_DATA_DIRS=/usr/share/mate:/usr/local/share/:/usr/share/
Failed to spawn gnome-keyring-daemon: Kindprozess »gnome-keyring-daemon« konnte nicht ausgeführt werden (Datei oder Verzeichnis nicht gefunden)
x-session-manager[641]: WARNING: Could not parse desktop file /home/dietpi/.config/autostart/dietpi-desktop_setup.desktop: Die Schlüsselwertedatei hat keinen Schlüssel »Name« in der Gruppe »Desktop Entry«
x-session-manager[641]: GLib-GObject-CRITICAL: object GsmAutostartApp 0x55ad46d9a0 finalized while still in-construction
x-session-manager[641]: GLib-GObject-CRITICAL: Custom constructor for class GsmAutostartApp returned NULL (which is invalid). Please use GInitable instead.
x-session-manager[641]: WARNING: could not read /home/dietpi/.config/autostart/dietpi-desktop_setup.desktop
x-session-manager[641]: WARNING: Unable to find provider '' of required component 'dock'
Warnung der Fensterverwaltung:Log level 128: unsetenv() is not thread-safe and should not be used after threads are created

(caja:819): Gtk-WARNING **: 10:52:42.007: Failed to register client: GDBus.Error:org.gnome.SessionManager.AlreadyRegistered: Unable to register client

(nm-applet:839): Gdk-CRITICAL **: 10:52:46.872: gdk_window_thaw_toplevel_updates: assertion 'window->update_and_descendants_freeze_count > 0' failed

(mate-panel:733): Gtk-WARNING **: 10:52:55.535: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.

(caja:819): Gtk-WARNING **: 11:02:46.204: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.

(caja:819): Gtk-WARNING **: 11:03:03.046: GtkLabel 0x558989df30 adjusted size horizontal min 72 natural 72 must not decrease below min 73 natural 73

(caja:819): Gtk-WARNING **: 11:03:03.051: GtkLabel 0x558989df30 attempted to adjust its size allocation from 208,8 73x24 to 207,8 74x24. adjust_size_allocation must keep allocation inside original bounds

(caja:819): Gtk-WARNING **: 11:03:03.339: GtkLabel 0x558989df30 adjusted size horizontal min 72 natural 72 must not decrease below min 73 natural 73

(caja:819): Gtk-WARNING **: 11:03:03.341: GtkLabel 0x558989df30 attempted to adjust its size allocation from 208,8 73x24 to 207,8 74x24. adjust_size_allocation must keep allocation inside original bounds

(caja:819): Gtk-WARNING **: 11:03:08.057: GtkLabel 0x558989df30 adjusted size horizontal min 72 natural 72 must not decrease below min 73 natural 73

(caja:819): Gtk-WARNING **: 11:03:08.057: GtkLabel 0x558989df30 attempted to adjust its size allocation from 208,8 73x24 to 207,8 74x24. adjust_size_allocation must keep allocation inside original bounds

(caja:819): Gtk-WARNING **: 11:03:08.075: GtkLabel 0x558989df30 adjusted size horizontal min 72 natural 72 must not decrease below min 73 natural 73

(caja:819): Gtk-WARNING **: 11:03:08.076: GtkLabel 0x558989df30 attempted to adjust its size allocation from 208,8 73x24 to 207,8 74x24. adjust_size_allocation must keep allocation inside original bounds

(caja:819): Gtk-WARNING **: 11:03:08.094: GtkLabel 0x558989df30 adjusted size horizontal min 72 natural 72 must not decrease below min 73 natural 73

(caja:819): Gtk-WARNING **: 11:03:08.095: GtkLabel 0x558989df30 attempted to adjust its size allocation from 208,8 73x24 to 207,8 74x24. adjust_size_allocation must keep allocation inside original bounds

(caja:819): Gtk-WARNING **: 11:03:08.114: GtkLabel 0x558989df30 adjusted size horizontal min 72 natural 72 must not decrease below min 73 natural 73

(caja:819): Gtk-WARNING **: 11:03:08.115: GtkLabel 0x558989df30 attempted to adjust its size allocation from 208,8 73x24 to 207,8 74x24. adjust_size_allocation must keep allocation inside original bounds

(caja:819): Gtk-WARNING **: 11:03:08.188: GtkLabel 0x558989df30 adjusted size horizontal min 72 natural 72 must not decrease below min 73 natural 73

(caja:819): Gtk-WARNING **: 11:03:08.189: GtkLabel 0x558989df30 attempted to adjust its size allocation from 208,8 73x24 to 207,8 74x24. adjust_size_allocation must keep allocation inside original bounds

(caja:819): Gtk-WARNING **: 11:03:08.206: GtkLabel 0x558989df30 adjusted size horizontal min 72 natural 72 must not decrease below min 73 natural 73

(caja:819): Gtk-WARNING **: 11:03:08.206: GtkLabel 0x558989df30 attempted to adjust its size allocation from 208,8 73x24 to 207,8 74x24. adjust_size_allocation must keep allocation inside original bounds

(caja:819): Gtk-WARNING **: 11:03:08.223: GtkLabel 0x558989df30 adjusted size horizontal min 72 natural 72 must not decrease below min 73 natural 73

(caja:819): Gtk-WARNING **: 11:03:08.224: GtkLabel 0x558989df30 attempted to adjust its size allocation from 208,8 73x24 to 207,8 74x24. adjust_size_allocation must keep allocation inside original bounds

(caja:819): Gtk-WARNING **: 11:03:08.241: GtkLabel 0x558989df30 adjusted size horizontal min 72 natural 72 must not decrease below min 73 natural 73

(caja:819): Gtk-WARNING **: 11:03:08.242: GtkLabel 0x558989df30 attempted to adjust its size allocation from 208,8 73x24 to 207,8 74x24. adjust_size_allocation must keep allocation inside original bounds

(caja:819): Gtk-WARNING **: 11:03:08.258: GtkLabel 0x558989df30 adjusted size horizontal min 72 natural 72 must not decrease below min 73 natural 73

(caja:819): Gtk-WARNING **: 11:03:08.259: GtkLabel 0x558989df30 attempted to adjust its size allocation from 208,8 73x24 to 207,8 74x24. adjust_size_allocation must keep allocation inside original bounds

(caja:819): Gtk-WARNING **: 12:00:20.081: GtkLabel 0x558a53cf30 adjusted size horizontal min 72 natural 72 must not decrease below min 73 natural 73

(caja:819): Gtk-WARNING **: 12:00:20.088: GtkLabel 0x558a53cf30 attempted to adjust its size allocation from 208,8 73x24 to 207,8 74x24. adjust_size_allocation must keep allocation inside original bounds`

Pi3B+ (aarch64)
Dietpi 8.15
bullseye
Kernel 6.1.19-v8+

Do you need more information? Do you know how to jump back to a particular kernel-version (that I can keep the 6er Version and don't have to jump back to 5.10.103)?

kai-row, 2023-03-30:
Hi,

I had the same problem.
Eventually I found the following in /var/log/kern.log
Mar 30 21:17:35  kernel: [   42.205542] vc4-drm gpu: [drm] *ERROR* [CRTC:96:pixelvalve-2] commit wait timed out
Mar 30 22:16:48  kernel: [   52.445682] vc4-drm gpu: [drm] *ERROR* flip_done timed out
Mar 30 22:16:48  kernel: [   52.445704] vc4-drm gpu: [drm] *ERROR* [CONNECTOR:32:HDMI-A-1] commit wait timed out
Mar 30 22:16:59  kernel: [   62.685833] vc4-drm gpu: [drm] *ERROR* flip_done timed out
Mar 30 22:16:59  kernel: [   62.685861] vc4-drm gpu: [drm] *ERROR* [PLANE:86:plane-3] commit wait timed out
Mar 30 22:17:09  kernel: [   72.925960] vc4-drm gpu: [drm] *ERROR* flip_done timed out

I had in /boot/config.txt the following setting: dtoverlay=vc4-kms-v3d
Somehow that caused the Xserver to crash.
Disabling it solved the problem.
Maybe that helps you as well.

kai-row, 2023-04-21:
Hi,

just another update on this topic. What I wrote above was fixing that the GUI was not starting anymore but also had the downside that the kernel module was not used anymore and video output gets sluggish.

I found that for the PI4 the best settings in /boot/config.txt are the following:

[hdmi:0]
disable_splash=0
config_hdmi_boost=4
enable_uart=1
display_rotate=0
start_x=1
gpu_mem=128
hdmi_force_hotplug=1
max_usb_current=1
hdmi_group=1
hdmi_mode=16
hdmi_drive=2
hdmi_blanking=0
disable_overscan=1
dtoverlay=vc4-fkms-v3d-pi4
max_framebuffers=2
[hdmi:1]
disable_splash=0
config_hdmi_boost=4
enable_uart=1
display_rotate=0
start_x=1
gpu_mem=128
hdmi_force_hotplug=1
max_usb_current=1
hdmi_group=1
#hdmi_mode=68
hdmi_mode=16
hdmi_drive=2
hdmi_blanking=0
disable_overscan=1
dtoverlay=vc4-fkms-v3d-pi4
max_framebuffers=2

[all]
arm_64bit=1
dtparam=audio=on
smsc95xx.macaddr=DC:A6:32:C4:B7:CB
hmdi_force_edid_audio=1

The above works fine with my setup.
I have a full HD screen connected to it. The GUI works comes up properly and video via VLC also works perfect.
In vlc I had to make sure that the video output is set to "XVideo output (XCB)".