raspberrypi/firmware #1798: SenseHat LED matrix not responding

DMHas, 2023-03-24:
Hello all,

I have the problem that on both of my Raspberry Pi (2 &4) the Sense Hat LED matrix is no longer responsive. After powering on the Pi, the LED matrix lights up in rainbow colors. But then the LED matrix does not respond to any commands - that is, it does not display anything when you want to display text on the LED matrix. No error message is displayed.
At first I thought it was a hardware defect. For testing I connected the Sense Hat to my old Raspberry Pi 2 - it worked as usual. Then I updated the Raspberry Pi 2 (sudo apt update && suso apt upgrade). After that, the error also occurred on the Raspberry Pi 2 - as described above. It must be a software / driver problem. The error occurs with a Sense Hat V1 like Sense Hat V2.

Have any of you experienced the same phenomenon or have an idea how I can narrow down or fix the error?
(Temperature, humidity and pressure sensor are working normally).

Greetings DMHas

see alos here: https://github.com/astro-pi/python-sense-hat/issues/138

DMHas, 2023-03-24:
@pelwell: Thank you very much for you fast response! I will update the kernel as soon as possible!

popcornmix, 2023-04-06:
The fixed kernel is now in apt. okay to close?

DMHas, 2023-04-06:
Yes - all fine now. Please close it.