raspberrypi/firmware #1799: Samsung T7 SSD fails USB boot unless is is plugged in while SD card boot is attempted

Brunnis, 2023-03-25:
**Describe the bug**
My Raspberry Pi 4 does not boot from a newly configured (via network install) Samsung Portable SSD T7 with Raspberry Pi OS 64-bit. Each attempt stops with "USB MSD stopped. Timeout: 25 seconds" on the boot screen. Pulling the SSD cable out and waiting until it performs an SD card boot attempt and reconnecting the SSD while the SD card attempt is ongoing seems to reliably trigger the USB MSD boot.

**To reproduce**

1. Power up Raspberry Pi 4 with an empty Samsung Portable SSD T7 connected to one of the USB3 ports and no SD card installed.
2. At the boot screen, choose the network installation. Install Raspberry Pi OS 64-bit.
3. When the installation is finished, the system ends up at the white/red screen looping between attempting USB MSD boot and SD card boot. This seems to go on forever.
4. Unplug the SSD.
5. Wait until an SD card boot attempt is being made. Plug the SSD back in while the SD card boot attempt is still ongoing.
6. Raspberry Pi OS is booted on the next USB MSD boot attempt.

**Expected behaviour**
The system should boot from the USB SSD without having to be unplugged and then reconnected.

**Actual behaviour**
The system ends up at the white/red screen looping between attempting USB MSD boot and SD card boot. This seems to go on forever.

**System**
Output of raspinfo here: https://pastebin.com/r33ZiuVR

rpi-eeprom-config outputs:
```[all]
BOOT_UART=0
WAKE_ON_GPIO=1
ENABLE_SELF_UPDATE=1
BOOT_ORDER=0xf41
```

danmrossi, 2023-05-10:
Just come across this, just a fellow enthusiast here and for anyone else that comes across this?

Did you try updating the bootloader/EEPROM? As i have found that alot of people don't necessarily do it and theres conflicting info out there as to wether or not this auto updates itself (never has for me on any of my pi's with any version of raspbian)..

On a SD Card - Install the latest raspbian bullseye, run sudo apt update and sudo apt full-upgrade to get the kernel etc all to latest and reboot by sudo reboot

Log back into your pi and then run sudo rpi-eeprom-update -d -a 

This will update the bootloader/EEPROM to the latest version.

Then sudo reboot.

If still not work try setting boot order with: https://linuxhint.com/change-boot-option-raspberry-pi/

wulfy23, 2023-06-04:
```
USB_MSD_BOOT_MAX_RETRIES
```
???

marcelgsteiger, 2023-08-08:
The specs of the Samsung T7 show a 1.5A requirement on +5V, which would be too high for the raspberry pi (max. 1.2 A total for all connected USB devices). The T5 spec was lower in this respect. However, normally both SSDs draw well below 1A in operation.
Use
`sudo vcgencmd get_throttled
`
this should return throttled=0x0. If it returns 0x50000 instead, the supply voltage has dropped too much at least once since last boot. I had to adjust the power supply voltage to 5.3V to achieve this - standard 'official' Pi supply delivers 5.1V. I'm powering the pi using a 24V industrial power supply feeding Joy-It StromPi UPS (modified for 5.3V output voltage) connected to the GPIO header. Works now since 2 years without interruption, with an intensive postgresql DB app running 24*7 (Zabbix).
A nice feature of the T5 I'm using (compared to other brands) is that it supports SMART through the UAS interface, so I can predict when it will fail. I did not yet verify this with the T7, though.
HTH
Marcel

timg236, 2024-01-08:
Closed due to inactivity.

closed 2024-01-08 by timg236