raspberrypi/firmware #1802: pi4(4GB) work in headless  and disable_fw_kms_setup=1,but Xorg.0.log still show  (II) modeset(0): Output HDMI-1 connected

shixudong2020, 2023-04-02:
I have a pi4 (4GB) running the latest 2023-02-21-raspios-bullseye-arm64 with a recent kernel upgrade to 6.1.19-v8+
uname -a
Linux raspberrypi4 6.1.19-v8+ #1637 SMP PREEMPT Tue Mar 14 11:11:47 GMT 2023 aarch64 GNU/Linux

shixudong2020, 2023-04-08:
headless generating Page flip failed log have been solved by upgrade to 6.1.21-v8+ ,but Xorg.0.log still show (II) modeset(0): Output HDMI-1 connected(disable_fw_kms_setup=1&headless)

popcornmix, 2023-04-08:
`disable_fw_kms_setup=1` prevents the addition of the `video=` setting in kernel command line.
The `video=` setting on only used when booting to console. The desktop behaviour isn't determined by that.

Yes, `hdmi_force_hotplug=1` adds the final `D` to the `video=` setting.

There will be no difference in behaviour between a 4G and 8G Pi4 in this regard.
If you are observing  a difference then make sure you are only changing the pi between the tests
(so sdcard, hdmi cable, monitor etc should all be identical).

shixudong2020, 2023-04-09:
two pi4,one is 4G,another is 8G,all use headless,so no need for hdmi cable and monitor,the same sdcard.
all use disable_fw_kms_setup=0 (#disable_fw_kms_setup=1 )
if hdmi_force_hotplug=1,Both perform exactly the same(video=HDMI-A-1:1920x1080M@60D)
if hdmi_force_hotplug=0,4G show video=HDMI-A-1:1920x1080M@60,but 8G have no video= setting
why?