raspberrypi/firmware #1804: rpi3 kernel: [61372.274158] WARN::dwc_otg_hcd_urb_dequeue:639: Timed out waiting for FSM NP transfer to complete on

iz8mbw, 2023-04-03:
Hello.
Using a Raspberry Pi 3 Model B+ with latest Raspberry Pi OS (64-bit):

`Linux rpi3 6.1.19-v8+ #1637 SMP PREEMPT Tue Mar 14 11:11:47 GMT 2023 aarch64 GNU/Linux`
`root@rpi3:~# cat /etc/debian_version
11.6`

On one USB port I have a USB to RS485 adapter (using PL2303 chip):
`
Apr  2 15:16:43 rpi3 kernel: [   12.199081] usbcore: registered new interface driver usbserial_generic
Apr  2 15:16:43 rpi3 kernel: [   12.200645] usbserial: USB Serial support registered for generic
Apr  2 15:16:43 rpi3 kernel: [   12.260137] usbcore: registered new interface driver pl2303
Apr  2 15:16:43 rpi3 kernel: [   12.273678] usbserial: USB Serial support registered for pl2303
Apr  2 15:16:43 rpi3 kernel: [   12.282038] pl2303 1-1.1.3:1.0: pl2303 converter detected
Apr  2 15:16:43 rpi3 kernel: [   12.317290] usb 1-1.1.3: pl2303 converter now attached to ttyUSB0
`

`
root@rpi3:~# lsusb -v

Bus 001 Device 004: ID 0930:6544 Toshiba Corp. TransMemory-Mini / Kingston DataTraveler 2.0 Stick
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x0930 Toshiba Corp.
  idProduct          0x6544 TransMemory-Mini / Kingston DataTraveler 2.0 Stick
  bcdDevice            1.00
  iManufacturer           1 Kingston
  iProduct                2 DataTraveler 2.0
  iSerial                 3 0019B92B28C9C8C135450044
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval             255
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
[... 763 lines cut ...]
Apr  3 08:37:06 rpi3 kernel: [60770.060092]
Apr  3 08:37:06 rpi3 kernel: [60770.060108] WARN::dwc_otg_hcd_urb_dequeue:639: Timed out waiting for FSM NP transfer to complete on 6
Apr  3 08:37:06 rpi3 kernel: [60770.060528]
Apr  3 08:37:06 rpi3 kernel: [60770.060531] WARN::dwc_otg_hcd_urb_dequeue:639: Timed out waiting for FSM NP transfer to complete on 1
Apr  3 08:38:25 rpi3 kernel: [60848.748986]
Apr  3 08:38:25 rpi3 kernel: [60848.749003] WARN::dwc_otg_hcd_urb_dequeue:639: Timed out waiting for FSM NP transfer to complete on 2
Apr  3 08:39:03 rpi3 kernel: [60887.181225]
Apr  3 08:39:03 rpi3 kernel: [60887.181240] WARN::dwc_otg_hcd_urb_dequeue:639: Timed out waiting for FSM NP transfer to complete on 3
Apr  3 08:39:39 rpi3 kernel: [60923.277746]
Apr  3 08:39:39 rpi3 kernel: [60923.277763] WARN::dwc_otg_hcd_urb_dequeue:639: Timed out waiting for FSM NP transfer to complete on 3
Apr  3 08:39:54 rpi3 kernel: [60937.773856]
Apr  3 08:39:54 rpi3 kernel: [60937.773873] WARN::dwc_otg_hcd_urb_dequeue:639: Timed out waiting for FSM NP transfer to complete on 5
Apr  3 08:39:54 rpi3 kernel: [60937.774306]
Apr  3 08:39:54 rpi3 kernel: [60937.774309] WARN::dwc_otg_hcd_urb_dequeue:639: Timed out waiting for FSM NP transfer to complete on 3
Apr  3 08:40:09 rpi3 kernel: [60952.845847]
Apr  3 08:40:09 rpi3 kernel: [60952.845863] WARN::dwc_otg_hcd_urb_dequeue:639: Timed out waiting for FSM NP transfer to complete on 5
Apr  3 08:40:09 rpi3 kernel: [60952.846280]
Apr  3 08:40:09 rpi3 kernel: [60952.846284] WARN::dwc_otg_hcd_urb_dequeue:639: Timed out waiting for FSM NP transfer to complete on 7
Apr  3 08:42:39 rpi3 kernel: [61103.375435]
Apr  3 08:42:39 rpi3 kernel: [61103.375459] WARN::dwc_otg_hcd_urb_dequeue:639: Timed out waiting for FSM NP transfer to complete on 2
```

I have the same warning on `messages` also if I try to use another USB-RS485 adapter using the ch341 chip.

Thanks for your support.

usaf-barnes, 2023-06-08:
I am also seeing this a lot on my installation.

**********************************************
RASPBERRY PI VERSION VERSION
**********************************************
### Raspberry Pi 3B+
RETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

Client: Docker Engine - Community
 Version:           24.0.2
 API version:       1.43
 Go version:        go1.20.4
 Git commit:        cb74dfc
 Built:             Thu May 25 21:50:59 2023
 OS/Arch:           linux/arm64
 Context:           default

**********************************************
DOCKER VERSION
**********************************************
Server: Docker Engine - Community
 Engine:
  Version:          24.0.2
  API version:      1.43 (minimum version 1.12)
  Go version:       go1.20.4
  Git commit:       659604f
  Built:            Thu May 25 21:50:59 2023
  OS/Arch:          linux/arm64
  Experimental:     false
 containerd:
  Version:          1.6.21
  GitCommit:        3dce8eb055cbb6872793272b4f20ed16117344f8
 runc:
  Version:          1.1.7
  GitCommit:        v1.1.7-0-g860f061
 docker-init:
  Version:          0.19.0
  GitCommit:        de40ad0

iz8mbw, 2023-06-09:
Hello.
For now I have changed my setup so instead to use https://github.com/gianfrdp/SDM120C software to poll my SDM120 meter, I'm using [mbusd](https://github.com/3cky/mbusd) and "call" it via `mbpoll` command.
So the software that uses the serial porti is `mbusd` and in this way I don't have any more `Timed out waiting for FSM NP transfer to complete` on the `/var/log/messages`.

vrobert78, 2023-11-09:
I had the same issue with the 64bits edition of Raspberry Pi OS. I reinstalled the 32bits version, and the `WARN::dwc_otg_hcd_urb_dequeue:639: Timed out waiting for FSM NP transfer` errors are completely gone.

At some points before reinstalling, the Raspberry Pi was even losing the network after a few hours.
Everything is solved by moving back to the 32bits edition.

My model is: Raspberry Pi 3 Model B Plus Rev 1.3

pelwell, 2023-11-09:
The dwc_otg's driver makes heavy use of FIQs to improve performance. Until fairly recently the arm64 support didn't recognise FIQs; it looks like they are now supported to some extent, but the dwc_otg driver hasn't been updated to use it - you can bet that won't be a simple task.

For now, the advice on arm64 is the same as it's always been - use "dtoverlay=dwc2" to select the upstream DWC2 driver.

sirjeannot, 2024-01-07:
thank you for the pointer. however it appears to not be possible to boot from a usb drive using the dwc2 driver.
I'll revert to armv7l as well.

mthmulders, 2024-01-23:
Same, 64-bit Debian Bookworm on a Raspberry Pi 3+. I'm also using a USB-to-RJ45 cable to read my smart energy meter.

popcornmix, 2024-01-23:
> also disabling fiq in config.txt isn't taken into account by the driver, if it is even still possible.
dwc_otg.fiq_enable=0
dwc_otg.fiq_fsm_enable=0

These setting go in cmdline.txt (on existing line), not config.txt.

defanator, 2025-03-01:
I found this issue by the same key message. I've been experiencing issues with running zigbee2mqtt on Raspberry Pi 3 with recent Debian 12 "bookworm" on kernel 6.6.74+rpt-rpi-v8 and SONOFF Zigbee 3.0 USB Dongle Plus V2 adapter: almost immediately after starting zigbee2mqtt system completely hangs (even watchdog doesn't help - I have to use remotely controlled smart socket to restart the device), and the following appears in the log:

```
Mar 01 10:24:21 rpi3-mon z2m[1508]: Starting Zigbee2MQTT without watchdog.
Mar 01 10:24:37 rpi3-mon kernel: ------------[ cut here ]------------
Mar 01 10:24:37 rpi3-mon kernel: Firmware transaction timeout
Mar 01 10:24:37 rpi3-mon kernel: WARNING: CPU: 2 PID: 1048 at drivers/firmware/raspberrypi.c:69 rpi_firmware_property_list+0x264/0x288
Mar 01 10:24:37 rpi3-mon kernel: Modules linked in: xt_addrtype xt_nat xt_tcpudp xt_mark xt_conntrack xt_comment nft_chain_nat xt_MASQUERADE nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nft_compat nf_tables veth bridge stp llc nfnetlink overlay vc4 brcmfmac_wcc snd_soc_hdmi_codec drm_display_helper cec drm_dma_helper binfmt_misc drm_kms_helper brcmfmac snd_soc_core snd_compress brcmutil snd_pcm_dmaengine hci_uart cfg80211 btbcm bcm2835_codec(C) bluetooth bcm2835_isp(C) bcm2835_v4l2(C) v4l2_mem2mem cdc_acm bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc raspberrypi_hwmon videobuf2_memops videobuf2_v4l2 ecdh_generic ecc videodev rfkill libaes snd_bcm2835(C) videobuf2_common mc snd_pcm snd_timer snd vc_sm_cma(C) raspberrypi_gpiomem uio_pdrv_genirq uio drm fuse drm_panel_orientation_quirks dm_mod backlight ip_tables x_tables ipv6 i2c_bcm2835
Mar 01 10:24:37 rpi3-mon kernel: CPU: 2 PID: 1048 Comm: kworker/2:1 Tainted: G         C         6.6.74+rpt-rpi-v8 #1  Debian 1:6.6.74-1+rpt1
Mar 01 10:24:37 rpi3-mon kernel: Hardware name: Raspberry Pi 3 Model B Plus Rev 1.3 (DT)
Mar 01 10:24:37 rpi3-mon kernel: Workqueue: events get_values_poll [raspberrypi_hwmon]
Mar 01 10:24:37 rpi3-mon kernel: pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Mar 01 10:24:37 rpi3-mon kernel: pc : rpi_firmware_property_list+0x264/0x288
Mar 01 10:24:37 rpi3-mon kernel: lr : rpi_firmware_property_list+0x264/0x288
Mar 01 10:24:37 rpi3-mon kernel: sp : ffffffc08122bca0
Mar 01 10:24:37 rpi3-mon kernel: x29: ffffffc08122bca0 x28: 0000000000000000 x27: ffffff8001fc1c40
Mar 01 10:24:37 rpi3-mon kernel: x26: 0000000000000004 x25: ffffffc080498008 x24: 0000000000001000
Mar 01 10:24:37 rpi3-mon kernel: x23: ffffff800979c5c0 x22: ffffffd2fe7ec2e0 x21: 0000000000000010
Mar 01 10:24:37 rpi3-mon kernel: x20: ffffff8001fc1c00 x19: ffffffc080498000 x18: 0000000000000000
Mar 01 10:24:37 rpi3-mon kernel: x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000000000
Mar 01 10:24:37 rpi3-mon kernel: x14: 0000000000000000 x13: 74756f656d697420 x12: 6e6f69746361736e
Mar 01 10:24:37 rpi3-mon kernel: x11: fffffffffffe0000 x10: ffffffd2fe6c64b0 x9 : ffffffd2fd10bdd4
Mar 01 10:24:37 rpi3-mon kernel: x8 : 00000000ffffefff x7 : ffffffd2fe6c3a38 x6 : 00000000000001c5
Mar 01 10:24:37 rpi3-mon kernel: x5 : ffffff803b1a5d88 x4 : 0000000000000000 x3 : 0000000000000027
Mar 01 10:24:37 rpi3-mon kernel: x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffffff8003941ec0
Mar 01 10:24:37 rpi3-mon kernel: Call trace:
Mar 01 10:24:37 rpi3-mon kernel:  rpi_firmware_property_list+0x264/0x288
Mar 01 10:24:37 rpi3-mon kernel:  rpi_firmware_property+0x78/0x120
Mar 01 10:24:37 rpi3-mon kernel:  get_values_poll+0x54/0x138 [rasp
```

(actually, I wasn't able to catch these for a long time as I don't have direct access to the device at the moment, and it turned out that it hangs so quickly that the call trace did not have a chance to reach the log monitored with `journalctl -f`)

I went through the `rpi-update` today to bring 6.12 kernel to the device, and it worked fine:
```
# uname -a
Linux rpi3-mon 6.12.16-v8+ #1859 SMP PREEMPT Mon Feb 24 13:14:16 GMT 2025 aarch64 GNU/Linux

# vcgencmd version
Feb 17 2025 20:04:08 
Copyright (c) 2012 Broadcom
version da7125f985dfe3cb0cf91e3c47f1ace88018e466 (clean) (release) (start)
```

However, now I'm getting this message after z2m starts:
```
Mar 01 11:02:40 rpi3-mon z2m[1583]: Starting Zigbee2MQTT without watchdog.
Mar 01 11:02:55 rpi3-mon kernel: 
Mar 01 11:02:55 rpi3-mon kernel: WARN::dwc_otg_hcd_urb_dequeue:639: Timed out waiting for FSM NP transfer to complete on 6
Mar 01 11:02:55 rpi3-mon kernel: ------------[ cut here ]------------
Mar 01 11:02:55 rpi3-mon kernel: Firmware transaction 0x00030046 timeout
```

I'm still trying to debug it further (i.e. could it be related to z2m version or Zigbee stick firmware - however I didn't update either for a long time).

Appreciate any advice.

P33M, 2025-03-05:
What's the firmware version that breaks dwc_otg? I'd be very interested to see what the fix was (and where the FSM hole is).

defanator, 2025-03-05:
@P33M previous firmware was https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.3.0-zbdonglee-115200.gbl, but it was working without issues since ~July 2024. I am still not sure the issue was caused by firmware (https://github.com/Koenkk/zigbee2mqtt/issues/26541#issuecomment-2692546523).

This is a remote setup, so hard to debug, but I could try to re-flash 7.4.3 and try to reproduce the failure.

amanteaux, 2025-04-17:
@defanator I have the same issue with the exact same setup (rpi3, SONOFF Zigbee 3.0 USB Dongle Plus), but even with an updated Sonoff firmware & raspbian, I encounter the same freeze issue. However, as you say, the freeze happens so quickly that I cannot catch any kernel logs during/after the crash (even with direct access). How have you been able to catch these logs?

defanator, 2025-04-17:
@amanteaux it was a pure luck - I've been able to caught those just once while debugging remotely (i.e. the only way to reach to the raspberry was using ssh + running e.g. `journalctl -f`). However, a few weeks ago I finally got to the location and spent some time trying to figure out what's happening - with keyboard and external display being connected. I was able to see the whole backtrace from kernel panic during one of these hiccups - will share it a bit later.

And yes - I've been running through the same issue with Sonoff dongle being updated to 7.4.5 and using 2 different RPis (both 3 model B), so for now I tend to think this is a software issue.

I'll update this comment with the backtrace soon.

amanteaux, 2025-04-17:
@defanator Thank you for your reply! I am glad you can get these logs because on my side, either remotely or with a keyboard and an external display (using `journalctl -f` or `dmesg -w`) I don't see anything :'(

pelwell, 2025-04-17:
Have you tried with `dtoverlay=dwc2` in config.txt? The DWC2 driver is lower performance, but it is more widely used and you may find that it is sufficient for your needs.

defanator, 2025-04-17:
@pelwell I did not. The only thing I've tried is this - https://github.com/raspberrypi/firmware/issues/1804#issuecomment-1906165603. Made no difference at all.

pelwell, 2025-04-17:
Consider it a suggestion.

amanteaux, 2025-04-18:
Uh with `dtoverlay=dwc2`, I confirm that everything works fine. So the issue is with the `dwc_otg` default/faster driver?

P33M, 2025-04-22:
> Uh with `dtoverlay=dwc2`, I confirm that everything works fine. So the issue is with the `dwc_otg` default/faster driver?

The dwc_otg driver only enables the FIQ on an armv7 kernel, and uses a regular interrupt on arm64, which makes it no better than dwc2. Except you get bonus failure cases because the FIQ makes timing assumptions based on not being delayed more than a trivial amount of time (hence fast-interrupt).

We should probably get the Videocore firmware to default to dwc2 in this case, unless someone has explicitly requested dwc_otg...

venglin, 2025-05-11:
I got the same exact issue on an RPi 3B+ with a deCONZ Conbee II dongle. After starting Zigbee2MQTT or the firmware flasher, the system freezes immediately (with no kernel panic on the console). Setting `dtoverlay=dwc2` fixes the issue.