raspberrypi/firmware #1805: cannot overwrite non-directory '//opt' with directory '//root/.rpi-firmware/vc/softfp/opt'

Ashcora, 2023-04-10:
When trying to run the rpi-update, I get the following error message displayed followed by cancelling the script:
`cannot overwrite non-directory '//opt' with directory '//root/.rpi-firmware/vc/softfp/opt'`

The good news is that I know where it is coming from:
=> I created a symbolic link for the "opt" directory to an external drive (in order to protect my sd-card).

Question: Are the only two options to either change the rpi-update (which I want to avoid) or not to use symbolic links for opt? Is there any way to provide the directory with parameters or similar?

Just if needed, below the complete output of the script:

pi@raspberrypi4:~ $ sudo rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
FW_REV:0b15297bf42d0dcbb463d702c739b281c272f0c1
 *** We're running for the first time
 *** Backing up files (this will take a few minutes)
 *** Remove old firmware backup
 *** Backing up firmware
 *** Remove old modules backup
 *** Backing up modules 6.1.21-v8+
WANT_32BIT:1 WANT_64BIT:1 WANT_PI4:1
#############################################################
WARNING: This update bumps to rpi-6.1.y linux tree
See: https://forums.raspberrypi.com/viewtopic.php?t=344246

'rpi-update' should only be used if there is a specific
reason to do so - for example, a request by a Raspberry Pi
engineer or if you want to help the testing effort
and are comfortable with restoring if there are regressions.

DO NOT use 'rpi-update' as part of a regular update process.
##############################################################
Would you like to proceed? (y/N)
 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  116M  100  116M    0     0  20.4M      0  0:00:05  0:00:05 --:--:-- 22.3M
 *** Updating firmware
 *** Updating kernel modules
 *** depmod 6.1.21+
 *** depmod 6.1.21-v8+
 *** depmod 6.1.21-v7+
 *** depmod 6.1.21-v7l+
 *** Updating VideoCore libraries
 *** Using SoftFP libraries
cp: cannot overwrite non-directory '//opt' with directory '//root/.rpi-firmware/vc/softfp/opt'

Ashcora, 2025-03-06:
Closed due to lack of interaction on this issue report.

closed 2025-03-06 by Ashcora

popcornmix, 2025-03-06:
This would have been better reported at https://github.com/raspberrypi/rpi-update/issues

Although I suspect this is no longer an issue as firmware repo no longer contains /opt (which contains deprecated tools/libs).

Ashcora, 2025-03-07:
Sometimes issues resolve themselves by giving them some time 😄