raspberrypi/firmware #1807: CM4S appears to have the wrong CMA default size

JinShil, 2023-04-20:
Booting a CM4S with `dtoverlay=vc4-kms-v3d,noaudio` results in a notification on the desktop that reads _"CMA size reduced to 256MB (total_mem 1024MB, gpu_mem 76MB)"_.  It's easy to silence that message by appending `,cma-256` to the `dtoverlay=vc4-kms-v3d` line, but it'd be nice if it had a sensible default out of the box, and would prevent us from having to creating separate config.txt files for multiple variations of our products.

If I am making an incorrect assumption, please clarify why the notification is appearing and what the default should be.  Also, is there any command I can use to check what the CMA size is configured to?

pelwell, 2023-04-20:
The 256MB cap applies to devices with less than 2GB of RAM. The warning message is there to inform users that they aren't getting the amount of CMA that they might have thought they were going to get.

The actual default value is 64MB. The 384MB value that is being reduced comes from the `vc4-kms-v3d` overlay (actually, it's the `vc4-kms-v3d-pi4` overlay that is loaded). You may find that 256MB of CMA isacceptable value across your devices, which would avoid the need for different values. Alternatively, if your product variations consist of a mixture of CM3 and CM4S units then you may prefer to use a customised `vc4-kms-v3d-pi4` overlay with a different default CMA value.

JinShil, 2023-04-21:
If I explicitly configure `vc4-kms-v3d` with `cma-256`, `cma-384`, or `cma-512` the notification message does not appear.  It only appears if the `cma-xxx` setting is not explicitly configured.  So, if there is a 256MB cap, then that doesn't appear to be working.

I can also see, when setting `cma-384` and `cma-512` explicitly, `grep CmaTotal /proc/meminfo` reports `393216 kB` and `524288 kB` respectively.  So it appears the cap is only enforced when the `cma-xxx` setting is *not* explicitly configured.

pelwell, 2023-04-21:
You are correct that the firmware can distinguish between a CMA size coming from a parameter and the baked-in defaults. By adding an explicit parameter you are overriding any cap.

Where there are multiple versions of overlays, the split is by architecture (`bcm283[567]` vs `bcm2711`), not model. In this regard, CM4S is treated like all the other Pi 4s - 4B, 400 and CM4.

If there is a problem here it is the warning message - something which any 1GB Pi 4B user would see. We may reconsider this.

JinShil, 2023-04-21:
> If there is a problem here it is the warning message - something which any 1GB Pi 4B user would see.

Ah, right.  Our CM4 1GB variation would also see the notification message.  We would probably want 256MB for any 1GB models and 512MB for all others.  What about adding a config.txt filter for installed RAM capacity?  That would work for us.

pelwell, 2023-04-21:
> We would probably want 256MB for any 1GB models and 512MB for all others.

That's the current behaviour, it's only the warning with a 32-bit kernel that is unwanted; we're internally discussing removing it.

pelwell, 2023-04-21:
It's agreed - that message will be gone in future firmware releases.

closed 2023-09-16 by JinShil