raspberrypi/firmware #1809: Firmware 1.20230405: CM4 with start4cd.elf/fixup4cd.dat fails to boot

srcshelton, 2023-05-02:
**Describe the bug**

I have a headless CM4008032 module which I'm trying to optimally configure for server-type operation.

As part of this, I've tried to reducing the GPU memory to the minimum necessary by manually specifying:

```
start_file=start4cd.elf
fixup_file=fixup4cd.dat
```

But with this change alone, the CM4 fails to boot.  Commenting these two lines restores expected operation.

Full config.txt, split over 3 files `include`d by `config.txt`:

```
disable_overscan=1
[cm4]
camera_auto_detect=0
display_auto_detect=0
dtparam=noant=on,hdmi=off,sd_poll_once=on                                                         
dtoverlay=disable-bt                                                                              
dtoverlay=disable-wifi                                                                            
dtoverlay=ramoops-pi4,console-size=16384
max_framebuffers=0
arm_boost=1
[all]
start_file=start4cd.elf
fixup_file=fixup4cd.dat
arm_64bit=1
boot_delay=0
disable_poe_fan=1
disable_splash=1
enable_uart=1
force_eeprom_read=0
[cm4]
otg_mode=1
[all]
uart_2ndstage=1

include overclock.txt
initial_turbo=60
temp_limit=75

include legacy.txt
hdmi_ignore_edid_audio=1
enable_tvout=0
ignore_lcd=1
disable_touchscreen=1
hdmi_ignore_hotplug=1
disable_fw_kms_setup=1
avoid_warnings=1
```

Does the `cd` firmware variant lose any other functionality compared to the default one beyond a couple of VideoCore decoders, such as the ability to process complex `config.txt` files?

Is the `cd` firmware variant automatically chosen for reduced VideoCore memory allocations - and if so, what's the boundary to switch to this version?  Is the minimum VideoCore memory still 16MB as documented, even with the latest firmware and when running on the CM4?

*Update*: I can successfully boot with `gpu_mem=16` specified, but this doesn't appear to automatically select a different pair of start/fixup files.

**To reproduce**
With the cut-down VideoCore firmware loaded as above, the CM4 fails to boot.  Making no change other than commenting the two lines in `config.txt` restores expected boot behaviour.

**Expected behaviour**
Assuming that the two lines above are syntactically valid, I'd expect the system to boot with reduced VideoCore features available.

**Actual behaviour**

On reboot:

[... 123 lines cut ...]
MESS:00:00:04.112343:0: *** Restart logging
MESS:00:00:04.114983:0: brfs: File read: 3902 bytes
MESS:00:00:04.119884:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_gp on reading EDID block 0
MESS:00:00:04.149040:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:04.154069:0: hdmi: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:04.159667:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:04.168431:0: HDMI0: hdmi_pixel_encoding: 300000000
MESS:00:00:04.173899:0: HDMI1: hdmi_pixel_encoding: 300000000
MEMESS:00:00:04.334779:0: brfs: File read: /mfs/sd/overlays/vc4-kms-v3d-pi4.dtbo
MESS:00:00:04.399992:0: Loaded overlay 'vc4-kms-v3d'
MESS:00:00:04.564841:0: brfs: File read: /mfs/sd/overlays/disable-bt.dtbo
MESS:00:00:04.583523:0: Loaded overlay 'disable-bt'
MESS:00:00:04.586925:0: brfs: File read: 2126 bytes
MESS:00:00:04.591669:0: brfs: File read: /mfs/sd/overclock.txt
MESS:00:00:04.596148:0: brfs: File read: 2208 bytes
MESS:00:00:04.601724:0: brfs: File read: /mfs/sd/legacy.txt
MESS:00:00:04.657251:0: brfs: File read: 3902 bytes
MESS:00:00:04.659535:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:04.664452:0: Read command line from file 'cmdline.txt':
MESS:00:00:04.670302:0: 'console=serial0,115200 kgdboc=ttyAMA0,115200 root=PARTUUID=e9a81d93-fc67-4c5e-b555-d696357afa46 rootfstype=ext4 fsck.repair=yes rootwait cgroup_enable=memory cgroup_memory=1 zswap.enabled=1 zswap.compressor=lz4 zswap.max_pool_p
MESS:00:00:04.791745:0: brfs: File read: 256 bytes
MESS:00:00:07.393865:0: Kernel relocated to 0x200000
MESS:00:00:07.395721:0: Device tree loaded to 0x2eff2a00 (size 0xd5d3)
MEfffffff]
```

pelwell, 2023-05-03:
No, I'm wrong. `start_x` and `start_debug` are things, `start_cd` isn't.

The differences in your bootloader log are strange - are both boots the same in all other respects?

srcshelton, 2023-05-03:
> No, I'm wrong. `start_x` and `start_debug` are things, `start_cd` isn't.
> 
> The differences in your bootloader log are strange - are both boots the same in all other respects?

I've recently been through the documentation with a fine toothed comb - I did notice that this option seemed to be absent (which also made me wonder whether the cutdown firmware is legacy/unsupported at this point?)

In terms of the (slightly) different messages logged during boot - yes, the only difference is that one was captured after doing `shutdown -h now` from a previous session, whilst the other is following a cold boot.

timg236, 2023-05-03:
gpu_mem=16 is the only way to boot start_cd.elf

srcshelton, 2023-05-03:
Okay, can confirm that with `gpu_mem=16` I get:

```
Read config.txt bytes     5292 hnd 0x38
Read start4cd.elf bytes   805756 hnd 0x6802
Read fixup4cd.dat bytes     3173 hnd 0xc4d8
```

(… which was previously hidden by the TP2 serial issues and #1810)

I'll make a PR against the documentation to note that that start_file & fixup_file can't be used with the `cd` firmware.

Is it documented anywhere exactly _which_ features the `cd` firmware lacks?

timg236, 2023-05-03:
Sorry, no idea.

popcornmix, 2023-05-03:
> Is it documented anywhere exactly which features the cd firmware lacks?

It's not documented, but basically cameras, codecs and (firmware side/deprecated) 3d.

And the initial simple framebuffer used at the start of kernel boot is limited to 16bpp 1080p
(but kms/X can replace that with a 32bpp 4K framebuffer later in boot).

srcshelton, 2023-05-04:
Closing as documentation updates are now raised as PRs.

closed 2023-05-04 by srcshelton

arretxe, 2025-12-01:
Just wanted to mention that in my CM4, even if I set gpu_mem=16 if fixup4.dat and start4.elf are not present it fails to boot (I thought it used the cd variants so it would not need the main ones).

I also tried plenty of combinations, like:

-  removing gpu_mem=16 but specifying start_file=start4cd.elf fixup_file=fixup4cd.dat FAIL
- keeping gpu_mem=16 but specifying start_file=start4cd.elf fixup_file=fixup4cd.dat FAIL
- keeping gpu_mem=16, removing start_file4cd.elf and fixup4cd.elf FAIL
- keeping gpu_mem=16, rename start4cd.elf and fixup4cd.dat to fixup4.dat and start4.elf FAIL
- keeping gpu_mem=16, rename start4cd.elf and fixup4cd.dat to fixup4.dat and start4.elf while keeping a copy of start4cd.elf and fixup4cd.dat SUCCESS

This took a while to discover, as I was compiling my buildroot with 
#BR2_PACKAGE_RPI_FIRMWARE_VARIANT_PI4 is not set
BR2_PACKAGE_RPI_FIRMWARE_VARIANT_PI4_CD=y
This generates only the cd variants, not the main ones, so I need to manuallly copy them.

Would appreciate if someone with deeper knowledge could confirm if this makes sense.
I thought maybe some earlier boot phase loooks for the non cd variants first, although apparently for that purpose the cd variants seem to work too if renamed.