raspberrypi/firmware #1810: 'gpu_mem' config.txt option doesn't appear to work from included configuration file

srcshelton, 2023-05-02:
**Describe the bug**
I've split my boot configuration into three files (`config.txt`, `overclock.txt`, `legacy.txt`).  I can confirm through the use of `vcgencmd` that settings from all three files are being loaded.

However, the `gpu_mem` parameter only appears to work when placed in `config.txt` (… or, potentially, there's a smaller maximum data-length which applies to `gpu_mem` and when set in a later `include`d file, there's too much preamble and it's not seen?).

**To reproduce**

* In `config.txt` add `gpu_mem=16`;
* Reboot;
* Confirm that `vcgencmd get_mem gpu` returns `gpu_mem=16M`;
* Remove `gpu_mem=16` from `config.txt` and add it to `legacy.txt`;
* Confirm that `config.txt` contains the line `include legacy.txt`;
* Reboot;
* Check the output of `vcgencmd get_mem gpu`;
* Check the output of `vcgencmd get_config ...` with various other settings also defined in `legacy.txt`.

**Expected behaviour**
`vcgencmd get_mem gpu` should return `gpu_mem=16M` regardless of which (`include`d) file it is specified in.

**Actual behaviour**
`gpu_mem=16M` is returned when specified in `config.txt`, but when this is set in another file (such as my `legacy.txt`) the output is the default of `gpu_mem=76M`, even though *other* settings from the same file have taken effect and are shown in `vcgencmd` output.

I've also tried `gpu_mem=16` at the top of the second (rather than third) file, `overclock.txt` - this line is also ignored whilst other settings further down the same file are processed.

**System**

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
CM4008032

* Which OS and version (`cat /etc/rpi-issue`)?
Gentoo Linux

* Which firmware version (`vcgencmd version`)?
1.20230405:

```
Mar 17 2023 10:50:39 
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)
```

* Which kernel version (`uname -a`)?
`6.1.21-v8+`

timg236, 2023-05-03:
include statements have never been supported by the bootloader

srcshelton, 2023-05-03:
> include statements have never been supported by the bootloader

It's unclear to me having read through all of the documentation at https://www.raspberrypi.com/documentation/computers/config_txt.html & https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#boot-sequence that there was any distinction between directives or parts of `config.txt` which are only valid for the bootloader (and so don't support the documented `include` directive?) and other directives of parts of `config.txt` which can be sourced from anywhere including `include`d files.

Can you provide more clarity on what directives are supported at what stage of boot?

(I'm happy to add documentation PRs as a result!)

srcshelton, 2023-05-04:
@pelwell @popcornmix Can you offer any additional guidance on the above?

timg236, 2023-05-04:
I'm not sure what needs adding, the include restriction is already documented
https://www.raspberrypi.com/documentation/computers/config_txt.html#include

srcshelton, 2023-05-16:
To rephrase the request here:

> _From the existing documentation, it's unclear which directives in `config.txt` are parsed by `bootcode.bin`/the EEPROM loader (and so don't support `include` statements), and which are processed by `start*.elf` (and so do)_

Again, happy to add PRs to the documentation repo with the above clarification.

cleverca22, 2023-07-31:
https://github.com/librerpi/rpi-open-firmware/blob/master/docs/bootcode_re.txt#L14-L30

i was also maintaining a list here, of every config flag the bootloader supports, and the type of each
but i havent updated it in a while

anything from that list, has to be in the main config.txt

srcshelton, 2023-09-16:
Closing as the documentation PR is now merged (albeit based on third-hand information rather than any list provided or confirmed by the maintainers of this firmware, which is unfortunate…)

closed 2023-09-16 by srcshelton