raspberrypi/firmware #1813: RPI4 Start*.elf disappeared

DigitalTrading, 2023-05-16:
**Describe the bug**
Have a several kiosk machines running custom software on a RPI 4 model B which has been working for the last few weeks but suddenly last week 3 of the machines have suddenly shown start4.elf not found and refused to boot. 

Looking at the files (FAT32 partition) on the SD card itself I can seem most of the files I would expect but not start elf files or recovery elf files.

As far as I'm aware there have been no updates or changes on the SD card. The software on the PI only interacts with the Ext4 partition. Its possible (maybe probable) that staff have been simply unplugging the devices instead of properly shutting down.

The kiosk software is just displaying a website locally on the pi and is not connected to the internet.

There are some buttons linked to the GPIO for user interaction.

**To reproduce**
Unsure on how to reproduce, these have been running well for 3 weeks, we also have a few others less often used that are still working.

**Expected behaviour**
In theory with no updates, I would expect the pi to keep booting.

**Actual behaviour**
Boot failure:
"Start4.elf not found"

System Information
------------------

Raspberry Pi 4 Model B Rev 1.5
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"

Raspberry Pi reference 2021-10-30
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 288b21fc27e128ea6b330777aca68e0061ebf4fe, stage2

Linux raspberrypi-blitX 5.10.92-v7l+ #1514 SMP Mon Jan 17 17:38:03 GMT 2022 armv7l GNU/Linux
Revision        : b03115
Serial          : 10000000176b5120
Model           : Raspberry Pi 4 Model B Rev 1.5
Throttled flag  : throttled=0x0

**Additional context**
I'm getting my team to check the software they built in order to ensure it wasn't something in that somehow accessing the FAT32 partition. But is there any other reason why the Start or recover elf files might disappear?

I can rewrite the SD cards from a backup to fix the immediate problem, but I need to find out what may have caused those files to disappear.

pelwell, 2023-05-16:
The firmware and bootloader never write to the SD card, so you can rule that out. The only Linux utilities I can think of that would write to /boot are `raspi-config`, `rpi-update` and `apt install/upgrade`, etc.

DigitalTrading, 2023-05-16:
The only ones that the software in the background are using are:

* vcgencmd measure_temp
* vcgencmd get_mem
* vcgencmd get_volts
* vcgencmd get_throttled
* Reading from /proc/stat

There's no internet connection on the machines so it wont be doing any updates in the background, and the staff don't have access to a keyboard etc so shouldn't be anyone fiddling with it or trying to hack it. I forgot to mention that these 3 machines were also geographically separated so I would rule out someone tinkering with them.
The software shouldn't have been doing any file deletions or anything itself (is it possible to access the boot partition from inside the OS?)

pelwell, 2023-05-16:
> (is it possible to access the boot partition from inside the OS?)

Yes, if you've mounted it. If you don't need it, don't mount it.

So far you are the only person reporting anything like this (that I've come across). The fact that it affects multiple devices running (I presume) the same software does suggest that it is at least something specific to your software environment, and most likely something triggered by your software.

> The firmware and bootloader never write to the SD card

A minor correction - the special EEPROM update "recovery" utility may delete/rename the installed image to prevent it being installed on each reboot.

DigitalTrading, 2023-05-16:
Thanks for that, I am assuming its something to so with the setup, though I wasn't sure if there was a something on the OS version perhaps.

So you don't think its just the switching off the machines that may be causing it?

What is the special EEPROM update recovery utility?

pelwell, 2023-05-16:
> So you don't think its just the switching off the machines that may be causing it?

I'm not going to rule it out, but unless something is writing to FAT partition there should be no reason for it to get corrupted (and nothing else).

> What is the special EEPROM update recovery utility?

The programming of a new EEPROM image is done by a special `recovery.bin` that is loaded directly by boot the ROM during a reboot in preference to the image in the EEPROM.

DigitalTrading, 2023-05-16:
Ah ok so likely its not the EEPROM.

Is there anyway I can test or track what has access to the FAT partition or if it is mounted?

pelwell, 2023-05-16:
Do you mean something that will notify you whenever it has been mounted? I can't think of one.

popcornmix, 2023-05-16:
You can check if /boot is currently mounted:
```
$ mount | grep /boot
/dev/mmcblk0p1 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
```
It almost certainly will be. If you are sure you don't want to access /boot from linux (i.e. you are not going to run apt upgrade or modify config.txt) you can prevent /boot from being mounted.
Comment out this line from `/etc/fstab` (your PARTUUID will be different). Add a # to start of line to comment it
```
PARTUUID=5201065d-01  /boot           vfat    defaults          0       2
```
and reboot. Now you won't be able to see contents of /boot and so, it would be very difficult for an errant script to delete a file from there.

Uncomment and reboot if you do want to upgrade firmware/kernel at a later date.

DigitalTrading, 2023-05-16:
Great thank you so much, with luck that will stop the problem happening again in the future.

timg236, 2024-01-06:
Seems to be resolved / not a firmware issue

closed 2024-01-06 by timg236