raspberrypi/firmware #1815: Missing BCM2837 DTBs in /boot

pedro-javierf, 2023-06-25:
I can't find any .dtb files for `BCM2837` based boards like the Raspberry Pi 3b+ under the /boot directory of the repo. Are these missing? There seems to be board revisions with different broadcom SoCs, all under the same name, so if someone could clarify all the possible existing combinations too, that would be really helpful.

pelwell, 2023-06-25:
I guess you didn't try `*3*` then? There aren't that many files to read through.

BCM2837 is also known as BCM2710. I've a feeling that one was the name of the bare die and one the packaged part, but that could be a false memory. On BCM2711 there was no 283x equivalent name (which came as a surprise to some - bits of firmware code still refer to BCM2838).

pedro-javierf, 2023-06-26:
The name on the package is BCM2837 yes, hence my confusion. 

I didn't try the BCM2710-pi-3-b-plus DTB because I'm attempting to boot a custom kernel8.img that isn't Linux based. It isn't working without providing any DTB (And I think no DTB is needed anyways for just a simple baremetal image that prints something over UART) but I needed a baseline where it works, so I was trying to get the appropriate file just in case it was needed. 

Some external repos like this https://github.com/icon299/rpi3-firmware/blob/master/bcm2837-rpi-3-b-plus.dtb do mention 2837 DTBs but no idea where they come from.

pelwell, 2023-06-26:
The bcm283x DTB names are used upstream (the common Linux kernel without our modifications). You can still build them from our kernel trees (`make ... dtbs`), but we don't include them in our images.

pedro-javierf, 2023-06-26:
Makes sense. So Raspberry Pi official uses the BCM2710 (silicon name) whereas others might use the package name, correct?

pelwell, 2023-06-26:
Something like that - they're either conveniently different (because they can coexist) or annoyingly different, depending on your point of view.

closed 2023-06-27 by pedro-javierf