raspberrypi/firmware #1816: Unstable bootcode/start.elf microSD access on Raspberry pi 3b+

pedro-javierf, 2023-06-27:
**Describe the bug**
The latest bootloaders are having issues accessing the micro SD card consistently during boot time.

**To reproduce**
Raspberry Pi 3 b+ 
Tried two different microSD cards:
* Sandisk EDGE 16Gb (micro SDHC I1 A1 class 10) (official microSD bought w/ the raspberry)
* Generic 16Gb card (micro SDHC I1 class 10)
* Latest bootcode.bin from https://github.com/raspberrypi/firmware/raw/master/boot/bootcode.bin (modified with UART_BOOT=1 to see traces)
* Latest fixup.dat
* Latest start.elf
* config.txt with the following options:
  * enable_uart=1
  * core_freq=250
  * dtoverlay=pi3-disable-bt
  * uart_2ndstage=1
  * arm_64bit=1 

**Expected behaviour**
The device should consistently boot up to the point of starting the kernel8.img (Linux) on the microSD card.

**Actual behaviour**
A plethora of different traces are seen across boots. **UART is always available,** which must mean that the GPU first stage bootloader is able to access the microSD card and start the UART_BOOT=1 modified bootcode.bin.

Then, the following scenarios have been observed on UART:
* `No SD card detected, Try USB` (How's this possible?? clearly the microSD is accessible, since I'm seeing UART traces)
* `Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 0` (now it progresses further but still fails to access sd reliably, tries to do USB boot anyways)
* `Found SD card, config.txt = 1, start.elf = 1, recovery.elf = 0, timeout = 0` but then it never manages to boot the kernel8 image, although it attempts to ("Loading kernel8.img at 0x80000) and so on.

I have tried formatting both microSDs in FAT32 repeatedly (with the SD foundation's official tool "SD card formatter", with Windows, and with MacOS). Formatting the card usually changes the odds of what of the scenarios is seen, but it appears to be very random.

Installing a mainstream OS (raspbian, etc) by flashing an .iso into the device (which installs all the needed partition) seems to work just fine and the OS always boots.

**Additional context**
These microSD cards work perfectly with mainstream distributions. Could power supply or the speed grade of the microSD cards have something to do? Am I using incorrect bootloaders? Thanks.

pelwell, 2023-06-27:
Have you considered using the `stable` firmware branch, which is used in the Debian package and image releases,  instead, i.e.: https://github.com/raspberrypi/firmware/raw/stable/boot/bootcode.bin ?

pedro-javierf, 2023-06-27:
Hadn't even realised there was a stable branch since it's not mentioned anywhere. I'll give it a try

pedro-javierf, 2023-06-27:
I have been able to reliably get to a `Loaded 'kernel8.img' to 0x80000 size 0x80064` with both the latest bootcode.bin (and also the stable bootcode.bin) by stripping the config.txt to just this:

```
enable_uart=1
#core_freq=250
#dtparam=audio=on_
#dtoverlay=pi3-disable-bt

# make start.elf print debug over the serial port
uart_2ndstage=1

# Let's boot the AArch64 kernel
arm_64bit=1
```

This got me thinking, could it be that the commented lines (e.g.: the dtoverlay) were making the bootcode.bin or the start.elf crash? **I don't have those overlays in the microSD**. Not sure if the `dtparam` and `core_freq` could have also had any effect. I wasn't expecting the different bootloaders to be so brittle to crash because of a missing file.

Now I'm just stuck at:

```
MESS:00:00:01.455269:0: HDMI0: hdmi_pixel_encoding: 162000000
MESS:00:00:01.460982:0: vec: vec_middleware_power_on: vec_base: 0x7e806000 rev-id 0x00002708 @ vec: 0x7e806100 @ 0x00000420 enc: 0x7e806060 @ 0x00000220 cgmsae: 0x7e806050
MESS:00:00:01.477756:0: kernel=
MESS:00:00:01.484401:0: dtb_file 'bcm2710-rpi-3-b-plus.dtb'
MESS:00:00:01.488429:0: dtb_file 'bcm2710-rpi-3-b.dtb'
MESS:00:00:01.493287:0: dtb_file 'bcm2837-rpi-3-b-plus.dtb'
MESS:00:00:01.498579:0: Failed to load Device Tree file '?'
MESS:00:00:01.503874:0: Failed to open command line file 'cmdline.txt'
MESS:00:00:01.543080:0: brfs: File read: /mfs/sd/kernel8.img
MESS:00:00:01.547046:0: Loaded 'kernel8.img' to 0x80000 size 0x80064
MESS:00:00:01.553199:0: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
MESS:00:00:01.562684:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:01.568976:0: uart: Baud rate change done...
MESS:00:00:01.572406:0: uart: Baud rate
```

So I probably have a broken baremetal Kernel, but I'll see what else I can find

pedro-javierf, 2023-06-27:
I couldn't manage to make it fail again with just those options mentioned above, so it must have been all the comments that I stripped from the original config.txt that I got from the docs (which contains a lot of commented options) and maybe I changed something else in there. However, doing this I found why I might have been seeing those errors before when playing with the config.txt, and I've opened issue #1817

closed 2023-06-30 by pelwell