raspberrypi/firmware #1817: Bootloader misleading/wrong UART error message when config.txt is not valid in SD boot path

pedro-javierf, 2023-06-27:
**Describe the bug**
When an invalid config.txt is found by the start.elf bootloader, SD boot is aborted with a misleading UART trace. Afterwards, the next boot option (e.g.: USB) is attempted.

The bootloader prints this to UART
```
Raspberry Pi Bootcode

No SD card found, try USB
Trying USB
``` 
Anytime a wrong line is found in config.txt. When using a Raspberry Pi 3 or lower, this is misleading since the SD card is obviously found (bootcode.bin is being loaded from microSD in these models lower than the Raspberry Pi 4.).

This bug is probably also present in the Raspberry Pi 4, and it would be even more confusing to users, since the second stage bootloader `bootcode.bin` is no longer present in the SD card but in EEPROM.

I have not checked it, but if the config.txt found in the next boot medium (USB, Ethernet, etc) were wrong, the message might also be misleading.

**To reproduce**
1. Use a Raspberry Pi 3b+
2. Put all needed bootloader stages (stage 2 and stage 3) into the microSD card: `bootcode.bin` (enable UART_BOOT=1 to see the traces), `start.elf` and `fixup.dat.
3. Write a wrong config.txt. For example:
```
aaaa=10
S9yasyduia as' :a_= =Qw9diujishdaauoix
D9uasjichjashciuahs8asy!·$"%&/("/&%/()IOKL\n\rdisuyd87\0AAA
#init_uart_clock=3000000
enable_uart=1
#core_freq=250
#dtparam=audio=on_
#dtoverlay=pi3-disable-bt

# make start.elf print debug over the serial port
uart_2ndstage=1

# Let's boot the AArch64 kernel
arm_64bit=1
```

4. This will consistently show the message above in 100% of the boot attempts. Very confusing since there's actually no problem at all with the sd card.

**Expected behaviour**
The bootloader message should not be misleading. Examples could be:
`Aborted SD card boot. Invalid config.txt`
`Invalid options found in config.txt`(bonus points if the parser could pinpoint in which line number the invalid option is found)

Just make sure that it's clear that the SD card boot has been aborted because the bootloader can't know what parameters to use. "No SD card found" is not a reasonable message since the device can indeed find and mount and access the SD card. 

**Actual behaviour**
```
Raspberry Pi Bootcode

No SD card found, try USB
Trying USB
``` 

**Additional context**
The `config.txt` I've provided is very wrong. However I have experienced this same issue just with very small changes or just invalid options in the file. Maybe invalid characters or bytes cause this too, but it's difficult to know for sure since the bootloader and the config.txt parser are closed source.

popcornmix, 2023-06-28:
I tried my Pi3+ which had a bootable system, and I replaced my config.txt with your one.
And I don't see your error.

Can you try with a different sdcard?

My uart output with your config.txt

```
Raspberry Pi Bootcode
Read File: config.txt, 319
Read File: start.elf, 2975616 (bytes)
Read File: fixup.dat, 7262 (bytes)
MESS:00:00:02.179463:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:02.183721:0: brfs: File read: 319 bytes
MESS:00:00:02.243919:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:02.475240:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:02.483635:0: *** Restart logging
MESS:00:00:02.486123:0: brfs: File read: 319 bytes
MESS:00:00:02.507827:0: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
MESS:00:00:02.524156:0: HDMI0: hdmi_pixel_encoding: 162000000
MESS:00:00:02.528906:0: kernel=
MESS:00:00:02.541058:0: dtb_file 'bcm2710-rpi-3-b-plus.dtb'
MESS:00:00:02.551784:0: brfs: File read: /mfs/sd/bcm2710-rpi-3-b-plus.dtb
MESS:00:00:02.556880:0: Loaded 'bcm2710-rpi-3-b-plus.dtb' to 0x100 size 0x743c
MESS:00:00:02.577753:0: brfs: File read: 29756 bytes
MESS:00:00:02.597635:0: brfs: File read: /mfs/sd/overlays/overlay_map.dtb
MESS:00:00:02.687127:0: brfs: File read: 1667 bytes
MESS:00:00:02.692174:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:02.702237:0: brfs: File read: 319 bytes
MESS:00:00:02.707114:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:02.711105:0: Read command line from file 'cmdline.txt':
MESS:00:00:02.716977:0: 'console=tty1 console=ttyAMA0,115200 nfsroot=10.3.14.140:/home/dom/bullseye64,vers=3 rw ip=dhcp rootwait'
MESS:00:00:03.389877:0: brfs: File read: 1841 bytes
MESS:00:00:03.930061:0: brfs: File read: /mfs/sd/kernel8.img
MESS:00:00:03.934011:0: Loaded 'kernel8.img' to 0x80000 size 0x86cd63
MESS:00:00:05.414217:0: Kernel relocated to 0x200000
MESS:00:00:05.417463:0: Device tree loaded to 0x2eff8600 (size 0x7993)
MESS:00:00:05.425102:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:05.431406:0: uart: Baud rate change done...
MESS:00:00:05.434818:0: uart: Baud rate
```

pedro-javierf, 2023-06-29:
You might be right. However I have already tried 3 different microSD cards, and different power supplies too, to no avail.
I am wondering if there might be some other hardware issue (e.g.: microSD slot not making proper contact) or something else causing these issues.

Is there any recommended partition scheme for the microSD's? I'm just doing a FAT32 one-partition formatting.

popcornmix, 2023-06-30:
```
Is there any recommended partition scheme for the microSD's? I'm just doing a FAT32 one-partition formatting.
```
To rule that out, you could start with a RPiOS lite image, and then just edit the config.txt and add your specific kernel.
But I wouldn't expect that to be significant. But it's useful information if it does seem to behave differently.