raspberrypi/firmware #1821: Raspberry Pi 4B device tree contains over long property name?

richlowe, 2023-08-02:
The device tree specification[1] says that property names are between 1 and 31 characters, the tree for raspberry pi 4B at least contains a property "arm,cpu-registers-not-fw-configured" which is overly long, per that specification.

```
; for elt in *.dtb; do fdtdump $elt 2>/dev/null | grep -q 'not-fw-configured'; [[ $? == 0 ]] && echo "$elt"; done
bcm2711-rpi-4-b.dtb
bcm2711-rpi-400.dtb
bcm2711-rpi-cm4-io.dtb
bcm2711-rpi-cm4.dtb
bcm2711-rpi-cm4s.dtb
```

I'm not sure if there's much you can do about this, or if something about how I've read the specification is incorrect.

[1] https://devicetree-specification.readthedocs.io/en/v0.3/devicetree-basics.html#properties

pelwell, 2023-08-02:
The source code for `dtc` and associated utilities includes the following definitions:
```
#define MAX_PROPNAME_LEN  31
#define MAX_NODENAME_LEN  31
```
They are completely unused.

popcornmix, 2024-02-23:
Closing as this is an upstream kernel issue.

closed 2024-02-23 by popcornmix