raspberrypi/firmware #1822: Cannot play audio from Raspberry Pi 2 jack output

AlphaJack, 2023-08-11:
## Describe the bug
I cannot play audio from my Raspberry Pi 2 jack output.

I know that in 2022-02-16 all I had to do was enabling `device_tree_param=audio=on`, installing `alsa-utils` and adding my user to the audio group

## To reproduce
Install the latest OS version, kernel 6.1.44
In `/boot/config.txt`, set

```ini
dtparam=audio=on

# also different combinations of the following have been tried:

#hdmi_group=1
#hdmi_mode=1

#audio_pwm_mode=2

#hdmi_group=2
#hdmi_drive=2
```

Add your user to the `audio` group: `sudo usermod -aG audio $USER`

Reboot

## Expected behaviour
`aplay -l` lists the jack audio card, `alsamixer` shows  control for the soundcard, `speaker-test -t wav -c 6` works properly

## Actual behaviour
`aplay -lv`:

```
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi [vc4-hdmi], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

`alsactl info`:
```
#
# Sound card
#
- card: 0
  id: vc4hdmi
  name: vc4-hdmi
  longname: vc4-hdmi
  driver_name: vc4-hdmi
  mixer_name: 
  components: 
  controls_count: 4
  pcm:
    - stream: PLAYBACK
      devices:
        - device: 0
          id: MAI PCM i2s-hifi-0
          name: MAI PCM i2s-hifi-0
          subdevices:
            - subdevice: 0
              name: subdevice #0
alsactl: rawmidi_device_list:105: snd_ctl_rawmidi_next_device
```

`lsmod | grep bcm2835`
```
bcm2835_codec          49152  0
bcm2835_v4l2           45056  0
bcm2835_isp            32768  0
videobuf2_vmalloc      16384  1 bcm2835_v4l2
v4l2_mem2mem           36864  1 bcm2835_codec
bcm2835_mmal_vchiq     32768  3 bcm2835_isp,bcm2835_codec,bcm2835_v4l2
videobuf2_dma_contig    20480  2 bcm2835_isp,bcm2835_codec
videobuf2_v4l2         32768  4 bcm2835_isp,bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem
videobuf2_common       65536  8 bcm2835_isp,bcm2835_codec,videobuf2_dma_contig,videobuf2_vmalloc,videobuf2_memops,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
videodev              258048  6 bcm2835_isp,bcm2835_codec,videobuf2_common,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
mc                     53248  6 bcm2835_isp,bcm2835_codec,videobuf2_common,videodev,v4l2_mem2mem,videobuf2_v4l2
i2c_bcm2835            16384  0
vc_sm_cma              36864  2 bcm2835_isp,bcm2835_mmal_vchiq
snd_bcm2835            28672  0
snd_pcm               118784  5 snd_compress,snd_pcm_dmaengine,snd_soc_hdmi_codec,snd_bcm2835,snd_soc_core
snd                    86016  6 snd_compress,snd_soc_hdmi_codec,snd_timer,snd_bcm2835,snd_soc_core,snd_pcm
```

`sudo vcgencmd version`
```
Apr 25 2023 18:32:24 
Copyright (c) 2012 Broadcom
version d7f9c2b4ef7e4a8c0b04374a879ce89d7a948453 (clean) (release) (start_cd)
```

`uname -a`
```
Linux $HOST 6.1.44-1-rpi-ARCH #1 SMP Wed Aug  9 15:17:57 MDT 2023 armv7l GNU/Linux
```

`sudo dmesg | grep bcm2835`
```
[    0.000000] Kernel command line: coherent_pool=1M snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_hdmi=0 video=Composite-1:720x480@60i,margin_left=32,margin_right=32,margin_top=32,margin_bottom=32 vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  root=UUID=d4dcf94d-8281-451b-bc01-c30e555f96c6 rw rootwait console=ttyAMA0,115200 console=tty1 selinux=0 fsck.repair=yes plymouth.enable=0 smsc95xx.turbo_mode=N dwc_otg.lpm_enable=0 kgdboc=ttyAMA0,115200 ipv6.disable=1 cgroup_enable=memory cgroup_memory=1
[    0.071669] bcm2835-mbox 3f00b880.mailbox: mailbox enabled
[    1.693373] bcm2835-dma 3f007000.dma: DMA legacy API manager, dmachans=0x1
[    2.325428] bcm2835-rng 3f104000.rng: hwrng registered
[    2.331665] gpiomem-bcm2835 3f200000.gpiomem: Initialised: Registers at 0x3f200000
[    4.718057] bcm2835-wdt bcm2835-wdt: Broadcom BCM2835 watchdog timer
[    4.730984] bcm2835-power bcm2835-power: Broadcom BCM2835 power domains driver
[    4.832951] mmc0: sdhost-bcm2835 loaded - DMA enabled (>1)
[   23.165291] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
[   30.063847] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
[   31.535835] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[   31.779631] bcm2835_isp: module is from the staging directory, the quality is unknown, you have been warned.
[   31.793073] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
[   31.908179] bcm2835_v4l2: module is from the staging directory, the quality is unknown, you have been warned.
[   31.918559] bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
[   31.929288] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
[   31.937152] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
```

# EDIT

From `dmesg` I saw a command line parameter that I haven't specified, `snd_bcm2835.enable_headphones=0`. Forcing it to 1 in `/boot/cmdline.txt` and rebooting made the sound card apppear:

`aplay -l`
```
**** List of PLAYBACK Hardware Devices ****
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 1: vc4hdmi [vc4-hdmi], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
However, I don't head audio, just a light electric buzz from the speaker when the audio is supposed to appear. The issue also happens on different headphones and by setting `audio_pwm_mode=2`

Is `dtparam=audio=on` setting `snd_bcm2835.enable_headphones=0`?

pelwell, 2023-08-11:
It works for me on a clean RPiOS image with no changes, running on a 2B:
```
pi@raspberrypi:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 1: vc4hdmi [vc4-hdmi], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@raspberrypi:~$ grep audio /boot/config.txt
# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# Enable audio (loads snd_bcm2835)
dtparam=audio=on
pi@raspberrypi:~$ cat /proc/cmdline
coherent_pool=1M snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0 video=Composite-1:720x480@60i vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=PARTUUID=219e5202-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
```
The cmdline may look strange, with two `enable_headphones` settings, but it is behaving as expected. The base dtb files set `snd_bcm2835.enable_headphones=0`, and `dtparam=audio=on` _appends_ `snd_bcm2835.enable_headphones=1` - the settings are processed in order, so the `=1` wins.

AlphaJack, 2023-08-11:
I'll check with the Arch Linux ARM guys then

`speaker-test -t wav -c 2` works properly, while `speaker-test -t wav -c 6` still outputs buzz

closed 2023-08-11 by AlphaJack