raspberrypi/firmware #1825: Automatic `over_voltage` unreliable

srcshelton, 2023-09-16:
**Describe the bug**
With recent firmware revisions, selection of an appropriate `over_voltage` matched to any specified `arm_freq` is intended to be automatic.

I have 4 RPi CM4 devices (1 x CM4008032, 3 x CM4008016), and have found that I can overclock them as follows:

```
CM4008032 - stable with arm_freq=2250 only, core volt=1.0350V;
CM4008016 - stable with arm_freq=2100 only, core volt=0.9920V;
CM4008016 - stable with arm_freq=2100 only, core volt=1.0020V;
CM4008016 - stable with arm_freq=1900 only.
```

… however, in the last case, specifying `over_voltage=5` allows `arm_freq=1950` and `over_voltage=6` allows `arm_freq=2000` (at `core volt=1.0200V`).

So for the CM4008032 the firmware has correctly selected a higher voltage, but with the third CM4008016 the firmware is failing to select a higher voltage which is still less then (manual) `over_voltage=6`.

**To reproduce**
Set `arm_freq` above 1900 in config.txt on certain CM4 modules, and instead of increasing the core voltage (within the range of `over_voltage` <= 6), the system is unable to boot.

**Expected behaviour**
Possibly the frequency/voltage relationship is based on a lookup-table or similar, and the third CM4 is particularly unsuited to overclocking - but my understanding was that the appropriate voltage would automatically be dynamically set to a suitable value to try to meet the selected `arm_freq`, up to the voltage which would be used if `over_voltage=6` were manually specified.

**Actual behaviour**
For `arm_freq=1950`, a value of no more than `over_voltage=4` seems to have been chosen rather than the `over_voltage=5` which is necessary for this particular CM4 module.

Perhaps a `over_voltage_table_offset` firmware variable would be useful to select a higher `over_voltage` than would otherwise be automatically chosen?

**System**
* Which model of Raspberry Pi? `CM4`
* Which firmware version (`vcgencmd version`)?
```
Mar 17 2023 10:50:56 
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start_cd)
```
* Which kernel version (`uname -a`)?
```
Linux rpi-cm4-8-16-24.seventytwo.miltonroad.net 6.1.21_p20230405 #20 SMP Wed Jun 28 23:03:43 BST 2023 aarch64 GNU/Linux
```

popcornmix, 2023-09-17:
No overclock setting is guaranteed.
If 1950MHz worked on all Pi4/cm4 devices we would likely have made that the default frequency.

The rules for overclocking is very much YMMV.
Some boards may never run reliably at 1950MHz (with any overvoltage).
Other boards may run, but need a manual, higher overvoltage.

This is expected behaviour and not a bug.

srcshelton, 2023-09-17:
Sorry, my point wasn't that I was expecting any degree of overclock - it was that my understanding was that `over_voltage` should now be automatic so that if it's possible to hit a given speed (without `over_voltage` going above 6), then this voltage is automatically applied without the need for further configuration.

Is this simply a lookup-table (thus why this one device which clearly doesn't have much headroom seems to be getting an `over_voltage` of 4 or below when it needs 5 - in which case see my request to add an ability to apply a correction-factor to trim/increase these default levels) - or is the voltage choice dynamic (in which case I believe there is an issue as an `over_voltage` of 5 seems to be both permissible and selected for other speeds on other devices, but isn't being selected here when it appears necessary).

I'm going by the documentation, which says:

> The latest firmware automatically scales up the voltage if the system is overclocked. Manually setting over_voltage disables automatic voltage scaling for overclocking.

… although in this case, the scale-up appears insufficient.  Should the documentation indicate that this scaling is on a best-effort only basis, and can't always be relied upon?

timg236, 2023-09-18:
Use over_voltage_delta=n (where n is integer uV)

This is a simple constant offset that is added to whatever the DVFS calculation is. For historical reasons the over_voltage parameter feeds into the start of the DVFS algorithm so is unlikely to do what you want.

N.B. over-clocking is not supported and the voltage/frequency relationship is not linear and will vary from board to board. The difference is more noticeable outside of the tested ranges which is one reason why it's not supported.

timg236, 2023-09-18:
It's signed, and you can specify any value up to whatever the built-in hardcoded limit on max-core voltage is. Although, much over whatever DVFS fixed (vcgencmd measure_voltage uncached) will cause the Pi to get hot.

geerlingguy, 2023-10-03:
(Just wanted to comment `vcgencmd measure_volts uncached` is the corrected command.)

xiongyw, 2023-11-30:
Is it possible to fixed the voltage at fixed value (say 0.8v)? I want to see how far it can goes in frequency at a fixed voltage.

timg236, 2024-01-06:
Closing as question answered

closed 2024-01-06 by timg236