raspberrypi/firmware #1827: RPi 4 - BCM43455 Cypress CLM database incorrect for region IN

3y3p4tch, 2023-09-20:
**Describe the bug**
I am trying to connect my Raspberry Pi 4 Model B to my 5GHz AP running on channel 149. Channel 149 is supported for the country IN as per the wireless regulations (can be verified with [wireless-regdb](https://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git/tree/db.txt)) but the Raspberry Pi fails to detect it. I am almost certain the issue is with the CLM blob file provided. Adding details below:

**To reproduce**
```bash
$ iw reg set IN
```
```
global
country IN: DFS-UNSET
	(2402 - 2482 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 30), (N/A)
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS
	(5470 - 5725 @ 160), (N/A, 24), (0 ms), DFS
	(5725 - 5875 @ 80), (N/A, 30), (N/A)
```
```bash
$ iw phy0 channels
```
```
Band 1:
	* 2412 MHz [1]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz
	* 2417 MHz [2]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz
	* 2422 MHz [3]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz
	* 2427 MHz [4]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz
	* 2432 MHz [5]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz
	* 2437 MHz [6]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz
	* 2442 MHz [7]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz
	* 2447 MHz [8]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz
	* 2452 MHz [9]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz
	* 2457 MHz [10]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz
	* 2462 MHz [11]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz
	* 2467 MHz [12]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz
	* 2472 MHz [13]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz
	* 2484 MHz [14] (disabled)
Band 2:
	* 5170 MHz [34] (disabled)
	* 5180 MHz [36]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz HT40+ VHT80
	* 5190 MHz [38] (disabled)
	* 5200 MHz [40]
	  Maximum TX power: 20.0 dBm
	  Channel widths: 20MHz HT40- VHT80
[... 246 lines cut ...]
**Actual behaviour**
enabled channels are not as per the regulatory information in `iw reg get`

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? `RPi4B`
* Which OS and version? `ArchLinux`
* Which firmware version? `kernel: brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Nov  1 2021 00:37:25 version 7.45.241 (1a2f2fa CY) FWID 01-703fd60`
* Which kernel version? `6.1.54`

The firmware blobs are from the latest release of archlinux package `firmware-raspberrypi`

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
I have gone through some related threads, but to no avail. Adding the threads here:
- [https://forums.raspberrypi.com//viewtopic.php?f=36&t=291824]() - CLM blob file discussions
- [https://forums.raspberrypi.com//viewtopic.php?f=117&t=291609]() - CLM blob file feedback
- [https://github.com/raspberrypi/linux/issues/4530]() - similar problem for CN country code

I have also noticed that there is a global CLM file, which enables all channels (or, to the best of my knowledge, fixes this issue). If such a CLM blob exists, I would like to get access to it so that I can use that till a fix is available.

Please let me know if any additional data is required. I'll be happy to add more logs

pelwell, 2023-09-21:
Creating a new CLM blob file has required us to submit all of the RF test reports. I believe that laborious process is almost complete, at which point we can ensure that the channel maps are correct before releasing the updated blob.

3y3p4tch, 2023-09-21:
Is this being done internally or is there any thread I can follow for updates on this? Thanks for a quick response @pelwell

pelwell, 2023-09-21:
Internally and between us and Infineon. I'll post a response when something changes, or you can watch this repo: https://github.com/RPi-Distro/firmware-nonfree/

pm4rcin, 2023-10-15:
@pelwell will you send the updated blob to upstream kernel? Because other distros that use upstream linux-firmware have already outdated firmware.

pelwell, 2023-10-16:
I've always been reluctant to replace a generic upstream firmware with one that has been tailored to our needs, but I guess it may be an option now that the kernel is qualifying firmware names with the model compatible string from Device Tree.

pm4rcin, 2024-02-10:
@pelwell is there any progress on that firmware upstreaming?

pelwell, 2024-02-10:
Unfortunately the only way to get a new CLM blob is to provide again all the RF test data, and we're still waiting for the person responsible here to get it done.

pm4rcin, 2024-05-02:
@pelwell you have recently switched to that generic firmware because of some issues with newer one. Does it mean that updated firmware will be in August this year per cypress policy? Is there a chance of upstreaming next version in August?

pelwell, 2024-05-02:
> Does it mean that updated firmware will be in August this year per cypress policy?

Pardon?

> Is there a chance of upstreaming next version in August?

It's not our firmware to upstream.

pm4rcin, 2024-05-02:
I thought cypress develops Wifi/BT chip firmware for you. I have taken the Aug 2024 from [here](https://community.infineon.com/t5/Wi-Fi-Bluetooth-for-Linux/Cypress-Linux-WiFi-Driver-Release-FMAC-2023-11-28/m-p/715365#M3024). My understanding was that they develop it for you and then you upstream it since you use that for your distro.