raspberrypi/firmware #1830: CM4S not supported by oldstable branch

johanradyn, 2023-10-01:
Is there any specific reason that the CM4 is supported in the oldstable and stable branches, but the CM4S is only supported in the stable branch?

pelwell, 2023-10-01:
Because the CM4S didn't exist when 5.10 was the mainline kernel, and by definition the stable and oldstable branches only receive bug fixes, not new features.

pelwell, 2023-10-01:
Both uname outputs are correct in that both reflect the kernel which is running. With newer Bullseye releases we choose to load the 64-bit kernel by default on devices based on BCM2711 because it runs better overall. If you prefer to use the old, BCM2711-specific 32-bit kernel then add `arm_64bit=0` to config.txt.

closed 2023-10-31 by johanradyn