raspberrypi/firmware #1833: Can't boot Pi 5 via NVMe behind PCIe switch / bridge

geerlingguy, 2023-10-17:
**Describe the bug**

I am unable to boot a Raspberry Pi 5 from an external NVMe SSD if used behind a PCIe switch (e.g. not as the root device on the external connection).

**To reproduce**

  1. Ensure you have an NVMe SSD that can boot the Raspberry Pi 5 when connected directly. (`BOOT_ORDER=0xf25416`)
  1. Connect a PCI Express switch to the external PCIe connection on the Raspberry Pi 5.
  1. Move the NVMe SSD you previously used to boot the Pi 5 to a port behind the PCIe switch.
  1. Attempt to boot the Raspberry Pi 5 off the NVMe SSD.

**Expected behaviour**

I would expect the NVMe SSD to be selected for boot, wherever it is enumerated on the PCIe bus.

**Actual behaviour**

The Raspberry Pi 5 bootloader attempts to load `nvme` but fails, likely due to it only enumerating devices directly attached to the external port, and not walking down the tree of any other connected PCIe bridges...

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
* Which OS and version (`cat /etc/rpi-issue`)?
* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)?

**Logs**

<details>
<summary>Click to expand full bootloader log (captured via UART)</summary>
```
RPi: BOOTSYS release VERSION:9d494316 DATE: 2023/09/13 TIME: 11:37:06
BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1694601426 serial e3ae743b boardrev d04170 stc 1604502
AON_RESET: 00000003 PM_RSTS 00001000
RP1_BOOT chip ID: 0x20001927
PM_RSTS: 0x00001000
part 00000000 reset_info 00000000
PMIC reset-event 00000000 rtc 65299da7 alarm 00000000 enabled 0
uSD voltage 3.3V
Initialising SDRAM 'Micron' 32Gb x2 total-size: 64 Gbit 4267
DDR 4267 1 0 64 152
RP1_BOOT chip ID: 0x20001927

RP1_BOOT chip ID: 0x20001927
RP1_BOOT: fw size 25968
PCI2 init
PCI2 reset
PCIe scan 00001de4:00000001
RP1_CHIP_INFO 20001927

RPi: BOOTLOADER release VERSION:9d494316 DATE: 2023/09/13 TIME: 11:37:06
BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1694601426 serial e3ae743b boardrev d04170 stc 4469863
AON_RESET: 00000003 PM_RSTS 00001000
 status
USB_PD CONFIG 0 43
Boot mode: NVME (06) order f41
PCI1 init
PCI1 reset
USB-PD: src-cap PDO object1 0x0a0191f4
Current 5000 mA
Voltage 5000 mV
USB-PD: src-cap PDO object2 0x0002d12c
Current 3000 mA
Voltage 9000 mV
USB-PD: src-cap PDO object3 0x0003c0e1
Current 2250 mA
Voltage 12000 mV
USB-PD: src-cap PDO object4 0x0004b0b4
Current 1800 mA
Voltage 15000 mV
Failed to open device: 'nvme'
Retry NVME 1
Failed to open device: 'nvme'
Boot mode: SD (01) order f4
SD HOST: 200000000 CTL0: 0x00800f00 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
SD HOST: 200000000 CTL0: 0x00800f00 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
EMMC
SD retry 1 oc 0
SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
SD retry 2 oc 0
SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
SDV1
SD CMD: 0x371a0010 (55) 0x0 0x1fff0001
Failed to open device: 'sdcard' (cmd 371a0010 status 1fff0001)
Retry SD 1
SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
SD HOST: 200000000 CTL0: 0x00800f00 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
EMMC
SD retry 1 oc 0
SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
SD retry 2 oc 0
SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
SDV1
SD CMD: 0x371a0010 (55) 0x0 0x1fff0001
Failed to open device: 'sdcard' (cmd 371a0010 status 1fff0001)
Boot mode: USB-MSD (04) order f
XHCI-STOP
xHC0 ver: 0 HCS: 00000085 00000085 00000085 HCC: 00000085
USBSTS 85
xHC0 ver: 0 HCS: 00000085 00000085 00000085 HCC: 00000085
xHC0 ports 0 slots 133 intrs 0
USB xHC init failed
Boot mode: RESTART (0f) order 0
Restart 0 max -1
Boot mode: NVME (06) order f41
PCI1 init
PCI1 reset
Failed to open device: 'nvme'
Retry NVME 1
Failed to open device: 'nvme'
```
</details>

**Additional context**

When behind the bridge, here is the hierarchy according to `lspci`:

```
pi@pi5:~ $ lspci -tv
-+-[0000:00]---00.0-[01-04]----00.0-[02-04]--+-01.0-[03]--
 |                                           \-02.0-[04]----00.0  KIOXIA Corporation Device 0010
 \-[0001:00]---00.0-[01]----00.0  Device 1de4:0001
```

I know for the Compute Module 4, the concern was a [lack of space in the bootloader](https://github.com/raspberrypi/firmware/issues/1684) to successfully enumerate all PCIe devices, no matter where they are on the bus. Does the Pi 5's bootloader overcome that limitation? I don't expect this to work on launch day, but it is something I think a _lot_ of people would like to do (e.g. stack an 'NVMe + 2.5G Ethernet' HAT, or 'NVMe + WiFi 7' HAT, etc. on top).

timg236, 2023-10-17:
It's not supported right now

peterharperuk, 2023-10-17:
What "PCIe switch" do you have?

geerlingguy, 2023-10-21:
@peterharperuk - I'm testing with [this I/O Crest switch](https://pipci.jeffgeerling.com/cards_adapter/iocrest-si-pex60016-2-port-pcie-switch.html):

```
pi@pi5:~ $ lspci
0000:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries Device 2712 (rev 21)
0000:01:00.0 PCI bridge: Pericom Semiconductor PI7C9X2G304 EL/SL PCIe2 3-Port/4-Lane Packet Switch (rev 05)
0000:02:01.0 PCI bridge: Pericom Semiconductor PI7C9X2G304 EL/SL PCIe2 3-Port/4-Lane Packet Switch (rev 05)
0000:02:02.0 PCI bridge: Pericom Semiconductor PI7C9X2G304 EL/SL PCIe2 3-Port/4-Lane Packet Switch (rev 05)
0000:04:00.0 Non-Volatile memory controller: KIOXIA Corporation Device 0010 (rev 01)
0001:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries Device 2712 (rev 21)
0001:01:00.0 Ethernet controller: Device 1de4:0001
```

cnxsoft, 2024-02-04:
I've just come across a HAT+ (Geekworm X1004) with an ASM1182e PCIe switch for two SSDs and the manufacturer claims it can't be used for booting at this time. It looks related.

geerlingguy, 2024-02-04:
@cnxsoft - It is most definitely related! And thanks for you post on that board, I really do hope Raspberry Pi can support at least a few switches (the asmedia ones seem extremely popular for smaller companies. I don't see Pericom that often).

ghakfoort, 2024-02-07:
I heard the guys from Pineberry are also already experimenting with a hat that supports 2 m.2 devices...  (no promises though.. ) 2 NVME drives or 1 NVME drive and room for a pcie coral TPU would be nice but of course we need to be able to boot from it too if a pcie switch is involved  :)

pbrueckner, 2024-02-19:
Similar problem often (50%) hangs after loading start4(db).elf and loading kernel:

CM4 with pericom switch:
```
lspci -tv
-[0000:00]---00.0-[01-05]----00.0-[02-05]--+-01.0-[03-04]----00.0-[04]----04.0  PLX Technology, Inc. PCI9056 32-bit 66MHz PCI <-> IOBus Bridge
                                           \-02.0-[05]--
lspci -v
00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2711 PCIe Bridge (rev 20) (prog-if 00 [Normal decode])
        Device tree node: /sys/firmware/devicetree/base/scb/pcie@7d500000/pci@0,0
        Flags: bus master, fast devsel, latency 0, IRQ 30
        Bus: primary=00, secondary=01, subordinate=05, sec-latency=0
        Memory behind bridge: 00000000-000fffff [size=1M] [32-bit]
        Prefetchable memory behind bridge: [disabled] [64-bit]
        Capabilities: <access denied>
        Kernel driver in use: pcieport

01:00.0 PCI bridge: Pericom Semiconductor Device b304 (rev 01) (prog-if 00 [Normal decode])
        Flags: bus master, fast devsel, latency 0, IRQ 31
        Bus: primary=01, secondary=02, subordinate=05, sec-latency=0
        I/O behind bridge: [disabled] [32-bit]
        Memory behind bridge: 00000000-000fffff [size=1M] [32-bit]
        Prefetchable memory behind bridge: [disabled] [64-bit]
        Capabilities: <access denied>
        Kernel driver in use: pcieport

02:01.0 PCI bridge: Pericom Semiconductor Device b304 (rev 01) (prog-if 00 [Normal decode])
        Flags: bus master, fast devsel, latency 0
        Bus: primary=02, secondary=03, subordinate=04, sec-latency=0
        I/O behind bridge: [disabled] [32-bit]
        Memory behind bridge: 00000000-000fffff [size=1M] [32-bit]
        Prefetchable memory behind bridge: [disabled] [64-bit]
        Capabilities: <access denied>

02:02.0 PCI bridge: Pericom Semiconductor Device b304 (rev 01) (prog-if 00 [Normal decode])
        Flags: fast devsel
        Bus: primary=02, secondary=05, subordinate=05, sec-latency=0
        I/O behind bridge: [disabled] [32-bit]
        Memory behind bridge: [disabled] [32-bit]
        Prefetchable memory behind bridge: [disabled] [64-bit]
        Capabilities: <access denied>

03:00.0 PCI bridge: PLX Technology, Inc. PEX 8111 PCI Express-to-PCI Bridge (rev 21) (prog-if 00 [Normal decode])
        Flags: bus master, fast devsel, latency 0, IRQ 32
        Bus: primary=03, secondary=04, subordinate=04, sec-latency=0
        I/O behind bridge: [disabled] [16-bit]
        Memory behind bridge: 00000000-000fffff [size=1M] [32-bit]
        Prefetchable memory behind bridge: [disabled] [32-bit]
        Capabilities: <access denied>

04:04.0 Bridge: PLX Technology, Inc. PCI9056 32-bit 66MHz PCI <-> IOBus Bridge
        Subsystem: DResearch Digital Media Systems GmbH PCI9056 32-bit 66MHz PCI <-> IOBus Bridge
        Flags: bus master, 66MHz, medium devsel, latency 64, IRQ 32
        Memory at 600000000 (32-bit, non-prefetchable) [size=512]
        Memory at 600000200 (32-bit, non-prefetchable) [size=256]
        Memory at 600000300 (32-bit, non-prefetchable) [size=16]
        Expansion ROM at <unassigned>
        Capabilities: <access denied>
        Kernel driver in use: xadio

```
logile:

```
RPi: BOOTLOADER release VERSION:8ba17717 DATE: 2023/01/11 TIME: 17:40:52
BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1673458852 serial 18f6edfd boardrev d03141 stc 476493
PM_RSTS: 0x00000020
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Micron' 32Gb x2 total-size: 64 Gbit 3200
[... 623 lines cut ...]

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1
dtoverlay=pcie-32bit-dma
dtoverlay=disable-bt
dtoverlay=disable-wifi
uart_2ndstage=1
dtparam=watchdog=on
dtparam=spi=on
dtoverlay=audremap
dtoverlay=spi-gpio40-45

[all]

enable_uart=1
start_debug=1
```

-------------------------------------- snap -------------------------------------------------------------

timg236, 2024-02-19:
Removed off topic comment about a CM4. This issue is specifically about Pi5 / BCM2712

Ronald1817, 2024-02-24:
As per a coincedence managed to instsall Ubuntu 23.10 on rpi 5 , with a NEO 5 NVME case.
Wrote the ubuntu image with NVME USB docking station , but woult not boot from it.
So i made a RPI OS  x64 usb stick and did de install drill and when rebooted it did enumerate the VNME and began to boot from NVME and the UBUNTU installer was started, id did the install procedure and after that , the Rpi 5 booted Ubuntu 23.10 from NVME SSD, was little bit surprised to see this happening this way, but it works.

Ronald1817, 2024-02-24:
played around little bit and what goes wrong is dtparam=pciex1_1=gen3 , then the nvme is not found anymore

pelwell, 2024-02-24:
That's covered in the documentation. From https://www.raspberrypi.com/documentation/computers/raspberry-pi-5.html#pcie-gen-3-0:
> The Raspberry Pi 5 is not certified for Gen 3.0 speeds, and connections to PCIe devices at these speeds may be unstable.

If Gen 3 was guaranteed to be stable then it would be enabled by default.

timg236, 2024-02-24:
This Issue about PCIe switches not assorted interop questions - please stay on topic

Ronald1817, 2024-02-24:
Don’t understand why you go this way

I don’t use pcie switch it’s directly connected to the pcie connector of the pi.

But  ok , I will look elsewhere …. No problem

geerlingguy, 2024-02-24:
@Ronald1817 - The title of this GitHub issue is "Can't boot Pi 5 via NVMe behind PCIe switch / bridge" and is meant to discuss that topic, not general NVMe SSD issues you may be having. The Raspberry Pi forums, or the manufacturer of the HAT / Bottom you're using, would be a better avenue for questions about Pi boot issues with NVMe.

That is all that is implied by @timg236—this GitHub repository is for issues pertaining to the Pi firmware, and this issue is about a specific feature request. These comments don't have anything to do with that, so typically the maintainers will clean them out at some point.

Ronald1817, 2024-02-24:
Hi,

 

I missed that , thought it was nvme in general .

 

Did you test the NEO 5 with NVME? , it’s quite good little case.

 

Regards

Ronald.

b0n541, 2024-03-09:
I have the same issue with the Geekworm X1003. I've tried both a K-M and a K-M&B connector SSD.

greyltc, 2024-03-15:
A new dual NVMe rpi5 product has appeared with a PCIe switch in it https://pimoroni.com/nvmeduo
Anyone know what the switch part number is?

geerlingguy, 2024-03-15:
@greyltc - Looks like `ASM1182e`. Same as the Geekworm—and Pineberry Pi!

@timg236 / @pelwell — it seems like by some stroke of luck, all the main vendors of PCIe addon boards have settled on the asmedia `ASM1182e` as their PCIe gen 2 switch of choice... would it be possible for that chip (at least for now) to be supported for switch traversal for NVMe boot?

Phlogistons, 2024-03-18:
https://pineberrypi.com/products/hatdrive-ai-coral-edge-tpu-bundle-nvme-2230-2242-gen-2-for-raspberry-pi-5
This new very sweet toy is also going to use the ASMedia switch, at this point I guess it's going to be the "standard" for most devices.

rushtoshankar, 2024-04-05:
Recently I have bought an PCIe HAT from aliexpress.com. The link is https://www.aliexpress.us/item/3256806347359812.html.
This has one 2.5Gb Eth and one NVM. This board has the 'ASMedia Technology Inc. ASM1182e 2-Port PCIe x1 Gen2 Packet Switch'. I have install an NVMe drive of 512G from my old laptop. I had issues booting it from. Then switched back to SD card boot using EEPROM recovery image.

But I found two interesting things:
- First, I was able to use NVMe for OS
- Second, I was able to fully boot from NVMe (without setting the boot order to NVMe but with the help of SD card)

Let me explain below what I did, I didn't go in depth to find out how it worked. But at a high level,
- To make bootable NVMe, instead of installing from the scratch in the NVMe, I just did a 'dd' from SD card to NVMe (This copied the UUIDs as well)
- Then changed the boot order to NVMe and removed the SD card
- First many attempts to boot from NVMe without SD card didn't work.
- Next, I have updated the boot order back to SD card and used my original SD card (which was used to duplicate to NVMe)
- Then it booted, I thought it booted from SD card.
But when I checked, the mounted partitions, I saw, the /boot/firmware was from SD card and the '/' (root) partition was from my NVMe. So basically, the OS was running on NMVe.

I didn't know what I did except expanding the root partition size to utilize the full NMVe (512G) size, (SD card was 128G), I resized the partition in NMVe. While preparing to answer this, I found that the below in lsblk command

mmcblk0          179:0    0 116.2G  0 disk
├─mmcblk0p1 179:1    0   512M  0 part
└─mmcblk0p2 179:2    0 115.7G  0 part
nvme0n1          259:0    0 476.9G  0 disk
├─nvme0n1p1 259:1    0   512M  0 part /boot/firmware
└─nvme0n1p2 259:2    0 476.4G  0 part /

Both boot and root partitions are used from NVMe only. Now this is running as my home router / gateway. So, I can't do much tests immediately. But I will try to reboot multiple times/ try without SD card and see what's happening when I get a chance and post it here.

mahtin, 2024-04-06:
Can I please add the [ASM1184e](https://www.asmedia.com.tw/product/556yQ9dSX7gP9Tuf/b7FyQBCxz2URbzg0) PCIe packet switch chip to this issue? It shows up on the [ZS ZHISHANG PCI-E X1 to 4 PCI-E X16 Expansion Riser](https://pipci.jeffgeerling.com//cards_adapter/zs-zhishang-pcie-x1-to-4-pcie-x16-expansion-riser.html) as documented by @geerlingguy.

The ASM1182e is mentioned above. It's a 2 by PCIe chip vs the ASM1184e which is 4 by PCIe chip; so similar(ish).

Boards available on Amazon via:
-   https://www.amazon.com/dp/B075DF9L5V (available)
-   https://www.amazon.com/dp/B07DDFXV99 (presently unavailable) - the one I purchased.
-   https://www.amazon.com/dp/B07KQ4D7ZS (available)

I will note that this PCIe bridge works fine once the system is booted.

EnziinSystem, 2024-04-07:
> As per a coincedence managed to instsall Ubuntu 23.10 on rpi 5 , with a NEO 5 NVME case. Wrote the ubuntu image with NVME USB docking station , but woult not boot from it. So i made a RPI OS x64 usb stick and did de install drill and when rebooted it did enumerate the VNME and began to boot from NVME and the UBUNTU installer was started, id did the install procedure and after that , the Rpi 5 booted Ubuntu 23.10 from NVME SSD, was little bit surprised to see this happening this way, but it works.
> 
> i think because of the rpi os installer , the nvme was enumerated and could be found en the pi could boot from it.

Hello,

Does this mean you can boot Ubuntu from the SSD card and also use the TPU afterwards?

We are talking about this product:

https://pineberrypi.com/products/hatdrive-ai-coral-edge-tpu-bundle-nvme-2230-2242-gen-2-for-raspberry-pi-5

The goal is to be able to boot the operating system from the SSD and use the TPU.

mikegapinski, 2024-04-11:
It looks like you can boot from the switch with https://github.com/worproject/rpi5-uefi

You still have to place the 2nd bootloader on the SD Card but it works. We'll do some testing internally and update the documentation for the HatDrive with a workaround that we'll use until this issue is solved in the firmware

timg236, 2024-04-12:
> @greyltc - Looks like `ASM1182e`. Same as the Geekworm—and Pineberry Pi!
>
> @timg236 / @pelwell — it seems like by some stroke of luck, all the main vendors of PCIe addon boards have settled on the [...]

In principle any switch follows the spec should work from a software perspective. Supporting / debugging individual switch/PCIe devices would be up to the HAT designer i.e. debugging low level electronic issues.

mikegapinski, 2024-04-12:
> > @greyltc - Looks like `ASM1182e`. Same as the Geekworm—and Pineberry Pi!
> 
> > 
> 
> > @timg236 / @pelwell — it seems like by some stroke of luck, all the main vendors of PCIe addon boards have settled on the asmedia `ASM1182e` as their PCIe gen 2 switch of choice... would it be possible for that chip (at least for now) to be supported for switch traversal for NVMe boot?
> 
> 
> 
> In principle any switch that follows the spec should work with generic firmware (once written). Supporting / debugging individual switch/PCIe devices would be up to the HAT designer i.e. debugging low level electronic issues.

That's exactly how it should be. Other switches also work but ASM1182e is perfect for the Pi 5. That's why all the manufacturers use it now. 4 port variants tend to have issues with 3 or more devices. It was a more pronounced issue on the CM4 but you can get PCIe errors once you start utilising all the devices at once (RAID etc). Having 4 ports on a single lane is also not good from the performance perspective. The Gen 3 switch doesn't help much too, the difference is small when divided between all the devices. It is also much more expensive.

We won't be touching the 4 port switch anytime soon, we are not happy with the samples. If someone wants to push the bus for science: we've successfully connected more than 4 PCIe devices at once with the HatBRICK! Commander.

timg236, 2024-04-12:
To be clear, my obeys the spec I mean that anything that requires a ton of quirks an interop fixes just won't be supported. No immediate plans to do this though

mikegapinski, 2024-04-12:
> Go easy with the adverts...

Wasn’t trying to plug anything, just geeking out over the tech specs a bit too much, maybe. There are no other solutions like this so it was almost impossible to keep the name of the board out of the discussion. Thanks for the nudge!

Going back to the topic. If someone desperately needs to boot from NVMe/SATA behind the switch it works with both the UEFI approach and uboot(it applies to the CM4 too). Having to place a secondary bootloader on the SD card is not perfect but no changes are required to the OS itself and this is a huge plus.

timg236, 2024-04-12:
> Wasn’t trying to plug anything, just geeking out over the tech specs a bit too much, maybe. There are no other solutions like this so it was almost impossible to keep the name of the board out of [...]

Agreed. The intention would be to support booting from an NVMe driver behind a switch, probably only simple hardware topologies would be supported

geerlingguy, 2024-04-12:
@timg236 - The other use case I have is to have multiple boot drives in a compact solution, so a Pi could boot into different OSes more easily than with a bunch of USB stuff hanging off it. That... would be interesting to do from the Pi OS default firmware experience, though a UEFI bootloader could have boot order priority.

I don't know the default way the bootloader picks which device to boot from first, if there are multiple on one bus like USB. Is it by device ID?

timg236, 2024-04-12:
The bootloader enumerates MSD in parallel and you get the first one that looks like valid OS. It’s  mentioned in the msd docs.
The design goal is to do the right thing based on hardware detection and sensible defaults  without ever inflicting a PC BIOS style UI on the end user. That said there’s scope for advanced users to have more flexibility eg via boot loader config specified GUIDs or ids. It very much a second order feature

EnziinSystem, 2024-04-23:
> Wasn’t trying to plug anything, just geeking out over the tech specs a bit too much, maybe. There are no other solutions like this so it was almost impossible to keep the name of the board out of [...]

Can you tell me specifically how to do it?

I'm having an NVMe drive behind an ASMedia hub like most people.

How to boot from it?

mikegapinski, 2024-04-23:
> Can you tell me specifically how to do it?
>
>
>
> I'm having an NVMe drive behind an ASMedia hub like most people.
>
>
>
> How to boot from it?

I will have a gist for this up after the weekend. 

In a nutshell you need the uefi on the sdcard and grub on the boot partion that has raspbian. I just want to automate everything and make it very easy to get running. (flash with rpi imager, run one command and that's it). We're also testing other operating systems

Supporting people with regular NVMe stuff in the past few months has though me where most of the users struggle the most.

timg236, 2024-04-23:
This issue is to track the development of native support for PCIe switches in the Raspberry Pi5 bootloader. When this is released we'll tag this bug so feel free to subscribe to this to get notified about early software release ... which will be via rpi-update.

I've deleted all the off topic comments about general NVMe support / booting from from EMMC first. 

The best place to discuss interop / workarounds or things that aren't directly related to the software change is on the Raspberry Pi forums e.g. Advanced Users.

peterharperuk, 2024-04-26:
```
$ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
nvme0n1     259:0    0 238.5G  0 disk 
├─nvme0n1p1 259:1    0   512M  0 part /boot/firmware
└─nvme0n1p2 259:2    0   238G  0 part /
nvme1n1     259:3    0 232.9G  0 disk 
├─nvme1n1p1 259:4    0   512M  0 part 
└─nvme1n1p2 259:5    0 232.4G  0 part 
```

geerlingguy, 2024-04-26:
But hoooow? And how long until I can play with this :D

marcelgsteiger, 2024-04-27:
I have just installed a Pimoroni NVMe Base Duo w/two 500GB disks (ASM1182e). I updated the firmware to version Sat 20 Apr 2024 and then used the HTTP installer with no SD Card or USB disk installed. This worked just fine and installed the system on the NVMe disk of choice. Of course, this won't boot directly at present. I then prepared a USB stick using RPI-Imager, then in /boot/firmware/cmdline.txt I simply replaced the disk ID for the ROOT=PARTUUID=xxxxxxxxx- stanza with the partuuid value from the cmdline.txt file on the NVMe drive. This leads to the lsblk situation depicted by @peterharperuk. above.

The USB stick just serves to circumvent the bootloader's inability to boot in this scenario. But then, everything runs as if it were booted from NVMe. Once booted, the USB key can be removed to save power - as long as nobody tries to reboot. I can even set the boot order to 0xf46 (NVMe first, then USB, then loop).

Of course, I have to manually copy updates to the /boot/firmware partition from NVMe to the USB Key later on. For me,  this appears the most simple way to have a system installed on NVMe at this time.

EnziinSystem, 2024-04-27:
> I have just installed a Pimoroni NVMe Base Duo w/two 500GB disks (ASM1182e). I updated the firmware to version Sat 20 Apr 2024 and then used the HTTP installer with no SD Card or USB disk installed. [...]

Can I use an SD card instead of a bootable USB?

marcelgsteiger, 2024-04-27:
> Can I use an SD card instead of a bootable USB?

...I can't imagine why this would not work technically. I used a USB because it's virtually impossible to insert a SD card behind the PCIe connector cable. And, the SDcard cannot be removed once the system is powered on.

marcelgsteiger, 2024-04-27:
> But hoooow? And how long until I can play with this :D

...because off-topic here, I moved my answer to https://forums.raspberrypi.com/viewtopic.php?t=369796

vitormfgoncalves, 2024-05-01:
Has been 6 months since this issue was open, and yes, I am aware that I ca use an SD card to circunvent this problem.
However, circunventing is not a solution.
I wonder if anyone could point us to any work done on this issue, if any.

pelwell, 2024-05-01:
[ Deleted comment with a bad attitude ]

peterharperuk, 2024-05-13:
A change has been pushed that updates the bootloader. When booting via NVMe it tries to use a drive behind a bridge. It's been tested with asmedia and pericom. It starts at the first downstream bridge it finds (oddly for the geekwork X1004 this the disk marked SSD2). It should try each drive. You can change the first drive used by specifying NVME_CONTROLLER=x, where x=0 is the first drive, x=1 is the second drive. Let us know if it works.

You can update by running sudo rpi-update. I think the following might just update the bootloader? SKIP_KERNEL=1 SKIP_FIRMWARE=1 SKIP_SDK=1 SKIP_VCLIBS=1 && sudo rpi-update

geerlingguy, 2024-05-13:
@peterharperuk - ooh, going to test this out in a few minutes, I'll let you know how it goes! I have one of Pineboards' dual NVMe HATs I've been meaning to test anyway.

mikegapinski, 2024-05-13:
In my very quick testing it all works fine. It goes through the drives in order defined by a switch (the slots are listed on the schematic as 0,1,2,3) etc.

geerlingguy, 2024-05-13:
@mikegapinski - I'm having some issues with NVMe stability behind the bridge. But trying to figure out if it's the Pineboard I'm using, the drives, firmware, or something else. Debugging over in https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/633

I will also try with another PCIe switch to see if that works any better.

geerlingguy, 2024-05-13:
I've tried three different PCIe switch boards (all three from Pineboards, but two I can confirm worked without any issues on some other testing I've done), and two different NVMe SSDs (a Cytron MakerDisk 2242 and a Pineboard 2242), with one or two drives inserted, and also with a separate M.2 HAT I can confirm works fine without the switch in front of it.

In all cases, when the Pi tries mounting the filesystem, it starts erroring out (see linked issue above), and eventually may boot fully... but then any time there's more than a tiny amount of disk access, the Pi freezes after a bunch of NVMe I/O timeouts and bus resets.

I will also try booting off microSD with unformatted NVMe drives to see if it could be a power issue or something else. The drives don't seem to draw more than 1-2W each (at least not these drives), so I don't think it's overloading the FFC connection power draw limits (IIRC 5W).

geerlingguy, 2024-05-13:
@peterharperuk - One thing I noticed is if I have an unformatted NVMe drive installed behind the switch, with the latest firmware it takes maybe 10 seconds or so before it'll pop over to the microSD card and try booting off it. Is that expected behavior? If I go into `raspi-config` and try to switch to SD card first, I get `Current EEPROM version 2024-05-13 or newer not found - aborting`.

Here's what's output during boot, after this it waits about 10 seconds(?) and then starts booting off microSD.

```
PCIe scan 00001987:00005013
Found device on bridge 4 with vid 1987 enable 1
PCIe scan 00001b21:00001182
PCIe scan 00001987:00005013
VID 0x1987 MN SSD94-2563CG-PB                        
NVME on 0
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
Trying partition: 0
Unable to read partition as FAT
type: 32 lba: 0 '' '            ' clusters 0 (0)
rsc 0 fat-sectors 0 root dir cluster 0 sectors 0 entries 0
Select partition rsts 0 C(boot_partition) 0 EEPROM config 0 result 0
Trying partition: 0
Unable to read partition as FAT
type: 32 lba: 0 '' '            ' clusters 0 (0)
rsc 0 fat-sectors 0 root dir cluster 0 sectors 0 entries 0
Boot mode: USB-MSD (04) order f1
NVME off
```

peterharperuk, 2024-05-13:
No, I wouldn't expect there to be a 10s delay. That trace shows an attempt to boot from USB. Are you sure that's not where the time is going?

geerlingguy, 2024-05-13:
@vitormfgoncalves - Can you run some benchmarks, like run [this FIO test](https://www.jeffgeerling.com/blog/2024/quick-nvme-performance-testing-fio) and see if there are any issues when running it? Also what speed do you get?

vitormfgoncalves, 2024-05-13:
@geerlingguy Yes.
I'm using a generic Intel SSD at gen2 speeds. Nothing fancy.

```
nvme0: (g=0): rw=read, bs=(R) 256KiB-256KiB, (W) 256KiB-256KiB, (T) 256KiB-256KiB, ioengine=libaio, iodepth=32
fio-3.33
Starting 1 process

nvme0: (groupid=0, jobs=1): err= 0: pid=1310: Mon May 13 22:39:23 2024
  read: IOPS=1580, BW=395MiB/s (415MB/s)(11.6GiB/30021msec)
    slat (nsec): min=10574, max=27185, avg=10796.75, stdev=375.23
    clat (usec): min=20094, max=39183, avg=20226.36, stdev=282.13
     lat (usec): min=20111, max=39193, avg=20237.15, stdev=282.13
    clat percentiles (usec):
     |  1.00th=[20317],  5.00th=[20317], 10.00th=[20317], 20.00th=[20317],
     | 30.00th=[20317], 40.00th=[20317], 50.00th=[20317], 60.00th=[20317],
     | 70.00th=[20317], 80.00th=[20317], 90.00th=[20317], 95.00th=[20317],
     | 99.00th=[20317], 99.50th=[20317], 99.90th=[20579], 99.95th=[24511],
     | 99.99th=[36439]
   bw (  KiB/s): min=404480, max=405290, per=100.00%, avg=404952.10, stdev=158.66, samples=60
   iops        : min= 1580, max= 1583, avg=1581.83, stdev= 0.62, samples=60
  lat (msec)   : 50=100.07%
  cpu          : usr=0.45%, sys=2.14%, ctx=47482, majf=0, minf=9
  IO depths    : 1=0.0%, 2=0.0%, 4=0.0%, 8=0.0%, 16=0.0%, 32=100.0%, >=64=0.0%
     submit    : 0=0.0%, 4=100.0%, 8=0.0%, 16=0.0%, 32=0.0%, 64=0.0%, >=64=0.0%
     complete  : 0=0.0%, 4=100.0%, 8=0.0%, 16=0.0%, 32=0.1%, 64=0.0%, >=64=0.0%
     issued rwts: total=47451,0,0,0 short=0,0,0,0 dropped=0,0,0,0
     latency   : target=0, window=0, percentile=100.00%, depth=32

Run status group 0 (all jobs):
   READ: bw=395MiB/s (415MB/s), 395MiB/s-395MiB/s (415MB/s-415MB/s), io=11.6GiB (12.4GB), run=30021-30021msec

Disk stats (read/write):
  nvme0n1: ios=55184/13, merge=0/8, ticks=1119148/114, in_queue=1119283, util=99.74%
```

I hope this helps.

peterharperuk, 2024-05-14:
@geerlingguy The output of lspci would be interesting. To fully debug you devices I'd need the output of "sudo lspci -xxx" If you attach it as a file I'll try and take a look. If you see the same problems when booting from an SD card with rootfs on the NVMe disk then that would rule out your issues being caused by some side effect from the bootloader.

mikegapinski, 2024-05-14:
@peterharperuk it seems like the Phison E13 controller has aspm issues behind the switch. The latest firmware is dated August 2022, maybe they'll update it at some point. I tested all the _problematic_ drives from WD and they boot normally with ASM1182 and the latest firmware.

@geerlingguy instead of globally disabling aspm please use: dtparam=pciex1_no_l0s=on and it'll work fine.

```
mgapinski@nvmebootpi:~$ sudo fio nvme-read.fio
nvme0: (g=0): rw=read, bs=(R) 256KiB-256KiB, (W) 256KiB-256KiB, (T) 256KiB-256KiB, ioengine=libaio, iodepth=32
fio-3.33
Starting 1 process
Jobs: 1 (f=1): [R(1)][100.0%][r=430MiB/s][r=1718 IOPS][eta 00m:00s]
nvme0: (groupid=0, jobs=1): err= 0: pid=2040: Tue May 14 12:29:45 2024
  read: IOPS=1715, BW=429MiB/s (450MB/s)(12.6GiB/30019msec)
    slat (nsec): min=8463, max=58890, avg=10840.75, stdev=605.38
    clat (usec): min=9271, max=36096, avg=18632.93, stdev=407.18
     lat (usec): min=9297, max=36107, avg=18643.77, stdev=407.15
    clat percentiles (usec):
     |  1.00th=[17695],  5.00th=[18220], 10.00th=[18482], 20.00th=[18744],
     | 30.00th=[18744], 40.00th=[18744], 50.00th=[18744], 60.00th=[18744],
     | 70.00th=[18744], 80.00th=[18744], 90.00th=[18744], 95.00th=[19006],
     | 99.00th=[19792], 99.50th=[20055], 99.90th=[21365], 99.95th=[25035],
     | 99.99th=[33162]
   bw (  KiB/s): min=438272, max=441344, per=100.00%, avg=439603.37, stdev=500.65, samples=60
   iops        : min= 1712, max= 1724, avg=1717.15, stdev= 1.95, samples=60
  lat (msec)   : 10=0.01%, 20=99.43%, 50=0.62%
  cpu          : usr=0.33%, sys=2.47%, ctx=51429, majf=0, minf=9
  IO depths    : 1=0.0%, 2=0.0%, 4=0.0%, 8=0.0%, 16=0.0%, 32=100.0%, >=64=0.0%
     submit    : 0=0.0%, 4=100.0%, 8=0.0%, 16=0.0%, 32=0.0%, 64=0.0%, >=64=0.0%
     complete  : 0=0.0%, 4=100.0%, 8=0.0%, 16=0.0%, 32=0.1%, 64=0.0%, >=64=0.0%
     issued rwts: total=51505,0,0,0 short=0,0,0,0 dropped=0,0,0,0
     latency   : target=0, window=0, percentile=100.00%, depth=32

Run status group 0 (all jobs):
   READ: bw=429MiB/s (450MB/s), 429MiB/s-429MiB/s (450MB/s-450MB/s), io=12.6GiB (13.5GB), run=30019-30019msec

Disk stats (read/write):
  nvme0n1: ios=60086/35, merge=0/14, ticks=1119131/482, in_queue=1119622, util=99.75%
mgapinski@nvmebootpi:~$ lspci -vvvv
0000:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries Device 2712 (rev 21) (prog-if 00 [Normal decode])
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Interrupt: pin A routed to IRQ 38
        Bus: primary=00, secondary=01, subordinate=04, sec-latency=0
        Memory behind bridge: 00000000-000fffff [size=1M] [32-bit]
        Prefetchable memory behind bridge: 00000000fff00000-00000000000fffff [disabled] [64-bit]
        Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort+ <SERR- <PERR-
        BridgeCtl: Parity- SERR- NoISA- VGA- VGA16- MAbort- >Reset- FastB2B-
                PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
        Capabilities: <access denied>
        Kernel driver in use: pcieport

0000:01:00.0 PCI bridge: ASMedia Technology Inc. ASM1182e 2-Port PCIe x1 Gen2 Packet Switch (prog-if 00 [Normal decode])
        Subsystem: ASMedia Technology Inc. ASM1182e 2-Port PCIe x1 Gen2 Packet Switch
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Interrupt: pin A routed to IRQ 38
        Bus: primary=01, secondary=02, subordinate=04, sec-latency=0
        I/O behind bridge: 0000f000-00000fff [disabled] [32-bit]
        Memory behind bridge: 00000000-000fffff [size=1M] [32-bit]
        Prefetchable memory behind bridge: 00000000fff00000-00000000000fffff [disabled] [64-bit]
        Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
        BridgeCtl: Parity- SERR+ NoISA- VGA- VGA16- MAbort- >Reset- FastB2B-
                PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
        Capabilities: <access denied>
        Kernel driver in use: pcieport

0000:02:03.0 PCI bridge: ASMedia Technology Inc. ASM1182e 2-Port PCIe x1 Gen2 Packet Switch (prog-if 00 [Normal decode])
        Subsystem: ASMedia Technology Inc. ASM1182e 2-Port PCIe x1 Gen2 Packet Switch
        Control: I/O- Mem- BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
[... 39 lines cut ...]
0001:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries Device 2712 (rev 21) (prog-if 00 [Normal decode])
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Interrupt: pin A routed to IRQ 47
        Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
        Memory behind bridge: 00000000-005fffff [size=6M] [32-bit]
        Prefetchable memory behind bridge: 00000000fff00000-00000000000fffff [disabled] [64-bit]
        Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
        BridgeCtl: Parity- SERR- NoISA- VGA- VGA16- MAbort- >Reset- FastB2B-
[... cut ...]
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Interrupt: pin A routed to IRQ 47
        Region 0: Memory at 1f00410000 (32-bit, non-prefetchable) [size=16K]
        Region 1: Memory at 1f00000000 (32-bit, non-prefetchable) [virtual] [size=4M]
        Region 2: Memory at 1f00400000 (32-bit, non-prefetchable) [size=64K]
        Capabilities: <access denied>
        Kernel driver in use: rp1
```

geerlingguy, 2024-05-14:
@mikegapinski - Ah, thanks. I have edited `/boot/firmware/config.txt` and added `dtparam=pciex1_no_l0s=on` to the end, and rebooted.

Everything seems to be working fine now, and my impromptu benchmark gives:

```
Run status group 0 (all jobs):
   READ: bw=435MiB/s (456MB/s), 435MiB/s-435MiB/s (456MB/s-456MB/s), io=12.8GiB (13.7GB), run=30019-30019msec
```

So getting PCIe Gen 2 without issue now. Thanks!

Now the race begins to figure out which drives work better than others out of the box behind a switch :D

mikegapinski, 2024-05-15:
> @mikegapinski - Ah, thanks. I have edited `/boot/firmware/config.txt` and added `dtparam=pciex1_no_l0s=on` to the end, and rebooted.
>
>
>
> Everything seems to be working fine now, and my impromptu [...]

I can think of at least 5 SKUs that throw errors only behind a switch (Not only Phison but also Maxio controllers). 

@peterharperuk at least initially pciex1_no_l0s could be set by default when you boot behind a switch. Having to remove a drive and put it in the enclosure to set it up after NVMe boot is enabled is not ideal.

I don't know why ASPM errors on certain drives are only thrown behind a switch, maybe it could be fixed in the firmware update.

timg236, 2024-05-15:
Changing global default to workaround drive specific issues is generally the wrong thing to do.  It's possible to set these as board level options via the bootloader flash config.

https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#config_txt

```
[config.txt]
dtparam=pciex1_no_l0s=on
```

EnziinSystem, 2024-05-15:
@timg236 

What you mean is that instead of adding "pcie_aspm=off" to "/boot/firmware/cmdline.txt" we should:

```
sudo nano /boot/firmware/config.txt

[all]
# Enable the PCIe External connector.
dtparam=pciex1
dtparam=pciex1_gen=3
dtparam=pciex1_no_l0s=on
kernel=kernel8.img
```

Is the above method correct?

mikegapinski, 2024-05-15:
@EnziinSystem 

dtparam=pciex1_gen=3 - does not make sense with a switch, those are mostly Gen 2
kernel=kernel8.img - this is only needed for Coral and other devices that need the 4K kernel

EnziinSystem, 2024-05-15:
> @EnziinSystem
> 
> dtparam=pciex1_gen=3 - does not make sense with a switch, those are mostly Gen 2 kernel=kernel8.img - this is only needed for Coral and other devices that need the 4K kernel

OK, we are interested in this configuration.

```
sudo nano /boot/firmware/config.txt

[all]
# Enable the PCIe External connector.
dtparam=pciex1
dtparam=pciex1_gen=2
dtparam=pciex1_no_l0s=on
```

timg236, 2024-05-15:
If the HAT meets the M2.HAT spec or the system has booted from NVMe then these lines are redundant. 
```
dtparam=pciex1
dtparam=pciex1_gen=2
```

peterharperuk, 2024-05-15:
I don't have any drives that suffer from this problem and I've got the cheapest drives I could find from Amazon. Where can I get these problematic drives from?

[32 more replies]

closed 2024-10-23 by peterharperuk