raspberrypi/firmware #1837: RPi5: wlan0 doesn't exist when Boot Order is USB->SDCARD

rh1972, 2023-10-28:
Using latest bookworm image, via the RPi Imager, the wlan0 interface only exists when Boot Order is SDCARD->USB.
When changing this to USB->SDCARD, it doesn't appear in "ifconfig", nor "rfkill list" and WLAN Locale is greyed out.
Changing back to SDCARD->USB restores the wlan0 interface after reboot.

Reproduce:

1. Burn image to SSD drive via RPi official imager, using 64bit full image.
2. wlan0 exists, if Boot Order is SDCARD->USB
3. Change Boot Order to USB->SDCARD, wlan0 doesn't exist after boot.
4. Changing it back to SDCARD->USB, wlan0 exists after boot.

pelwell, 2023-10-29:
Is it the boot order or the boot medium that matters? You haven't said what happens if you have SD card first but no valid SD image, or USB first and no USB image.

rh1972, 2023-10-29:
Fair point. My issue above is always with no sdcard in the slot. I'll do some more testing. ---- On Sun,29 Oct 2023 15:05:20 +0000  ***@***.***  wrote ----
Is it the boot order or the boot medium that matters? You haven't said what happens if you have SD card first but no valid SD image, or USB first and no USB image.

rh1972, 2023-10-29:
OK, did some more testing. At least with my setup, wlan0 is never created under the following conditions:

1. Boot Order is USB->SDCARD
2. USB drive is attached (with or without sdcard being present)

Happy to provide some logs, but would need some guidance.

pelwell, 2023-10-29:
Thanks. The common link between WiFi and SD cards is SD(IO), so my guess is that there is a common PLL that needs to be enabled or magic register but that must be set in order for WiFi to work, and that by skipping the SD card initialisation the bootloader is also skipping that necessary setup.

peterharperuk, 2023-10-30:
Yes - I can repro this.

pelwell, 2023-10-30:
The cause has been determined and a fix created. Expect a bootloader update in the near future.

timg236, 2023-10-30:
It should be resolved by this PR

https://github.com/raspberrypi/rpi-eeprom/pull/494

If you are able to installed the EEPROM directly and verify that would be great
`
wget https://github.com/timg236/rpi-eeprom/raw/pieeprom-2023-10-30-2712/firmware-2712/default/pieeprom-2023-10-30.bin
sudo rpi-eeprom-config --edit pieeprom-2023-10-30.bin
`

rh1972, 2023-10-30:
EEPROM was updated

```
Updating bootloader EEPROM
Reading EEPROM: 2097152 bytes 0x3c960000
1363ms
Writing EEPROM
..................................................................................................................................................................................
..................................................................................................................................................................................
...........................................................................................................................................................+
151ms

Verify BOOT EEPROM
Reading EEPROM: 2097152 bytes 0x3c960000
1363ms
BOOT-EEPROM: UPDATED
EEPROMs updated. Rebooting
```

but I get this when changing the boot order to USB->SDCARD in raspi-config:

`No EEPROM bin file found for version 2023-10-30 -  aborting `

Am I missing a step?

timg236, 2023-10-30:
Yes, you are installing an 'out-of-package' EEPROM image so raspi-config / APT won't know about it.

If you do "rpi-eeprom-config --edit" and set the BOOT_ORDER to 0xf14 in the popup editor then sudo reboot it will have the same effect

rh1972, 2023-10-30:
OK, done. But no wlan0, I'm afraid.
Change dit back to SDCARD->USB, and wlan0 appears.

timg236, 2023-10-30:
Please can you run the following commands in the bad-state? 

`
vcgencmd bootloader_version
vcgencmd bootloader_config
`

rh1972, 2023-10-30:
Weird. I re-flashed the EEPROM and it works now:
```
RPi: BOOTLOADER release VERSION:30de0ba5 DATE: 2023/10/30 TIME: 16:45:10
BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1698684310 serial 792ffc3b boardrev d04170 stc 4596606
AON_RESET: 00000003 PM_RSTS 00001000
usb_pd_init status 3
USB_PD CONFIG 0 41
Boot mode: USB-MSD (04) order f1
```
And, I see wlan0 - which is the main thing.

timg236, 2023-10-30:
Thanks for confirming. If you ran raspi-config as well as rpi-eeprom-update then one change might have overridden the other. I've merged the WLAN fix so it should be come part of the normal APT package soon

geerlingguy, 2023-11-09:
I wonder if this is the same bug that was causing me to not get WiFi when I was booting off NVMe? I had boot order: `BOOT_ORDER=0xf416`

pelwell, 2023-11-09:
Yes, that's the scenario for which the fix was created.

timg236, 2023-11-09:
Resolved by https://github.com/raspberrypi/rpi-eeprom/commit/aded0825e3a2b6108c5bba7e0d823ddcdeb0d2d3

closed 2023-11-09 by pelwell

mrg9999, 2023-11-24:
Not wanting to add a red herring. 2023-10-10 Bookworm current OS

I had NFR RPI 5 4GB a few months ago and it worked fine any boot order only ever used USB device. It may have been modified by multiple people before me and to my shame I did not check the prom version.

This week I purchased a retail RPI 5 4GB and 8GB from the Pi Store in Cambridge, UK. 
I haven't used a MicroSD card for years
bootloader shows
$ sudo rpi-eeprom-update -a
BOOTLOADER: up to date
   CURRENT: Thu 28 Sep 10:24:57 UTC 2023 (1695896697)
    LATEST: Thu 28 Sep 10:24:57 UTC 2023 (1695896697)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader-2712/default)

4GB Wi-Fi works when boot set to USB > SD
8GB Wi-Fi Is not detected then USB >SD. changed backed to SD >USB and Wi-Fi fine.

pelwell, 2023-11-24:
Look at the dates - your EEPROM image is too old to include the fix mentioned above. Update them both (`sudo apt update && sudo apt install rpi-eeprom`) to grab the latest images.

badokami, 2024-01-14:
I'm still having problems with my PI5 (8GB). I created a bootable USB and boot it up on a PI4 (4GB) and wifi works as expected. When I boot the same USB on my PI5, I can see the router SSID but cannot connect to it. I was using the Dec/6 firmware, which I upgraded to 5/Jan but still no joy. I've switched the boot order back from USB first to SD first but that hasn't helped either. I'm at a loss now?

badokami, 2024-01-14:
Oops, for got to add both the PI4 and PI5 are using their respective stock power adapted. Sorry!

badokami, 2024-01-15:
Additional information: 

I thought the problem might be power related so I added a self-powered USB hub but it's not made any difference. I have discovered that booting does work on the USB2 ports but not the USB3 ports. Hopefully I'm not the only one who still has this problem and it can be fixed in some future firmware update.

timg236, 2024-01-15:
@badokami That sounds more like RF interference from USB3. This bug addressed an issue where the WiFI would not receive a clock and would therefore not work at all on USB2 or USB3. It would not be possible to see the SSID with the original firmware.

badokami, 2024-01-16:
> @badokami That sounds more like RF interference from USB3. This bug addressed an issue where the WiFI would not receive a clock and would therefore not work at all on USB2 or USB3. It would not be possible to see the SSID with the original firmware.

I'm not so sure about the RF interference angle... I have a 2"ft USB3 Male to Female extension cable which allowed me to move the various USB3 storage devices and a 2.5" spinner I tested, well away from the PI5 with the same results. The system consistently boot's with working WiFi if USB2 ports are used but no joy with USB3. I guess I'm just S.O.L for the moment, until RPI foundation releases another version of the board. Disappointing to say the least.

f0c, 2024-01-21:
> I wonder if this is the same bug that was causing me to not get WiFi when I was booting off NVMe? I had boot order: `BOOT_ORDER=0xf416`

When I was boot Raspberry pi 5 with nvme, I get the same trouble. I try many boot sequence, when I change to BOOT_ORDER=0xf61, wifi is working normally.