raspberrypi/firmware #1838: Black Screen after loading vc4-kms-v3d

Chris56, 2023-11-03:
RPI 4, bookworm raspian , kernel 6.1.0-rpi4-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.54-1+rpt2 (2023-10-05) aarch64 GNU/Linux

I have a couple of TV sets which are all fine but one picke one.
RPI boots at 720p: all fine.
But when vc4-kms-v3d get loaded, screnn goes black, no signal.
I used to force HDMI into a mode, that the picky TV accepts (basically it should continue with 720p whet it boots.
However I cannot tell (or haow can I) vc4-kms-v3d to use a certain mode.

vlog says:
HDMI0:EDID error reading EDID block 0 attempt 0
HDMI0:EDID giving up on reading EDID block 0

dmesg keeps saying: 
EDID block 0 (tag 0x00) checksum is invalid, remainder is 170

So: how can I make vc4-kms-v3d ignore everything and just use a given HDMI mode.?

Chris56, 2023-11-03:
There is no display line. It tries sd and usb, then finds nothing and loops around this.

Without knowning, just guessing: EDID is the data returned by the TV Set (this data is broken ...). Thus proper information cannot be acquired from the TV. Therefore I would like to pin the resolution to the working 720p.
This has worked with respian buster. However not anymore with the vc4-kms-v3d driver.

The irformation is here: https://paste.debian.net/hidden/86da8d01

Also another piece of information:
Using another TV Set where the is not blank screen:
When I force the inital booting to 720p by config.txt:

hdmi_force_hotplug=1
hdmi_group=1
hdmi_mode=4
hdmi_force_mode=1

Then initally it uses 720p, BUT when vc4-kms-v3d gets loaded it switches to 1080p.
How can I force vc4-kms-v3d to NOT change the mode ???

popcornmix, 2023-11-03:
> There is no display line. It tries sd and usb, then finds nothing and loops around this.

Are you seeing a diagnostic screen on display?
If you are and you don't see a "display:" line, then [update the bootloader](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#bootloader_update_stable).

popcornmix, 2023-11-03:
> Connector 0 (32) HDMI-A-1 (disconnected)
  Encoder 0 (31) TMDS
Connector 1 (42) HDMI-A-2 (connected)
  Encoder 1 (41) TMDS

If using one display, use the first one (i.e. connector closest to power connector).

Chris56, 2023-11-03:
Same with the other hdmi port.

Chris56, 2023-11-03:
There is a config.txt parameter:
display_auto_detect=1
I did not see it before.
When set to
display_auto_detect=0
vc4-kms-v3d keeps the mode.

closed 2023-11-03 by Chris56

6by9, 2023-11-03:
Weird.
`display_auto_detect` determines whether the firmware looks for the Pi DSI display and tries to configure the kernel for it or not. It shouldn't affect HDMI at all.