raspberrypi/firmware #1839: RPi5 specific section in config.txt also configure pi4

digitalLumberjack, 2023-11-04:
Hello.

On kernel 392b9849c8d147c25557fea5d6dbcddbcb9e72d5 and firmware 7f14c6192dd536392efbad2280e990285447591a,

Adding
```
[pi5]
force_eeprom_read=0
```

Makes rpi4 also avoid eeprom read.

Have a good day.

timg236, 2023-11-06:
Yes that's the behavior for undefined conditionals. Future start.elf files will support [pi5] conditionals. You can use the [board-type=0xN] for a generic approac

pelwell, 2023-11-06:
Where the board type is bits 4-11 of the board revision:
```
pi@raspberrypi:~$ tail -3 /proc/cpuinfo
Revision        : d04170
Serial          : 3c4045e75bb203ac
Model           : Raspberry Pi 5 Model B Rev 1.0
```
Here the board type (in hexadecimal) is `17`, a.k.a. 0x17 or 23.

timg236, 2023-11-06:
See also https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#new-style-revision-codes-in-use

digitalLumberjack, 2023-11-06:
I will use the board type, thank you :)
I close :)

closed 2023-11-06 by digitalLumberjack

pelwell, 2023-12-19:
FYI support for the `[pi5]` filter appeared in the 6.1.58 kernel release of October 17th: https://github.com/raspberrypi/firmware/commit/fdb9eafae4b83e553593937eae8e77b0193903c3

digitalLumberjack, 2023-12-19:
Thank you for the update !