raspberrypi/firmware #1840: Can't boot pi4 from disk with 4096 logical block size

abaumgaertner, 2023-11-09:
Trying to boot of 4k logical sector size disk.
Manually setup GPT layout on large disk on Raspberry Pi 4 Model B Rev 1.4

Device Model:     ST18000NM003D-3DL103
Firmware Version: SN03
Sector Size:      4096 bytes logical/physical

Disk is connected via ASM235 (USB to SATA) and detected by bootloader.

Bootloop with error: MSD unsupported block size 4096
bootloader CURRENT: Wed Jan 11 17:40:52 UTC 2023 (1673458852)
VL805_FW: CURRENT: 000138c0

Same setup boots fine with disk set to 512bytes logical sector emulation.

timg236, 2023-11-09:
Yes, that's not supported by the bootloader.

diegocjorge, 2024-01-29:
I'm having the same problem. Is this ever going to be supported by the bootloader?

macromorgan, 2024-02-01:
Ditto for the Pi 5. Looks like I can't use a 4kn NVME drive without enabling 512e mode.

doubleuerner, 2024-11-07:
https://man.archlinux.org/man/nvme-format.1.en   bootloader does not support 4096, only 512

macromorgan, 2024-11-07:
> https://man.archlinux.org/man/nvme-format.1.en bootloader does not support 4096, only 512

I understand that, but was hoping that support *could* be added.

timg236, 2024-11-07:
It's on the TODO list (with many other things), priority would be Pi5 NVMe first, USB later, Pi4 etc afterwards

abaumgaertner, 2025-02-26:
tracking for pi5: Add support for 4K sectors raspberrypi/rpi-eeprom#577

timg236, 2025-02-26:
Good point - closing this feature request as duplicate

closed 2025-02-26 by timg236