raspberrypi/firmware #1841: RPI 4 - unable to enumerate USB device

Lab-Labovych, 2023-11-09:
**Problem:**
Periodically there is an issue 'unable to enumerate USB device' using an IoT device connected to the Raspberry Pi 4B.
The similar behaviour is also observed while using a USB-UART TTL converter (FTDI FT232RL).
The issue reproduces on 'bullseye' and 'bookworm' versions as well. 

**Symptoms:** 
RPi does not recognise connected USB device.

**Possible temporary solutions:**
- manual reconnect USB cable;
- reboot an RPi.

**dmesg logs:**
```
[78509.741458] usb 1-1.1: device descriptor read/64, error -71
[78509.961493] usb 1-1.1: device descriptor read/64, error -71
[78510.181280] usb 1-1.1: new high-speed USB device number 17 using xhci_hcd
[78510.281500] usb 1-1.1: device descriptor read/64, error -71
[78510.511590] usb 1-1.1: device descriptor read/64, error -71
[78510.641806] usb 1-1-port1: attempt power cycle
[78511.301306] usb 1-1.1: new high-speed USB device number 18 using xhci_hcd
[78511.301606] usb 1-1.1: Device not responding to setup address.
[78511.521602] usb 1-1.1: Device not responding to setup address.
[78511.741296] usb 1-1.1: device not accepting address 18, error -71
[78511.841300] usb 1-1.1: new high-speed USB device number 19 using xhci_hcd
[78511.841584] usb 1-1.1: Device not responding to setup address.
[78512.061544] usb 1-1.1: Device not responding to setup address.
[78512.281293] usb 1-1.1: device not accepting address 19, error -71
[78512.281751] usb 1-1-port1: unable to enumerate USB device
```
**System:**
```
Raspberry Pi reference 2023-05-03
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 47eee1f0ddcf8811559d51eea1c1bb48335e3e88, stage2

Mar 17 2023 10:50:39 
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)

Linux raspberrypi 6.1.21-v7l+ #1642 SMP Mon Apr  3 17:22:30 BST 2023 armv7l GNU/Linux
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
```
and
```
Raspberry Pi reference 2023-05-03
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 47eee1f0ddcf8811559d51eea1c1bb48335e3e88, stage2

Oct 17 2023 15:39:16 
Copyright (c) 2012 Broadcom
version 30f0c5e4d076da3ab4f341d88e7d505760b93ad7 (clean) (release) (start)

Linux raspberrypi 6.1.61-v7l+ #1696 SMP Thu Nov  2 16:40:28 GMT 2023 armv7l GNU/Linux
PRETTY_NAME="Raspbian GNU/Linux 12 (bookworm)"
```

kcorreaj, 2023-11-29:
Same issue, could you please have a look?

`device descriptor read/64, error -71
[3898879.821602] usb 1-1.1: new high-speed USB device number 61 using xhci_hcd
[3898879.931754] usb 1-1.1: device descriptor read/64, error -71
[3898880.151765] usb 1-1.1: device descriptor read/64, error -71
[3898880.272048] usb 1-1-port1: attempt power cycle
[3898880.931551] usb 1-1.1: new high-speed USB device number 62 using xhci_hcd
[3898880.931857] usb 1-1.1: Device not responding to setup address.
[3898881.151814] usb 1-1.1: Device not responding to setup address.
[3898881.371544] usb 1-1.1: device not accepting address 62, error -71
[3898881.471535] usb 1-1.1: new high-speed USB device number 63 using xhci_hcd
[3898881.471821] usb 1-1.1: Device not responding to setup address.
[3898881.691833] usb 1-1.1: Device not responding to setup address.
[3898881.911552] usb 1-1.1: device not accepting address 63, error -71````

JamesH65, 2023-11-29:
> Same issue, could you please have a look?
>
> ```
> `device descriptor read/64, error -71
> [3898879.821602] usb 1-1.1: new high-speed USB device number 61 using xhci_hcd
> [3898879.931754] usb 1-1.1: device [...]

The same device?

kcorreaj, 2023-11-29:
is its a camera but is the same type of device, IOT device.

Tanmayi-Indukrui, 2024-01-15:
same issue

sudo lsusb -v 
[with usb connection to pico]
...
Port 8: 0000.0101 power connect

[removed from usb port]
Port 8: 0000.0100 power

there should be power **enable** connect    ,  how to make it enabled ??