raspberrypi/firmware #1842: raspi-firmware fails upgrade over nfs

Philius, 2023-11-13:
After setting up nfs/tftpboot/PXE-boot I ran apt update && apt upgrade.
The upgrade process failed as it determined that /boot/firmware wasn't a mount point.

**To reproduce**
Configure the Raspberry Pi to boot over the network, set up nfs so it gets the files it needs.
My test case was a Raspberry Pi 4.

**Expected behaviour**
Things should go as expected.

**Actual behaviour**
The upgrade failed for raspi-firmware as well as its depencencies - the kernels, because `/etc/initramfs/post-update.d/z50-raspi-firmware` exited with return code 1.

> raspi-firmware: missing /boot/firmware, did you forget to mount it?

Here are my changes to `/etc/initramfs/post-update.d/z50-raspi-firmware`
```
firmware_dst="/boot/firmware"
rmp="$(df -Th / | tail -n 1 | awk '{print $2}')"
if ischroot ; then
  true # chroot detected - skip mount point check
elif [ -e /usr/bin/systemd-detect-virt ] && systemd-detect-virt -q ; then
  true # virtualization detected - skip mount point check
elif [ "nfs" = "$rmp" ] ; then
  true # nfs filesystem
elif ! mountpoint -q "${firmware_dst}" ; then
  echo "raspi-firmware: missing ${firmware_dst}, did you forget to mount it($rmp)?" >&2
  exit 1
fi
```
The `did you forget to mount it($rmp)?` change was there purely for testing - it's not needed.

For the `raspi-firmware` upgrade I also had to change `/var/lib/dpkg/info/raspi-firmware.postinst`

```
    rmp="$(df -Th / | tail -n 1 | awk '{print $2}')"
    if ischroot ; then
      true # chroot detected - skip mount point check
    elif test -e /usr/bin/systemd-detect-virt && systemd-detect-virt -q ; then
      true # virtualization detected - skip mount point check
    elif [ "nfs" = "$rmp" ] ; then
      true # nfs filesystem
    elif ! mountpoint -q /boot/firmware; then
      echo "Error: missing /boot/firmware, did you forget to mount it($rmp)?" >&2
      exit 1
    fi
```

I can paste the results of the raspinfo command if someone needs it. I'm leaving it out for now to avoid clutter.

* Which kernel version (`uname -a`)?
`Linux b5c83920 6.1.0-rpi4-rpi-v7l #1 SMP Raspbian 1:6.1.54-1+rpt2 (2023-10-05) armv7l GNU/Linux`

Note that this is the version of the running kernel, not the upgrade that included raspi-firmware.

After successful install and a reboot,
`Linux b5c83920 6.1.0-rpi6-rpi-v7l #1 SMP Raspbian 1:6.1.58-1+rpt2 (2023-10-27) armv7l GNU/Linux`

XECDesign, 2023-11-14:
Thanks for the report and workaround. I'll add something like that to the package.

XECDesign, 2023-11-14:
Does this variant of the check work for you?

```
elif [ "$(stat -f -c %T /boot/firmware)" = "nfs" ]; then
  true
```

Philius, 2023-11-14:
Yes. I never used `stat` before. Good to know.

Philius, 2023-11-14:
I just got it. All 8 kernels updated.
Synaptic or dpkg complained about device or resource busy.
That's normal with nfs as the Rpi's OS is on spinning rust that takes a moment to spin up.

closed 2023-11-14 by Philius