raspberrypi/firmware #1843: Pi5 fails to load initramfs with comma separated initrds

paul-1, 2023-11-16:
unexpected error @ 0x3fc1c642
BOOT ERROR: code 45 - ' Fatal firmware Error'

Using the firmware https://github.com/raspberrypi/firmware/commit/83dafbc92c0e63f76ca2ecdd42462d56489d1c77 with a 4GB pi5 running the 2023/10/30 bootloader

Concatenating the two initrd's into a single file, then everything loads fine.

pelwell, 2023-11-16:
I think this is another manifestation of the file length/alignment problem. Are any of the initramfs files (but not the last) of a word-odd length (length not a multiple of 4)?

paul-1, 2023-11-16:
In this case, neither of the two files are.

pelwell, 2023-11-16:
Really? That error certainly looks like an alignment assertion. Can you post your initramfs files somewhere I can try them?

paul-1, 2023-11-16:
I think we misunderstood each other.  Neither file is evenly divisible by 4.

4201562 bytes and 1640060 bytes.

pelwell, 2023-11-16:
The code can cope with odd lengths, but not odd alignment (but it's not immediately apparent why since we've been round this loop before).

paul-1, 2023-11-16:
Here is a copy of the boot partition.

https://www.dropbox.com/scl/fi/68lfqgkbzy4s8lvi8m3aq/bootpart.tar.gz?rlkey=rah12veoz4jkugw1lnccft83i&dl=0

pelwell, 2023-11-16:
Thanks, but I can reproduce here. Watch this space.

paul-1, 2023-11-16:
Glad you found it.  So many configurations to test.  I can concat my initrds for the short term while I do testing.  I just got my pi5 hardware, so I have lots to work on before I can do a release.

timg236, 2023-11-16:
We've churning the APT packages for betas and are attaching the release to (optionally) verify on the Github Issues. I think there will be a 'latest' release bump early next week

[rpi-eeprom-recovery.zip](https://github.com/raspberrypi/firmware/files/13378677/rpi-eeprom-recovery.zip)

paul-1, 2023-11-16:
So for my own sanity.   On a pi5, all this functionality is now handled by code in the eeprom, vs the pi4 (and previous devices).  I was debating last night which repo was the correct repo to tag with the issue. 

I can test with the attached when I get home this evening.  I'm not using RaspiOS for this project anyway, so the APT's don't help me.

timg236, 2023-11-16:
> So for my own sanity. On a pi5, all this functionality is now handled by code in the eeprom, vs the pi4 (and previous devices). I was debating last night which repo was the correct repo to tag with the issue.

Yeah, I think these are rpi-eeprom and rpi-firmware are interchangeable on Pi5 now. We monitor both :)

paul-1, 2023-11-16:
:)   BTW, the attached rpi-eeprom-recovery zip file has a readme that has no reference to the RPI5.

timg236, 2023-11-16:
Good spot. Yes, this was just grabbed from the internal build server which doesn't currently have separate README files.

The EEPROM is 2MB and the Pi5 recovery.bin will be ignored on a Pi4 and vice versa

paul-1, 2023-11-17:
Flashed and confirmed working fine.

timg236, 2023-11-17:
> Flashed and confirmed working fine.

Great. Thanks for verifying

closed 2023-11-17 by pelwell

paul-1, 2023-12-06:
@timg236 I see this made its way into the default firmware.  Is there a timeline to release new bootloader images so that rpi-imager can be used to update boards?

timg236, 2023-12-06:
The APT package has been updated so you should be able to "apt update" "rpi-eeprom-update -a"

pelwell, 2023-12-06:
(and possibly an "apt upgrade" as well, since apt update doesn't do very much)

paul-1, 2023-12-06:
While that is helpful, since this is a for a 3rd party OS, it's much easier to use the imager bootloader image to update.  Loading RaspiOS and, then updating takes a fair amount of time.

timg236, 2023-12-06:
> While that is helpful, since this is a for a 3rd party OS, it's much easier to use the imager bootloader image to update. Loading RaspiOS and, then updating takes a fair amount of time.

I can't comment on when other OS's update the rpi-eeprom package or equivalent. The bootloader RPi Imager will be updated in the next week or two for other changes so it would pull in those changes.

paul-1, 2023-12-06:
I'm the OS maintainer, I have provisions to update firmware via the rpi-eeprom repo, but I have to be able to boot the OS first.

A week or two would be great.

timg236, 2023-12-06:
To help OS maintainers and expert users we are modifying rpi-update to also update the bootloader EEPROM. On Pi5 there is no external start5.elf so this should address the gap in beta test that we had before with start4.elf
The intention is that new bootloader changes will appear in Github (but not APT) for a few days before we update the APT repo assuming there are no major regressions
https://github.com/raspberrypi/rpi-update/pull/24