raspberrypi/firmware #1845: v4l2 h264 decoder stops working after a few reuses in a loop

zocker-160, 2023-11-17:
**Is this the right place for my bug report?**
I am not 100% sure, but given that there is no dmesg output, I suspect this here is the right place.

**Describe the bug**
I am coming from a bug that I [reported over at gstreamer repo](https://gitlab.freedesktop.org/gstreamer/gstreamer/-/issues/3137) where the firmware misbehaving was mentioned as a potential cause of the problem.

The report here is essentially the same as the linked one above.

When running the script below, after the 4 - 5th loop no stream is displaying anymore, there is just an empty window and it never recovers.

**To reproduce**
```python
import time
import subprocess
import gi
gi.require_version("Gst", "1.0")
gi.require_version("GstApp", "1.0")
from gi.repository import Gst, GstApp

# public RTSP stream for testing purposes
URL = "rtsp://31.160.161.51/axis-media/media.amp"

Gst.init(None)

def mainLoop():
    loopcount = 0

    while True:
        loopcount += 1
        print("LOOP", loopcount)

        pipeline = Gst.parse_launch(
            f"rtspsrc location={URL} latency=0"
            " ! rtph264depay"
            " ! h264parse"

            " ! v4l2h264dec"

            # using this instead of v4l2h264dec works without issues
            #" ! avdec_h264"

            " ! videoscale"
            " ! videoconvert"
            " ! video/x-raw, width=854, height=480"
            " ! xvimagesink"
        )
        assert pipeline, "failed to create pipeline"

        print("starting video")
        ret = pipeline.set_state(Gst.State.PLAYING)        

        print("RET:", ret)
        if ret == Gst.StateChangeReturn.FAILURE:
            print("ERROR: failed to set pipeline to PLAYING")
            pipeline = None
            return

        print("waiting 10 seconds")
        time.sleep(10)

        print("stopping video")
        pipeline.set_state(Gst.State.NULL)
        pipeline = None
        
        print("waiting 1 second")
        time.sleep(1)

try:
    mainLoop()
except Exception as e:
    print(e)
```

**Expected behaviour**
I expect this to keep working and displaying the video stream, just like it does when I replace `v4l2h264dec` with `avdec_h264`.

**Actual behaviour**
It stops displaying any image and looks like this:

[image]

**System**

Pastebin: https://pastebin.com/RmcXvcbQ

* Which model of Raspberry Pi? 
Pi 4B
* Which OS and version (`cat /etc/rpi-issue`)?
```
Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage4
```
* Which firmware version (`vcgencmd version`)?
```
Oct 17 2023 15:39:16 
Copyright (c) 2012 Broadcom
version 30f0c5e4d076da3ab4f341d88e7d505760b93ad7 (clean) (release) (start)
```
* Which kernel version (`uname -a`)?
`Linux raspberrypi 6.1.61-v8+ #1696 SMP PREEMPT Thu Nov  2 16:44:46 GMT 2023 aarch64 GNU/Linux`

**Logs**
No errors in dmesg when this issue appears.

**Additional context**
- See my gstreamer bug report
- Script above uses public RTSP stream, so should be ready to go for testing

6by9, 2023-11-18:
`alloc_compact_internal ... failed` means the firmware has run out of memory (gpu_mem allocation).

Does setting `pipeline = None` actually destroy the pipeline, or leave it lying around and the garbage collector might get around to freeing it at some random point in the future? If it hasn't released the V4L2 decoder, then I can see the firmware having retained the resources required for it.

zocker-160, 2023-11-18:
Originally I have used `pipeline.unref()`, but one of the Gstreamer developers told me to [never call it](https://gitlab.freedesktop.org/gstreamer/gstreamer/-/issues/3137#note_2171808) and set the variable to None instead.

That being said, using unref() does not change the behavior.

I am not sure if this has to do with CmaFree, but when running `grep CmaFree /proc/meminfo` while the issue occurs I get: `CmaFree:          446476 kB`

I wonder though: is it normal to run out of memory just after 4 streams? In the past I have been able to play 6 - 8 streams at the same time using omxh264dec, which doesn't work anymore.

zocker-160, 2023-11-18:
Update:

I have redone the example in C and it suffers from the same issue:

```c
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <unistd.h>
#include <gst/gst.h>

#define URL "rtsp://31.160.161.51/axis-media/media.amp"

int loop() {
    GstElement *pipeline = gst_parse_launch(
        "rtspsrc location=" URL " latency=0"
        " ! rtph264depay"
        " ! h264parse"
        //" ! avdec_h264"
        " ! v4l2h264dec"
        " ! videoscale"
        " ! videoconvert"
        " ! video/x-raw, width=854, height=480"
        " ! xvimagesink",
        NULL
    );

    if (!pipeline) {
        g_printerr("Failed to create pipeline\n");
        return -1;
    }

    g_print("Starting video\n");
    if (gst_element_set_state(pipeline, GST_STATE_PLAYING) == GST_STATE_CHANGE_FAILURE) {
        g_printerr("Failed to set pipeline to PLAYING\n");
        return -1;
    }

    g_print("Waiting 10 seconds\n");
    sleep(10);

    g_print("Stopping video\n");
    gst_element_set_state(pipeline, GST_STATE_NULL);
    gst_object_unref(pipeline);

    g_print("Waiting 1 second\n");
    sleep(1);
}

int main(int argc, char *argv[]) {
    gst_init(&argc, &argv);

    int loopcount = 0;

    while (TRUE) {
        loopcount++;

        printf("LOOP: %d\n", loopcount);
        loop();
    }

    return 0;
}
```

6by9, 2023-11-18:
gpu_mem allocation, not CMA heap.

`sudo vcdbg reloc` will tell you the gpu_mem allocations.

At the point it fails but before you allow the application quits, `lsof | grep video10` would list the number of file handles open on the decoder. If more than 1, then the decoder hasn't been freed.

zocker-160, 2023-11-18:
Can confirm this is indeed happening, after the loop ends and playback stops, there are still a few entries left in `lsof | grep video10` and with each loop the number goes up until it doesn't work anymore.

So this could be gstreamer issue, I will update my issue report there and see what they have to say.

6by9, 2023-11-18:
I'll try and remember to test on Monday, but if they're not closing the file handle on the decoder, you will slowly (or not so slowly) leak resources.

6by9, 2023-11-20:
Hmm, some suboptimal behaviour from both sides.
GStreamer is calling 
```
[ 1020.564552] video10: VIDIOC_STREAMOFF: type=vid-out-mplane
[ 1020.564695] video10: VIDIOC_REQBUFS: count=0, type=vid-out-mplane, memory=mmap
[ 1020.564985] video10: VIDIOC_STREAMOFF: type=vid-cap-mplane
[ 1020.565086] video10: VIDIOC_REQBUFS: count=0, type=vid-cap-mplane, memory=mmap
```
when stopping the stream, but never closes the fd.
Closing the app closes all handles and releases all the resources (as long as we haven't locked up before that).

V4L2 has some distinct niggles.
One of the conformance tests opens the node 100 times, and all must succeed. That means we can't allocate the constrained resources on the firmware side until later (queue_setup which is part of REQ_BUFS), but take the approach that once assigned, you want to keep using the node until close.
We could destroy everything on detecting the above sequence of both input and output having been stopped, but that means that a media player that is just being efficient and reusing the one handle suffers a performance loss for unnecessary churn of state.

I can change the firmware to release the image pool on both ports going inactive, but that only defers the lockup as the kernel MMAL implementation has a limit of 64 components allocated at any one time.

So looping back to why is GStreamer not closing the file handle. Using GST_DEBUG=v4l2:7 I am getting
```
0:00:21.278751860  1113   0x5587147a10 DEBUG                   v4l2 v4l2_calls.c:712:gst_v4l2_close:<v4l2h264dec0:sink> Trying to close /dev/video10
0:00:21.278776953  1113   0x5587147a10 DEBUG                   v4l2 v4l2_calls.c:465:gst_v4l2_empty_lists:<v4l2h264dec0:sink> deleting enumerations
0:00:21.278802416  1113   0x5587147a10 DEBUG                   v4l2 v4l2_calls.c:712:gst_v4l2_close:<v4l2h264dec0:src> Trying to close /dev/video10
0:00:21.278824101  1113   0x5587147a10 DEBUG                   v4l2 v4l2_calls.c:465:gst_v4l2_empty_lists:<v4l2h264dec0:src> deleting enumerations
```
so it appears to be trying, but the kernel isn't seeing a call that is triggering release in the kernel driver.

zocker-160, 2023-11-20:
You have probably already looked into that part of the Gstreamer code, but it does seem to be doing a close() syscall in that part of the code:

[v4l2_calls.c#L723](https://gitlab.freedesktop.org/gstreamer/gstreamer/-/blob/main/subprojects/gst-plugins-good/sys/v4l2/v4l2_calls.c#L723)
[gstv4l2object.h#L213](https://gitlab.freedesktop.org/gstreamer/gstreamer/-/blob/main/subprojects/gst-plugins-good/sys/v4l2/gstv4l2object.h#L213)

6by9, 2023-11-20:
I'd prefer to defer to the GStreamer devs here, as they'll be far faster to know what is going on where.

Yes, I'd already looked, and generally the close hook is going to be `close` (https://gitlab.freedesktop.org/gstreamer/gstreamer/-/blob/main/subprojects/gst-plugins-good/sys/v4l2/gstv4l2object.c?ref_type=heads#L561).

`gst_v4l2_close` is obviously being called from the logging, so close  should be being called with `v4l2object->video_fd`. `v4l2object->video_fd` is set from `gst_v4l2_open` or `gst_v4l2_dup`.

`lsof | grep video10` is giving me 12 handles on /dev/video10 per playback instance, increasing to 60 when playback actually starts.

Changing the pipeline to play a local file also appears to have the same issue, but with less handles. I'm not running X, hence using kmssink instead of xvimagesink.
```
    GstElement *pipeline = gst_parse_launch(
        "filesrc location=../bbb_sunflower_1080p_60fps_normal.mp4"
        " ! qtdemux"
        " ! h264parse"
        //" ! avdec_h264"
        " ! v4l2h264dec"
        " ! videoscale"
        " ! videoconvert"
        " ! video/x-raw, width=854, height=480"
        " ! kmssink ",
//        " ! xvimagesink",
        NULL
    );
```
During playback
```
pi@bookworm64:~ $ lsof | grep video10
out       1629                              pi  mem       CHR               81,1                  344 /dev/video10
out       1629                              pi    9u      CHR               81,1       0t0        344 /dev/video10
out       1629                              pi   10u      CHR               81,1       0t0        344 /dev/video10
out       1629                              pi   12u      CHR               81,1       0t0        344 /dev/video10
out       1629                              pi   17u      CHR               81,1       0t0        344 /dev/video10
out       1629 1631 qtdemux0:               pi  mem       CHR               81,1                  344 /dev/video10
out       1629 1631 qtdemux0:               pi    9u      CHR               81,1       0t0        344 /dev/video10
out       1629 1631 qtdemux0:               pi   10u      CHR               81,1       0t0        344 /dev/video10
out       1629 1631 qtdemux0:               pi   12u      CHR               81,1       0t0        344 /dev/video10
out       1629 1631 qtdemux0:               pi   17u      CHR               81,1       0t0        344 /dev/video10
out       1629 1632 v4l2h264d               pi  mem       CHR               81,1                  344 /dev/video10
out       1629 1632 v4l2h264d               pi    9u      CHR               81,1       0t0        344 /dev/video10
out       1629 1632 v4l2h264d               pi   10u      CHR               81,1       0t0        344 /dev/video10
out       1629 1632 v4l2h264d               pi   12u      CHR               81,1       0t0        344 /dev/video10
out       1629 1632 v4l2h264d               pi   17u      CHR               81,1       0t0        344 /dev/video10
```
During the delay between loops
```
out       1629                              pi   12u      CHR               81,1      0t0        344 /dev/video10
out       1629                              pi   17u      CHR               81,1      0t0        344 /dev/video10
out       1629 1632 v4l2h264d               pi   12u      CHR               81,1      0t0        344 /dev/video10
out       1629 1632 v4l2h264d               pi   17u      CHR               81,1      0t0        344 /dev/video10
```
During 2nd loop
```
pi@bookworm64:~ $ lsof | grep video10
out       1629                              pi  mem       CHR               81,1                  344 /dev/video10
out       1629                              pi    9u      CHR               81,1       0t0        344 /dev/video10
out       1629                              pi   10u      CHR               81,1       0t0        344 /dev/video10
out       1629                              pi   12u      CHR               81,1       0t0        344 /dev/video10 <<< STILL HERE
out       1629                              pi   17u      CHR               81,1       0t0        344 /dev/video10 <<< STILL HERE
out       1629                              pi   22u      CHR               81,1       0t0        344 /dev/video10
out       1629                              pi   27u      CHR               81,1       0t0        344 /dev/video10
out       1629 1698 qtdemux1:               pi  mem       CHR               81,1                  344 /dev/video10
out       1629 1698 qtdemux1:               pi    9u      CHR               81,1       0t0        344 /dev/video10
out       1629 1698 qtdemux1:               pi   10u      CHR               81,1       0t0        344 /dev/video10
out       1629 1698 qtdemux1:               pi   12u      CHR               81,1       0t0        344 /dev/video10
out       1629 1698 qtdemux1:               pi   17u      CHR               81,1       0t0        344 /dev/video10
out       1629 1698 qtdemux1:               pi   22u      CHR               81,1       0t0        344 /dev/video10
out       1629 1698 qtdemux1:               pi   27u      CHR               81,1       0t0        344 /dev/video10
out       1629 1699 v4l2h264d               pi  mem       CHR               81,1                  344 /dev/video10
out       1629 1699 v4l2h264d               pi    9u      CHR               81,1       0t0        344 /dev/video10
[... cut ...]
out       1629 1699 v4l2h264d               pi   27u      CHR               81,1      0t0        344 /dev/video10 <<< GAINED
```
I don't immediately see a route for the video_fd within `GstV4l2Object` to be exported elsewhere, but it all appears to be pointing at a refcount accounting issue.
It doesn't look like debug symbols are available in the Pi OS builds of GStreamer, and I'm not in the mood for building it at present, so let's see what Nicolas can find.

`echo 0x7 > /sys/class/video4linux/video10/dev_debug` is the easiest way to enable logging of the basic V4L2 functions.

6by9, 2023-11-20:
Just to ensure I wasn't going crazy, the simple test program
```
#include <fcntl.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <unistd.h>

void main(void)
{
        int fd = open("/dev/video10", O_RDWR);
        int fd2;

        fd2 = dup(fd);

        sleep(10);

        close(fd);

        sleep(5);

        close(fd2);

        sleep(10);
}
```
Does call release after `close(fd2)`, so it is behaving how I would have expected.

zocker-160, 2023-11-20:
Thank you for your effort, I have forwarded this information.

Interestingly from my testing, I always had entries which seem to be related to gdbus left after each loop:
[image]

which is something you don't seem to experience.

zocker-160, 2023-11-21:
Nicolas from Gstreamer is sadly unable to reproduce this issue :(.

I have retried with the latest stable version of Gstreamer (1.22.7) compiled from source as requested, but with the same results.

zocker-160, 2023-11-23:
Here some of my testing results on a Pi 4:

| Pi OS  | Gstreamer | Kernel | Issue? | 
| :------: | :------: | :------: | :------: |
| 11     | 1.18   | 5.x 32bit   | NO |
| 11     | 1.18   | 5.x 64bit   | NO |
| 11     | 1.18   | 6.1.x 64bit   | NO |
| 11     | 1.18   | 6.1.63 64bit w/ firmware (1) | NO |
| 11     | 1.22.7   | 6.1.x 64bit   | NO |
| 11     | 1.22.7   | 6.1.63 64bit w/ firmware (1) | NO |
| 12     | 1.22.0 | 6.1.x 64bit | YES |
| 12     | 1.22.7 | 6.1.x 64bit | YES |
| 12     | 1.22.7 | 6.1.63 64bit w/ firmware (1) | YES |

(1): `8e2a862c9c7a0cc5cd505c8f89761d723af8768c` (latest available as of 23.11.23)

Different kernel, firmware or Gstreamer versions don't seem to make a difference. The only connection I can see is Pi OS / Debian 11 vs 12.

Can you make any sense of this? 

For some reason staying on the same firmware, gstreamer and kernel but only changing from bullseye to bookworm introduces this issue.

zocker-160, 2023-11-23:
> Is that a self-built GStreamer in each case? I would have expected Bullseye and Bookworm to have different GStreamer versions.

1.18 on bullseye and 1.22.0 on bookworm are both default from the repo, 1.22.7 are always compiled from source.

zocker-160, 2023-11-24:
Final update:

The issue is fixed in `plugins-good` version 1.22.7. Stupid me was only building the gstreamer base plugins from source, but not the one that actually matters (v4l2 is part of plugins-good).

So closing here, as this is not firmware related.

PS: @6by9 would it be possible to update the `gstreamer1.0-plugins-good` package (or all of gstreamer for that matter) to 1.22.7 on Raspberry Pi OS?

closed 2023-11-24 by zocker-160

6by9, 2023-11-24:
I thought the meson build system with [gstreamer monorepo](https://gstreamer.freedesktop.org/documentation/frequently-asked-questions/mono-repository.html) built and installed everything. Last time I tried it, it was far easier than the half-dozen repos that needed pulling and building.

We're still generally following Debian for these packages. If we can identify the fix, then we can backport it, however there's a fair amount of churn on the 1.22 branch for sys/v4l2, so it may not be a simple backport.
The curious one is I can't even see an obvious route to leak the handle, but I'd need to check that I'm looking at the PiOS 1.22 source and not just upstream.

6by9, 2023-11-27:
Ah the joys of a bisect when the build takes a good while (I could trim down the build, but getting the runes right probably takes longer than the builds).

1.22.0 has the issue. 1.22.4 does not. The next 2 steps will have to wait for tomorrow.

zocker-160, 2023-11-27:
> I thought the meson build system with [gstreamer monorepo](https://gstreamer.freedesktop.org/documentation/frequently-asked-questions/mono-repository.html) built and installed everything. Last time I tried it, it was far easier than the half-dozen repos that needed pulling and building.

You are ofc correct about that, but I used the source from `https://gstreamer.freedesktop.org/src/gstreamer` not knowing it does not contain any of the plugins, those are in separate folders.

> Ah the joys of a bisect when the build takes a good while (I could trim down the build, but getting the runes right probably takes longer than the builds).

1.22.0 has the issue. 1.22.4 does not. The next 2 steps will have to wait for tomorrow.

LMK if I can assist / help you in any way.

6by9, 2023-11-28:
1.22.2 OK.
1.22.1 OK.

I'm going back and retesting 1.22.0 now, which is needed anyway as I'll then be trying cherry-picking patches to try and find the fix.