raspberrypi/firmware #1846: margin_*overscan_* doesn't work in bookworm/kernel 6.1.x anymore

BobOmbuzal, 2023-11-17:
I'm using "Raspberry Pi 3 Model B Rev 1.2" with Bookworm
Console mode only, no Xorg running

I set up the /boot/firmware/config.txt like this: 
```
disable_overscan=0
overscan_left=-10                                                     
overscan_right=-10                                                    
overscan_top=10                                                       
overscan_bottom=10                                                    
hdmi_group=1
hdmi_mode=4
```
to get 1280x720 with overscan pixels.
it did work in bullseye, but it doesn't seem to work on boorworm anymore.

neod123, 2023-11-28:
Apparently this is due to the new wayfire system which replace X11.

I was also not able to modify the overscan.

According to the release note, this is currently not possible:

> Compensation for displays which use overscan is tricky under Wayland, and we haven’t quite got it working yet, so this has been removed for now. The vast majority of displays nowadays don’t need it, but we will be putting it back when we have worked out how best to do it!

[source](https://www.raspberrypi.com/news/bookworm-the-new-version-of-raspberry-pi-os/)

I have tried several combinations in `.config/wayfire.ini `but nothing change.

Several people mentioned this issue also:
[Raspberry forum 1](https://forums.raspberrypi.com/viewtopic.php?t=357687)
[Raspberry forum 2](https://forums.raspberrypi.com/viewtopic.php?p=2144558&hilit=margin#p2144558)

I am hesitating to coming back to bulleyes version.

6by9, 2023-11-29:
You can switch back to X11 on Bookworm using raspi-config. That would be preferable over switching back to Bullseye.

neod123, 2024-01-13:
Ok, thank for the help but this was still did not work.

I have founded another solution that I will explain here for those who need it:

1. Download and flash Raspian OS bookworm-arm64 from 2023-12-05
2. After the basic setup, switch to X11 as mentionned below.
3. Use the randr command as below to setup the screen properly.
`   export DISPLAY=:0 ; xrandr --output HDMI-2 --fb 1280x720 --panning 1280x720 --scale 990x550 --transform 1.3,0,-65,0,1.3,-100,0,0,1`
4. Apply on crontab at startup   

The approch is different but this work in the same way that old ini files possibilities.

schroray, 2024-05-19:
Initial source: https://raspberrypi.stackexchange.com/questions/142512/how-to-get-overscan-working-on-raspi-4/148024

I am using this successfully on Raspberry Pi 5 / Wayland hosts:

`echo "--- Before -" && sudo cat /boot/firmware/cmdline.txt && echo && echo "--- Remove video configuration -" && sudo sed -i 's/[[:space:]]*video=.*margin_.*[[:space:]]//g' /boot/firmware/cmdline.txt && sudo cat /boot/firmware/cmdline.txt && echo && echo "--- Final - Append video configuration -" && sudo sed -i "1 s/$/ video=HDMI-A-1:1920x1080@60,margin_left=60,margin_right=60,margin_top=40,margin_bottom=40 /" /boot/firmware/cmdline.txt && sudo cat /boot/firmware/cmdline.txt && echo && echo "--- missing just a 'sudo reboot'"`