raspberrypi/firmware #1848: Raspberry Pi 5 fails to boot with a large config.txt

leon-anavi, 2023-11-24:
**Is this the right place for my bug report?**

Yes, because it is related to the boot process and config.txt size.

**Describe the bug**

While adding Raspberry Pi 5 support in Yocto/OpenEmbedded BSP meta-raspberrypi we noticed that the board doesn't boot with a large `config.txt` file: https://github.com/agherzan/meta-raspberrypi/pull/1237#

I haven't seen this issue before on previous Raspberry Pi models and versions. @floion started [a discussion in the Raspberry Pi forum](https://forums.raspberrypi.com/viewtopic.php?t=359838&sid=91ba6454119bc581013fa2f3fe922c60).  [jamesh, Raspberry Pi Engineer & Forum Moderator, recommended to get rid of the comments and reduce the file size](https://forums.raspberrypi.com/viewtopic.php?p=2159238#p2159656). In meta-raspberrypi we can do this easily with `sed -i '/^##/d' $CONFIG` (at least temporary until we have a better solution) but let's also report the issue here too.

**To reproduce**

Try booting Raspberry Pi 5 with a large config.txt: https://raw.githubusercontent.com/Evilpaul/RPi-config/master/config.txt

**Expected behaviour**

The expected behavior is Raspberry Pi 5 to boot.

**Actual behaviour**

Raspberry Pi 5 doesn't boot.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW

Raspberry Pi 5

* Which OS and version (`cat /etc/rpi-issue`)?

Raspberry Pi OS or image built with https://github.com/agherzan/meta-raspberrypi/pull/1237

* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)?

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

[raspberrypi5-large-config-20231124.txt](https://github.com/raspberrypi/firmware/files/13458749/raspberrypi5-large-config-20231124.txt)

**Additional context**

timg236, 2023-11-24:
The limit is 32K

leon-anavi, 2023-11-24:
@timg236 thank you for the quick feedback and the shared technical details. Noted down. In the particular case `config.txt` is 49K so your explanation makes sense:
```
$ ls -lah config.txt
-rw-r--r-- 1 leon leon 49K ное 24 13:55 config.txt
```
Regarding these circumstances in short term at Yocto/OpenEmbedded BSP layer meta-raspbebrrypi we will just remove various comments from `config.txt` to reduce the file size and in long term we will come up with more sustainable solution for `config.txt`.

closed 2023-11-24 by leon-anavi

timg236, 2023-11-24:
For larger config.txt files using include statements may be helpful. I think there's possibly room for improvement with error messages in future bootloader updates so we can add that to the backlog.
Amongst other reasons the bounded size is that config.txt is now parsed earlier by the bootloader code which is has a bounded size for signature checks