raspberrypi/firmware #1849: Banned connecting to mirror.datacenter.by

Lab-Labovych, 2023-11-30:
**Describe the bug**
Since all russian (ru) and belarusian (by) domains are banned by most of Ukrainian Internet providers because of russian invasion of Ukraine we have troubles during the installing some of the packages. Vim for instance and other. 
The necessary package can't be downloaded from mirror.datacenter.by server and just freezes.

How can we avoid using the ru an by mirrors without manual changing the `/etc/apt/sources.list` file?

**To reproduce**
`sudo apt install vim`

**Expected behaviour**
vim package is installed

**Actual behaviour**
```
pi@raspberrypi:/etc/apt/preferences.d $ sudo apt install vim
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  libgpm2 libsodium23 vim-runtime
Suggested packages:
  gpm ctags vim-doc vim-scripts
The following NEW packages will be installed:
  libgpm2 libsodium23 vim vim-runtime
0 upgraded, 4 newly installed, 0 to remove and 66 not upgraded.
Need to get 8209 kB/8388 kB of archives.
After this operation, 40.6 MB of additional disk space will be used.
Do you want to continue? [Y/n] Y
12% [Connecting to mirror.datacenter.by (2a02:2208:1:3::12)] 
```

**System**

```
Raspberry Pi reference 2023-05-03
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 47eee1f0ddcf8811559d51eea1c1bb48335e3e88, stage2

Mar 17 2023 10:50:39 
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)

Linux raspberrypi 6.1.21-v7l+ #1642 SMP Mon Apr  3 17:22:30 BST 2023 armv7l GNU/Linux
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
```

XECDesign, 2023-12-04:
I'm not aware of a way to get around this without changing `/etc/hosts` or `/etc/apt/sources.list`.

Since mirrordirector is not managed by us, there's not much we can do on our end. You could try reporting the issue over at https://bugs.launchpad.net/raspbian/+bugs, but I doubt that will help.

popcornmix, 2024-02-23:
Closing as this isn't a firmware issue.

zhil, 2025-02-09:
@Lab-Labovych how did you solve this issue for you?

@XECDesign can you please explain, how can you solve this issue by changing /etc/hosts or /etc/apt/sources.list ?
I tried different options, but still locked

XECDesign, 2025-02-10:
You can edit /etc/apt/sources.list and change the URL there to a mirror that's closer to you, for example: `http://fastmirror.pp.ua/raspbian/raspbian/`

The Raspbian site seems a bit broken right now, but here's an archive of [a list of mirrors](https://web.archive.org/web/https://www.raspbian.org/RaspbianMirrors) you can try.

Avamander, 2025-07-08:
I personally just don't want any (especially unencrypted) traffic going to Russia or Belarus, even if I'm not in Ukraine.

Avamander, 2025-07-22:
E77FC0EC34276B4B should be [Raspbian multiarch automatic signing key. <root@raspbian.org>](https://keyserver.ubuntu.com/pks/lookup?search=E77FC0EC34276B4B&fingerprint=on&op=index), but other than that, yes.