raspberrypi/firmware #1850: Rpi4 tftp boot doesn’t respect option 67

Smiril, 2023-11-30:
**Describe the bug**
Fail to boot from tftpd-hpa server with nfs4 subsystem 

**To reproduce**
Load the firmware from raspios bookworm arm64
Make boot from network with raspi-config

**Actual behaviour**
DHCP Option 67 or Filename not respected

**System**
* Raspberry Pi4
* Debian bookworm
* firmware version latest from update
* kernel version Linux pxe-master 6.1.0-13-arm64 #1 SMP Debian 6.1.55-1 (2023-09-29) aarch64 GNU/Linux

**Logs**
TFTP 1: File not found
TFTP 1: File not found
TFTP 1: File not found
TFTP 1: File not found
TFTP 1: File not found
TFTP 1: File not found 
TFTP 1: File not found 
TFTP 1: File not found
Firmware not found

**Additional context**
Dhcpd.conf_host

		host rpi1 {
			option root-path "/mnt/usb/tftpboot/";
			hardware ethernet d8:3a:dd:0b:10:8f;
			option option-43 "Raspberry Pi Boot";
			option tftp-server-name "192.168.50.1";
			option bootfile-name "53649973/cmdline.txt";	
			next-server 192.168.50.1;
			filename "53649973/cmdline.txt";
			fixed-address 192.168.50.11;
			option host-name "rpi1";
		}

timg236, 2023-12-01:
That's correct it's not supported.

Sorry, there are no plans to support this or have full PXE support. TFTP is just a remote file-system.

The TFTP prefix options may help when managing multiple devices
https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#TFTP_PREFIX

Smiril, 2023-12-01:
> That's correct it's not supported.
> 
> Sorry, there are no plans to support this or have full PXE support. 
> 

Any option to run pxeboot on a rpi4 at least 8gb ram?

timg236, 2023-12-04:
Network DHCP + TFTP boot has been supported for a few years now

see https://www.raspberrypi.com/documentation/computers/remote-access.html#network-boot-your-raspberry-pi

closed 2024-01-06 by timg236