raspberrypi/firmware #1851: Garbled output with official touchscreen if DRM_CAP_ASYNC_PAGE_FLIP is used

maru-sama, 2023-12-01:
**Describe the bug**
I recently upgraded to bookworm and in this process also updated the SDL libraries I am using for my KIVY based touchscreen application. I used the python3-kivy deb and started my application which runs on the console. Instead of the normal GUI I just saw garbled output. 
Initially I thought that it was a SDL issue but after bisecting it I found out that DRM_CAP_ASYNC_PAGE_FLIP was the culprit. Starting in SDL-2.0.16 async page flip is supported by SDL and probed for. 
Apparently the rpi drm implementation supports it so SDL enables it but this results in garbled output. For testing purposes I recompiled SDL, turned off ASYNC_PAGE_FLIP and it worked fine,
I am not sure if this is a mesa/drm or kernel issue, but as it is right now kivy on the console is not working with the official touchscreen display rpi4 and bookworm.

**To reproduce**
- Install python3-kivy
- Run one of the demo applications on the console using DRM

**Expected behaviour**
Normal output of the Kivy GUI

**Actual behaviour**
Garbled output

**System**

Raspberry Pi 4 Model B Rev 1.4
PRETTY_NAME="Raspbian GNU/Linux 12 (bookworm)"
NAME="Raspbian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"

Raspberry Pi reference 2016-03-18
Generated using Pi-gen, https://github.com/RPi-Distro/Pi-gen, stage2

Linux rpi 6.1.0-rpi6-rpi-v7l #1 SMP Raspbian 1:6.1.58-1+rpt2 (2023-10-27) armv7l GNU/Linux
Revision	: d03114
Serial		: 10000000d53fdaf2
Model		: Raspberry Pi 4 Model B Rev 1.4
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0, libcamera interfaces=0

Videocore information
---------------------

Oct 17 2023 15:39:16 
Copyright (c) 2012 Broadcom
version 30f0c5e4d076da3ab4f341d88e7d505760b93ad7 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

6by9, 2023-12-01:
> Actual behaviour
Garbled output

A picture is worth 1000 words.

https://dri.freedesktop.org/docs/drm/gpu/drm-uapi.html#c.DRM_MODE_PAGE_FLIP_ASYNC
```
DRM_MODE_PAGE_FLIP_ASYNC ()

Parameters

Description

Request that the page-flip is performed as soon as possible, ie. with no delay due to waiting for vblank. This may cause tearing to be visible on the screen.

When used with atomic uAPI, the driver will return an error if the hardware doesn't support performing an asynchronous page-flip for this update. User-space should handle this, e.g. by falling back to a regular page-flip.

Note, some hardware might need to perform one last synchronous page-flip before being able to switch to asynchronous page-flips. As an exception, the driver will return success even though that first page-flip is not asynchronous.
```
I'd actually want to read up on that, as the HVS latches the display list at the start of each frame, so I don't know exactly what it means by async here.

https://dri.freedesktop.org/docs/drm/gpu/drm-uapi.html#c.DRM_CAP_ASYNC_PAGE_FLIP
```
If set to 1, the driver supports [DRM_MODE_PAGE_FLIP_ASYNC](https://dri.freedesktop.org/docs/drm/gpu/drm-uapi.html#c.DRM_MODE_PAGE_FLIP_ASYNC) for legacy page-flips.
```
If it's the legacy API, then it sounds like it isn't the call you are wanting.

My recollection is that it was used by X11 for cursor plane updates only.

maru-sama, 2023-12-01:
Sdl is checking for this and enables async pageflips
https://github.com/libsdl-org/SDL/blob/058213366bccf7f62d4c4c6061477d7a69ea336d/src/video/kmsdrm/SDL_kmsdrmvideo.c#L957

https://github.com/libsdl-org/SDL/blob/058213366bccf7f62d4c4c6061477d7a69ea336d/src/video/kmsdrm/SDL_kmsdrmopengles.c#L175

If I hardcore this to disabled it works fine.

maru-sama, 2023-12-01:
Please note that a mesa bug was opened for this as well.
https://gitlab.freedesktop.org/mesa/mesa/-/issues/10242

maru-sama, 2023-12-02:
Here pictures with pageflip disabled and enabled...

Enabled
[image]

Disabled
[image]

Both on a rpi4

appnjoy, 2023-12-05:
Hi guys, just a stupid question :)

Is it planned somewhere when a Mesa driver fix will be released for Raspberry Pi?

Thanks

6by9, 2023-12-05:
> Is it planned somewhere when a Mesa driver fix will be released for Raspberry Pi?

A fix for this specific issue, or a general update?

For this specific issue would require this issue to be resolved in the first place.

Generally Mesa on Raspberry Pi OS was updated to 23.2.1 on 20th Nov 2023. This is ahead of Debian Bookworm which has stuck on 22.3. Updates take a fair amount of testing effort, so they generally happen when something significant dictates it.

maru-sama, 2023-12-05:
Furthermore I think nobody knows yet if this is a Kernel or Mesa library related issue or a combination of both.
The only "good" thing IMO is that it is easily reproducible. :)

cmitu, 2024-02-08:
I think this is fixed with https://github.com/raspberrypi/linux/pull/5827, https://github.com/raspberrypi/linux/pull/5861.

popcornmix, 2024-02-08:
Yes, the fix should be present in apt now. Closing.

maru-sama, 2024-02-19:
Hello,

I just tried the latest kernel with pageflip enabled and I am still seeing this issue with my rpi4

` ` `
Linux rpi 6.1.0-rpi8-rpi-v7l #1 SMP Raspbian 1:6.1.73-1+rpt1 (2024-01-25) armv7l GNU/Linux
` ` `

popcornmix, 2024-02-19:
It looks like that kernel came from [this tag](https://github.com/raspberrypi/linux/commits/stable_20240124/) which appears to contain the fix.

The 6.1 kernel is unlikely to receive any more updates. It would be useful if you can test the [6.6 kernel](https://forums.raspberrypi.com/viewtopic.php?t=361116) using rpi-update.

The 6.6 will become the stable (apt) kernel soon.

maru-sama, 2024-02-19:
Ok, I will test it again as soon as it is available via apt.