raspberrypi/firmware #1853: Rpi 5 cannot boot to multi-boot partitions

procount, 2023-12-06:
The Pi5 boot loader _can_ use autoboot.txt (boot_partition=n) to boot to an alternative partition, but neither `sudo reboot n` nor `::syscall(SYS_reboot, LINUX_REBOOT_MAGIC1, LINUX_REBOOT_MAGIC2, LINUX_REBOOT_CMD_RESTART2, n)` seem to work.
I believe this feature might have been delayed til post-launch along with some other improvements in that area.
Any idea when it might be implemented?

timg236, 2023-12-07:
We are looking into it. Possibly a kernel issue since the magic numbers don't seem to get to the bootloader.

procount, 2023-12-07:
Just FYI - I'm using the 6.1.54 aarch64 kernel. 
The same kernel reboots fine on a Pi4 and other models, whether in PINN, Raspios64-lite or raspios64 (standard).
It's only the Pi5 where it fails. 
I upgraded raspios 64 lite to the latest 6.1.63 and it still fails to reboot to the correct partition.

procount, 2023-12-13:
Many thanks.
Please advise when it is complete so I can test it.
I guess I will also need to advise users to upgrade their firmware in eeprom, so an indication of the minimum eeprom version or commit id etc would be useful. Or maybe I could just push out the new version via PINN if the user's present version is not new enough?

timg236, 2023-12-13:
Please find attached a bench-tested version of the firmware.
[rpi-eeprom-recovery.zip](https://github.com/raspberrypi/firmware/files/13663251/rpi-eeprom-recovery.zip)

I think this can go to the bootloader 'latest' release pretty soon because it's a very small fix. 

FYI rpi-update can now update the bootloader "SKIP_BOOTLOADER rpi-update". I'm hoping that this will allow us to get bug fixes tested earlier so we can reduce the delay before making bootloader updates generally available.
https://github.com/raspberrypi/rpi-update/commit/62ef428cf9796a327ef44b92caa41c59c02c46e2

procount, 2023-12-18:
I updated the eeprom, applied the kernel fix to 6.1.54 and It works - PINN is multi-booting again on the Pi5!
Many thanks 😄

closed 2024-02-29 by timg236