raspberrypi/firmware #1855: Not possible to configure ethernet leds in bookworm

toy, 2023-12-17:
**Describe the bug**
After switching to bookworm it is not possible to switch off ethernet leds from `/boot/config.txt`.
I used [latest 32-bit release](https://downloads.raspberrypi.com/raspios_armhf/images/raspios_armhf-2023-12-06/2023-12-05-raspios-bookworm-armhf.img.xz) and was not able to affect ethernet leds, tried different values. Using exact steps but with [latest bullseye](https://downloads.raspberrypi.com/raspios_oldstable_armhf/images/raspios_oldstable_armhf-2023-12-06/2023-12-05-raspios-bullseye-armhf.img.xz) turns off ethernet leds.
Looks like https://github.com/raspberrypi/firmware/issues/1742#issuecomment-1396177708 is reporting the same thing, but it is not answered in the thread.

**To reproduce**
Write the image to the microSD card, modify `/boot/config.txt` to contain:
```
dtparam=eth_led0=4
dtparam=eth_led1=4
```
Start the Pi

**Expected behaviour**
The ethernet leds are off.

**Actual behaviour**
The ethernet leds are on (one showing activity).

**System**

* Which model of Raspberry Pi?
  ```
  Raspberry Pi 4 Model B Rev 1.5
  ```
* Which OS and version (`cat /etc/rpi-issue`)?
  ```
  Raspberry Pi reference 2023-12-05
  Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 02b371de1a87bbe70cfcbd498527067a7d2ee6ca, stage4
  ```
* Which firmware version (`vcgencmd version`)?
  ```
  Oct 17 2023 15:39:16
  Copyright (c) 2012 Broadcom
  version 30f0c5e4d076da3ab4f341d88e7d505760b93ad7 (clean) (release) (start)
  ```
* Which kernel version (`uname -a`)?
  ```
  Linux ds 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux
  ```

pelwell, 2023-12-18:
Bookworm moved the configuration file to `/boot/firmware/config.txt`. The rest should be unchanged.

toy, 2023-12-18:
@pelwell Thank you, indeed it works.

Unfortunate that I didn't see that `/boot/config.txt` is originally a symlink to `firmware/config.txt` – I was adding the dtparams using ansible playbook which replaced the symlink with the changed content.

closed 2023-12-18 by toy