raspberrypi/firmware #1856: Pi 5 8GB Unable to boot from USB SSD

Babyyoda777, 2023-12-24:
**Is this the right place for my bug report?**
This repository contains the GPU firmware used on the Raspberry Pi. This software is the closed source part of the Raspberry Pi system, it includes booting (including network booting and USB booting), low-level power and clock control, FKMS and legacy HDMI control (not full KMS that is in the linux kernel), hardware legacy codecs (MPEG2, H264 and VC1), encode hardware including the ISP (image sensor pipeline) and camera control, audio output (analogue and HDMI audio).

If you believe that the issue you are seeing is within this area, this is the right place. If not, we have other repositories for the linux kernel at [github.com/raspberrypi/linux](https://github.com/raspberrypi/linux) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
Multiple clean flashes of SSD using Raspberry Pi imager. Tried both 64bit Lite and Full versions for RPi 5 with no luck. The raspberry pi turns on and then nothing, the fan just spins for ever until I unplug and replug. After re plugging fan stops after 8 seconds like a normal boot but still no sign of life when actually sshing or connecting to pi. Tried using SD card and everything worked normally. I am using the official power supply and active cooler.

**To reproduce**
List the steps required to reproduce the issue.

**Expected behaviour**
Pi simply boots normally from usb SSD and connects like normal. 
K
**Actual behaviour**
Pi turn on, LED turns green but Ethernet shows only solid orange and no green. Fan spins for ever and pi is never actually accessible.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
* Which OS and version (`cat /etc/rpi-issue`)?
* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)?
* 
unbke to access

**Logs**
If applicable, add the relevant output from `dmesg` or similar.
Unable to access 
**Additional context**
Add any other relevant context for the problem.

using official power supply with 5.1V 5A. Same setup worked fine on Raspi4B. Boots normally using SD card as well.

Paddy4711, 2023-12-27:
I do have the same issue exact like described above. And the same Hardware Pi 5 8 GB.

timg236, 2024-01-02:
Generic support / interop issues should be asked on the forums.
Closing because there’s no logs or description of the hardware so can’t reproduce

closed 2024-01-02 by timg236