raspberrypi/firmware #1859: bootcode.bin to fallback to booting sdcard if PXE fails

kmously, 2024-01-09:
I can currently boot a Raspberry Pi 2B v1.1 from pxe using a bootcode.bin on the first partition without any other files. If the rest of the files (like start.elf) are also on the same partition, it boots normally from the local SD. Again, no problems.

However I cannot find a way to make the RPI2 v1.1 boot from the SD card if PXE fails or times out. Is there a way to do this? The same bootcode.bin has no problem booting either way so I assume it must be possible.

It is obviously very useful as it allows a person to remotely determine exactly how to boot the RPI2 v1.1. 

Thanks

kmously, 2024-01-10:
Looks like bootcode.bin does USB then PXE - is there a way to make it do "PXE then USB" or "PXE then microSD"?

timg236, 2024-01-10:
No. I don’t believe it’s configurable it’s the same as the ROM code.