raspberrypi/firmware #1862: Enhancement Proposal: 16GB RAM Option for Future Raspberry Pi 5b+ Model

aravhawk, 2024-01-27:
Hello Raspberry Pi Team,

I'm writing to propose an enhancement for the next iteration of the Raspberry Pi, potentially a 5b+ model. Given the evolution and increasing demands of modern computing, I believe that offering a 16GB RAM option would significantly benefit power users and expand the Raspberry Pi's capabilities in various advanced applications.

Current Context:
The Raspberry Pi 5 has made impressive strides with its 2.4GHz quad-core, 64-bit Arm Cortex-A76 CPU, and options for 4GB or 8GB LPDDR4X-4267 SDRAM. These specs have undoubtedly elevated the device's utility in both consumer and industrial segments. However, as applications grow more complex and resource-intensive, there's an emerging need for even greater memory capacity.

Proposed Enhancement:

16GB LPDDR4X SDRAM Option: Introducing a 16GB RAM variant would cater to users engaged in high-level tasks such as complex simulations, intensive coding projects, and AI development. This upgrade would make the Raspberry Pi an even more versatile tool for educational purposes, professional development, and industrial applications.

Accompanying CPU Upgrade: To complement the increased RAM, a minor refreshment in the CPU, possibly with improved efficiency and processing power, would balance the system's performance. This ensures that the Raspberry Pi can leverage the additional memory effectively without bottlenecking.

Potential Challenges: While advocating for this upgrade, it's also important to consider factors like RAM bandwidth improvements to support the higher memory capacity efficiently. Additionally, the impact on the device's pricing and power consumption should be taken into account, maintaining the Raspberry Pi's reputation for affordability and energy efficiency.

Rationale:
The suggested enhancements align with the Raspberry Pi's tradition of continuous improvement and adaptation to the evolving needs of its user base. As the complexity of projects undertaken by Raspberry Pi users increases, so too should the device's capabilities. A 16GB RAM option would not only meet current demands but also future-proof the device for emerging technologies and applications.

In conclusion, I believe that incorporating a 16GB RAM option, along with corresponding CPU and bandwidth improvements, would be a significant step forward for the Raspberry Pi, aligning with its mission to democratize access to advanced computing.

Thank you for considering this proposal. I look forward to seeing the continuous evolution of the Raspberry Pi and its growing impact in the world of computing and education.

Best regards,
Arav Jain

closed 2024-01-27 by pelwell

JamesH65, 2024-02-11:
Not quite sure what you are expecting as a follow-up. We do not usually announce products in advance. Requests for 16GB have been frequent on the forums, so it's not like this is an unusual request, and our engineers are perfectly aware of what would be needed to implement it, and what benefits it would bring.

aravhawk, 2024-02-11:
Ok, understood. Thanks for letting me know.