raspberrypi/firmware #1865: Raspberry Pi 5 | "sdram_freq_min" set by default at 0 instead of 4267

mann1x, 2024-02-06:
**Describe the bug**
In the config.txt documentation the default value for "sdram_freq_min" is 4267 while actually it's set to 0.

**To reproduce**
Query the value via vcgencmd

**Expected behaviour**
```shell
mannix@raspberrypi:~ $ vcgencmd get_config sdram_freq_min
sdram_freq_min=4267
```

**Actual behaviour**
```shell
mannix@raspberrypi:~ $ vcgencmd get_config sdram_freq_min
sdram_freq_min=0
```

**System**
Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW

Pi 5 Model B Rev 1.0

* Which OS and version (`cat /etc/rpi-issue`)?

>Raspberry Pi reference 2023-12-05
>Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 70cd6f2a1e34d07f5cba7047aea5b92457372e05, stage4

* Which firmware version (`vcgencmd version`)?

>2024/01/05 15:57:40
>Copyright (c) 2012 Broadcom
>version 30cc5f37 (release) (embedded)

* Which kernel version (`uname -a`)?

>Linux raspberrypi 6.1.0-rpi8-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux

**Logs**

**Additional context**

Just2Confuse, 2024-02-06:
Same behavior. 

```
me@Pi5:~ $ vcgencmd version
2024/01/22 14:44:36
Copyright (c) 2012 Broadcom
version e891ded6 (release) (embedded)
me@Pi5:~ $ vcgencmd get_config sdram_freq_min
sdram_freq_min=0
me@Pi5:~ $ uname -a
Linux Pi5 6.1.0-rpi8-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux
me@Pi5:~ $

```

timg236, 2024-02-06:
Neither Pi5 nor Pi4 provide an option to change the SDRAM frequency so the option will return zero because it's in common code.

closed 2024-02-06 by timg236