raspberrypi/firmware #1866: Fast boot RPI4

thefrankk, 2024-02-07:
Is there any way to speed up the bootloader? I have 6s from power up to the rainbow screen and i need to speed up, I need to run a simple QT app and i need to fast boot as fast as possible.  I think is to much time having 6-7s from power up to the kernel initalization.

Thank you.

pelwell, 2024-02-07:
You are far from the first person to ask that question. Unfortunately for you, initialising the DRAM controller accounts for most of this time, and that can't be skipped. FYI it has taken quite a bit of effort to get this delay as low as it is.

thefrankk, 2024-02-07:
I know, I'm just want to make sure that I did all the research and there is no solution for this. 
It's kinda crazy that a rpi3 can boot on 2 sec but in rpi4 took a lot of time more.
Can you help me by recommend another board to get a 5-7s complete boot time? I'm watching the rockpro64 but I'm not sure.
Thank you.

timg236, 2024-02-07:
Pi5 ?

timg236, 2024-02-07:
Closing this because it’s not actually a bug report and the forums are the place for discussions

closed 2024-02-07 by timg236