raspberrypi/firmware #1869: PiZ2W: different use of WiFi firmware

jayofelony, 2024-02-14:
What is the reason for using different firmware for the same wifi chip in 2 different OS releases? This is taken from a Pi Zero 2W

[image]
RaspiOS Bullseye arm64
-
[image]
RaspiOS Bookworm arm64
-

pelwell, 2024-02-15:
> What is the reason for using different firmware for the same wifi chip in 2 different OS releases?

The firmware is the same - look at the version strings (*). However, the two OSes load it via different names.

Bullseye used a hacked brcmfmac driver that pulled the firmware names from Device Tree. This was necessary because the Synaptics 43436s has the same SDIO ID as the Cypress 43430/43438, but they should run the corresponding firmware. The kernel used in Bookworm had gained the ability to qualify the firmware names with the board model string (also from DT), which means we can create bcrmcfmac43430-sdio.raspberrypi,model-zero-2-w.bin as a symbolic link to brcmfmac43436s-sdio.bin and drop the kernel patch.

But obviously, wait to see what the experts on the nexmon repo say.

(*) I don't recognise that version string - it's not what ships with Bookworm - so I think you've already modified your system.

jayofelony, 2024-02-15:
Yeah this is running nexmon indeed. But we came across some weird things. As mentioned in the nexmon issue. It seems a hardware vs software thing.

pelwell, 2024-02-15:
I think I've answered your question, even though this isn't the forum for general questions, and we don't and won't support nexmon.

closed 2024-02-15 by pelwell