raspberrypi/firmware #1874: start4cd.elf/ starting with gpu_mem=16 has no debug logging

BubuOT, 2024-03-05:
**Describe the bug**
When booting with the cut down start4cd.elf firmware (via gpu_mem=16) there is no debug log output beyond the eeprom bootloader output. Maybe this is intentional because, well, it's a cut-down firmware, but from the available docs I would not have expected logging to be one of the features that are removed in this firmware version.

**To reproduce**
Add `gpu_mem=16` to your config while also enabling all debug output options (uart_2ndstage, enable_uart, BOOT_UART=1).

**Expected behaviour**
Log outputs like you get from start4.elf:
```
Firmware: f4e2138c2adc8f3a92a3a65939e458f11d7298ba Feb 29 2024 12:24:53
Starting start4.elf @ 0xfec00200 partition 0
+

MESS:00:00:02.430522:0: arasan: arasan_emmc_open
MESS:00:00:02.432183:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:02.552418:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:02.565289:0: arasan: arasan_emmc_set_clock C0: 0x00800f00 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 390000 max: 400000 delay: 5
MESS:00:00:02.616252:0: arasan: arasan_emmc_set_clock C0: 0x00800f02 C1: 0x000e0407 emmc: 200000000 actual: 25000000 div: 0x00000004 target: 25000000 min: 0 max: 25000000 delay: 1
MESS:00:00:02.780699:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:02.783861:0: brfs: File read: 1209 bytes
MESS:00:00:02.809753:0: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:02.814262:0: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:02.831048:0: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:02.835551:0: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:02.844051:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:02.847181:0: gpioman: gpioman_get_pin_num: pin DISPLAY_SDA not defined
MESS:00:00:02.853723:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:03.025190:0: gpioman: gpioman_get_pin_num: pin FLASH_0_ENABLE not defined
MESS:00:00:03.029826:0: gpioman: gpioman_get_pin_num: pin FLASH_0_INDICATOR not defined
MESS:00:00:03.037568:0: gpioman: gpioman_get_pin_num: pin FLASH_0_ENABLE not defined
MESS:00:00:03.045017:0: gpioman: gpioman_get_pin_num: pin FLASH_0_INDICATOR not defined
MESS:00:00:03.388864:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:03.394503:0: *** Restart logging
MESS:00:00:03.397146:0: brfs: File read: 1209 bytes
[...]
```

**Actual behaviour**
```
RPi: BOOTLOADER release VERSION:4fd8f1f3 DATE: 2023/05/11 TIME: 07:26:03
BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1683786363 serial fe336bb6 boardrev b03141 stc 476326
PM_RSTS: 0x00000020
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Samsung' 16Gb x1 total-size: 16 Gbit 3200
DDR 3200 0 0 16 152

Boot mode: SD (01) order f2564
SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
SD HOST: 200000000 CTL0: 0x00800f00 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
EMMC
SD retry 1 oc 0
SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
OCR c0ff8080 [0]
CID: 00150100414a5444345206f19b58d429
SD HOST: 200000000 CTL0: 0x00800f04 BUS: 25000000 Hz actual: 25000000 HZ div: 8 (4) status: 0x1fff0000 delay: 4
SD HOST: 200000000 CTL0: 0x00800f04 BUS: 50000000 Hz actual: 50000000 HZ div: 4 (2) status: 0x1fff0000 delay: 2
MBR: 0x00002000,   65536 type: 0x0c
MBR: 0x00022000,  204800 type: 0x83
MBR: 0x00054000,    2048 type: 0x00
MBR: 0x00054800,30189568 type: 0x0f
Trying partition: 0
type: 16 lba: 8192 oem: 'mkfs.fat' volume: '  V       ^ '
rsc 4 fat-sectors 64 c-count 16343 c-size 4
root dir cluster 1 sectors 32 entries 512
FAT16 clusters 16343
Trying partition: 0
type: 16 lba: 8192 oem: 'mkfs.fat' volume: '  V       ^ '
rsc 4 fat-sectors 64 c-count 16343 c-size 4
root dir cluster 1 sectors 32 entries 512
FAT16 clusters 16343
Read config.txt bytes      639 hnd 0x25
gpio_cmd: '26=ip,pu' pins: 26-26 drive: -1 fsel: 0 term: 2
Read start4cd.elf bytes   808892 hnd 0x103f
Read fixup4cd.dat bytes     3204 hnd 0x103d
0x00b03141 0x00000000 0x00001fff
MEM GPU: 16 ARM: 998 TOTAL: 1014
Firmware: f4e2138c2adc8f3a92a3a65939e458f11d7298ba Feb 29 2024 12:25:16
Starting start4cd.elf @ 0xff000200 partition 0
+

```

**System**

* Which model of Raspberry Pi? CM4 rev 1.1
* Which OS and version (`cat /etc/rpi-issue`)? n/a
* Which firmware version (`vcgencmd version`)? See logs (f4e2138c2adc8f3a92a3a65939e458f11d7298ba)
* Which kernel version (`uname -a`)? n/a

BubuOT, 2024-03-05:
@timg236 That is fair, I guess, but should probably be added to the documentation then?

closed 2024-03-05 by timg236

popcornmix, 2024-03-05:
Feel free to submit an issue [there](https://github.com/raspberrypi/documentation/issues).
Or even better submit a PR.