raspberrypi/firmware #1875: rpi400 boot regression

nullr0ute, 2024-03-08:
When booting the RPi 400 on Fedora with U-Boot  we're seeing a regression with commit 2a1e5bf (the bump to 6.6 kernel). The regression is very early as I don't see any U-Boot output on serial console. The issue is still there with the last time the firmware was updated with commit dc94391 (Feb 29th). Not seeing the issue with RPi4B (4Gb rev1.1, 8Gb rev1.4).

nullr0ute, 2024-03-08:
> Do you have an issue with latest firmware?

Yes.

Three was a uboot related fix(*) a [few weeks back](https://github.com/raspberrypi/firmware/commit/b208f8c7ce1bc3279b8bb89a3ec47c11e87c09bf)

The RPi4B works fine with the latest U-Boto, the RPi400 does not.

> (*) I believe technically the issue was in uboot rather than firmware, and was fixed in uboot v2024.04.

Possibly, but if I change the firmware from 93d3f79 to 2a1e5bf without changing anything else the rpi400 stops booting. Neither b208f8c (Feb 14th) nor dc94391 (Feb 29th) work either.

The RPi4Bs (4Gb rev1.1, 8Gb rev1.4) work with the later Feb 14th+ firmwares, but the rpi400 does not.

nullr0ute, 2024-03-08:
Going to play around with uart_2ndstage=1 from one of the comments on that other bug.

popcornmix, 2024-03-08:
> When booting the RPi 400 on Fedora with U-Boot we're seeing a regression with commit https://github.com/raspberrypi/firmware/commit/2a1e5bfd933d977a57ba4cd62f3381fdc339e29a (the bump to 6.6 kernel).

Just to confirm - are you just using the firmware (start*.elf/fixup*.dat) from that repo and not other files (the pre-built kernel/device trees etc)?

nullr0ute, 2024-03-08:
> > When booting the RPi 400 on Fedora with U-Boot we're seeing a regression with commit [2a1e5bf](https://github.com/raspberrypi/firmware/commit/2a1e5bfd933d977a57ba4cd62f3381fdc339e29a) (the bump to 6.6 kernel).
> 
> Just to confirm - are you just using the firmware (start*.elf/fixup*.dat) from that repo and not other files (the pre-built kernel/device trees etc)?

No, we're also using the DT, U-Boot will load the Linux DT for kernel by default but we're not even getting to U-Boot to do that.

popcornmix, 2024-03-08:
Can you confirm if updating the firmware (start*.elf/fixup*.dat) or other files (dtbs or anything else) caused the issue?

nullr0ute, 2024-03-18:
So with uart_2ndstage=1, I also tuned on the uart in the eeprom too, using the same card for the original RPi4B I get:

```
MESS:00:00:07.407161:0: dtb_file 'bcm2711-rpi-4-b.dtb'
MESS:00:00:07.412861:0: brfs: File read: /mfs/sd/bcm2711-rpi-4-b.dtb
MESS:00:00:07.416110:0: Loaded 'bcm2711-rpi-4-b.dtb' to 0x100 size 0xd619
MESS:00:00:07.435976:0: brfs: File read: 54809 bytes
MESS:00:00:07.479355:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:07.481956:0: dtparam: audio=on
MESS:00:00:07.496578:0: brfs: File read: 1305 bytes
MESS:00:00:07.499338:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:07.503725:0: Failed to read command line file 'cmdline.txt'
MESS:00:00:07.660763:0: brfs: File read: /mfs/sd/rpi-u-boot.bin
MESS:00:00:07.663582:0: Loaded 'rpi-u-boot.bin' to 0x200000 size 0x99d40
MESS:00:00:07.670653:0: Kernel relocated to 0x80000
MESS:00:00:07.674599:0: Device tree loaded to 0x2eff2400 (size 0xdb75)
MESS:00:00:07.682572:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:07.689922:0: uart: Baud rate change done...
MESS:00:00:07.691944:0:

U-Boot 2024.04-rc4 (Mar 12 2024 - 00:00:00 +0000)

DRAM:  947 MiB (effective 3.9 GiB)
RPI 4 Model B (0xc03111)
```

But with the RPi400 it stops when the RPi4B should be starting to display U-Boot output:
```
MESS:00:00:07.435890:0: dtb_file 'bcm2711-rpi-400.dtb'
MESS:00:00:07.441105:0: brfs: File read: /mfs/sd/bcm2711-rpi-400.dtb
MESS:00:00:07.444354:0: Loaded 'bcm2711-rpi-400.dtb' to 0x100 size 0xd61d
MESS:00:00:07.464256:0: brfs: File read: 54813 bytes
MESS:00:00:07.508896:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:07.511495:0: dtparam: audio=on
MESS:00:00:07.528605:0: brfs: File read: 1305 bytes
MESS:00:00:07.531388:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:07.535756:0: Failed to read command line file 'cmdline.txt'
MESS:00:00:07.682156:0: brfs: File read: /mfs/sd/rpi-u-boot.bin
MESS:00:00:07.684974:0: Loaded 'rpi-u-boot.bin' to 0x200000 size 0x99d40
MESS:00:00:07.692132:0: Kernel relocated to 0x80000
MESS:00:00:07.695993:0: Device tree loaded to 0x2eff2400 (size 0xdb8a)
MESS:00:00:07.704216:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:07.711316:0: uart: Baud rate change done...
MESS:00:00:07.713339:0:
```

I'm not sure why the rpi400 should be any different here, I'm still digging.

nullr0ute, 2024-03-18:
If I swap in the DT from 7273369ad and leave everything the same it boots

pelwell, 2024-03-19:
I think this was caused by a reorganisation of the downstream Pi 400 DTB as a modification to the upstream version, a change which lost the override of the stdout-path property, which is different to upstream because of the way UARTs are numbered in the downstream tree.

There's a PR (https://github.com/raspberrypi/linux/pull/6054) with a patch, but you can download a modified bcm2711-rpi-400.dtb now from the link: https://drive.google.com/file/d/1SVWO3kjNBc_alqLwRlyeq4LNR93-MzFM/view?usp=sharing

nullr0ute, 2024-03-19:
Confirmed that fixes it for me. Thanks for the update.

pelwell, 2024-03-19:
Thanks. The PR is now merged, so closing.

closed 2024-03-19 by pelwell